crankyoldgit/IRremoteESP8266 #822: About timer support in general AC model and Gree model YB0F2

reasly, 2019-07-18:
Thank you for this project. I am trying to do the same thing but found this project eventually. It is excellent!  I hope to provide some suggestions:

1. Can the generic AC model provde Timer support ? This is very important for home automation. Sometimes AC is turned on by HA, and I hope to always attach a timer setting in case close IR message is missed.

2.  Could you help add Gree model YB0F2 to the library ? Here are some format in Python for your reference:
`        self.bit_define = (
        # byte, offset, length,name, {"value":value,}
        (0,0,3,'Mode',{ 'Auto'      :0b000,\
                        'Cool'      :0b001,\
                        'Dehumid'   :0b010,\
                        'Fan'       :0b011,\
                        'Warm'      :0b100,\
                        }),\
        (0,3,1,'Power',{ 'On'      :0b1,\
                        'Off'      :0b0,\
                        }),\
        (0,4,2,'FanSpeed',{ 'Auto'      :0b00,\
                            'Low'       :0b01,\
                            'Mid'       :0b10,\
                            'High'      :0b11,\
                        }),\
        (0,6,1,'SwingFlag',{'On'     :0b01,\
                            'Off'    :0b00,\
                        }),\
        (0,7,1,'SleepFlag',{'On'     :0b01,\
                            'Off'    :0b00,\
                        }),\
        (1,0,4,'Temp',dict(zip(map(str,range(16,31)),range(0,15)))),\
        (1,4,8,'Timer',dict(zip(map(str,[t/2.0 for t in range(49)]),[(t%2+8+(t/2/10*2)+(t/2%10)*16)*(t>0) for t in range(49)]))),\
        (2,4,1,'Turbo',{'On'     :0b01,\
                        'Off'    :0b00,\
                        }),\
        (2,5,1,'Light', {'On'       :0b01,\
                         'Off'      :0b00,\
                        }),\
        (2,6,1,'Health', {'On'       :0b01,\
                         'Off'      :0b00,\
                        }),\
        (2,7,1,'DryHot', {'On'       :0b01,\
                         'Off'      :0b00,\
                        }),\
        (3,0,1,'NewAir', {'On'       :0b01,\
                         'Off'      :0b00,\
                        }),\
        (4,0,4,'VSwing', {'OFF'     :0x0,\
                          'SWING'      :0x1,\
                          'UP'         :0x2,\
                          'MID_UP'     :0x3,\
                          'MID'        :0x4,\
                          'MID_DOWN'   :0x5,\
                          'DOWN'       :0x6,\
                          'SWING_UP'   :0xB,\
                          'SWING_MID'  :0x9,\
                          'SWING_DOWN' :0x7,\
                        }),\
        (4,4,4,'HSwing', {'OFF'     :0x0,\
                          'SWING'      :0x1,\
                          'LEFT'       :0x2,\
                          'MID_LEFT'   :0x3,\
                          'MID'        :0x4,\
                          'MID_RIGHT'  :0x5,\
                          'RIGHT'      :0x6,\
                          'WIDE'       :0xC,\
                          'SWING_SPREAD'  :0xD,\
                        }),\
        (5,0,2,'TempShow', {'Off'     :0b00,\
                            'Show'    :0b01,\
                            'ShowInhouse'  :0b10,\
                            'ShowOut' :0b11,\
                        }),\
        (7,2,1,'SaveEnergy', {  'On'     :0b01,\
                                'Off'    :0b00,\
                        }),\
        )

    def cal_crc(self):
        crc = self.get_field_bits('Mode') \
            +self.get_field_bits('Temp')+5 \
            +self.get_field_bits('HSwing') \
            +self.get_field_bits('NewAir') \
            -self.get_field_bits('Power')
        return crc%16
    def fix_swing(self):
        if (self.get_field_bits('HSwing')==0x01 or self.get_field_bits('VSwing')==0x01):
            set_field_bits('SwingFlag',0x1)
        else:
            set_field_bits('SwingFlag',0x0)
    def fix_crc(self):
        crc = self.cal_crc()
        self.set_bits(7,4,4,crc)
`

crankyoldgit, 2019-07-18:
1. Currently, no, timers are not supported yet in the common a/c stuff. They get complicated fast between all the different models and implementations. I understand what you mean and why you want it, but you could (for now) just repeat the "off" message from HA etc if you are worried about a failed transmission. Because of that, adding that feature is a low priority.

2. See #814 / #815 for some similar/expanded work on the Gree protocol.

Can you please save me the effort and break down the expr:
`dict(zip(map(str,[t/2.0 for t in range(49)]),[(t%2+8+(t/2/10*2)+(t/2%10)*16)*(t>0) for t in range(49)])))`
i.e. is it storing it in mins_past_midnight, or units of 10 mins, or 15 mins etc? Is it a relative time or an absolute time? What are the maximums and minimums?
i.e. Turn off at 8am, or turn off in 8 hours etc?

Roping in @sillyfrog who has done python more recently than me to help decode this.

reasly, 2019-07-18:
Thanks for your reply and explaination. 
The python code is for the timer coding:
(1,4,8,'Timer',dict(zip(map(str,[t/2.0 for t in range(49)]),[(t%2+8+(t/2/10*2)+(t/2%10)*16)*(t>0) for t in range(49)])))

Meaning:  timer code is start from byte #1 +   4 bit , length is 8 bit (so {byte2[3:0], byte1[7:4]}  ), with the following mapping :   timer in hours (0.5, 1.0, 1.5 ... ) -> code of  {byte2[3:0], byte1[7:4]}    MSB -> LSB
Timer does not impact CRC

{'1.5': 25, '1.0': 24, '12.5': 43, '19.5': 155, '0.0': 0, '6.0': 104, '0.5': 9, '23.5': 61, '7.0': 120, '3.5': 57, '21.0': 28, '3.0': 56, '7.5': 121, '19.0': 154, '2.0': 40, '2.5': 41, '4.5': 73, '8.0': 136, '13.5': 59, '10.5': 11, '4.0': 72, '11.5': 27, '11.0': 26, '17.0': 122, '16.5': 107, '15.5': 91, '13.0': 58, '16.0': 106, '9.5': 153, '9.0': 152, '5.0': 88, '10.0': 10, '17.5': 123, '8.5': 137, '20.0': 12, '5.5': 89, '14.5': 75, '14.0': 74, '24.0': 76, '6.5': 105, '23.0': 60, '20.5': 13, '22.0': 44, '12.0': 42, '15.0': 90, '22.5': 45, '18.0': 138, '18.5': 139, '21.5': 29}

sillyfrog, 2019-07-18:
@reasly Just looking at the python code, also not that clear to me as to the target output format (trying both in Python 2 and 3), what's the actual mapping for the 8-bits of the timer? Can you give a couple of examples maybe (including the minimum resolution)?

This is what I'm getting with binary output at the moment:
```
In [18]: (1,4,8,'Timer',dict(zip(map(str,[t/2.0 for t in range(49)]),['{:08b}'.format(t%2+8+(t/2/10*2
    ...: )+(t/2%10)*16)*(t>0) for t in range(49)])))
Out[18]:
(1,
 4,
 8,
 'Timer',
 {'0.0': '',
  '0.5': '00001001',
  '1.0': '00011000',
  '1.5': '00011001',
  '10.0': '00001010',
  '10.5': '00001011',
  '11.0': '00011010',
  '11.5': '00011011',
  '12.0': '00101010',
  '12.5': '00101011',
  '13.0': '00111010',
  '13.5': '00111011',
  '14.0': '01001010',
  '14.5': '01001011',
  '15.0': '01011010',
  '15.5': '01011011',
  '16.0': '01101010',
  '16.5': '01101011',
  '17.0': '01111010',
  '17.5': '01111011',
  '18.0': '10001010',
  '18.5': '10001011',
  '19.0': '10011010',
  '19.5': '10011011',
  '2.0': '00101000',
  '2.5': '00101001',
  '20.0': '00001100',
  '20.5': '00001101',
  '21.0': '00011100',
  '21.5': '00011101',
  '22.0': '00101100',
  '22.5': '00101101',
  '23.0': '00111100',
  '23.5': '00111101',
  '24.0': '01001100',
  '3.0': '00111000',
  '3.5': '00111001',
  '4.0': '01001000',
  '4.5': '01001001',
  '5.0': '01011000',
  '5.5': '01011001',
  '6.0': '01101000',
  '6.5': '01101001',
  '7.0': '01111000',
  '7.5': '01111001',
  '8.0': '10001000',
  '8.5': '10001001',
  '9.0': '10011000',
  '9.5': '10011001'})
```

crankyoldgit, 2019-07-18:
@sillyfrog haha. I came to pretty much the same conclusion. I was just typing _"My gawd that's disgusting encoding"_ when your update popped up.

crankyoldgit, 2019-07-18:
My analysis so far is 4th LSB (i.e. `0b1000`) is the "is the timer enabled flag?"

The time units are half hours.
```
String timer = String((x & 0b110) >> 1);  // Tens of hours. e.g. [0-2]
timer += String(x >> 4);  // 1's of hours. e.g. [0-9]
timer += (x & 0b1) ? ".5" : ".0";  // half hours.
```
@reasly  Can we please have some raw data from `IRrecvDumpV2` with the timer set to:
"0.5", "1.0", "1.5", "9.5", "10.0", "11.0", "19.0", "20.0", "21.0", and finally, the max value, which I think is "24.0", correct?

crankyoldgit, 2019-07-18:
Or as a integer: (Untested)
`uint8_t num_of_half_hours = (((x & 0b110) >> 2) * 10 + (x >> 4)) * 2 + (x & 0b1);`
`bool is_timer_set = x & 0b1000;`

reasly, 2019-07-19:
Leaving all 8 bits to zero is disableing the timer.

`1. turn on the AC without timer
Last IR Received: 24,0C0C005000200048 (2 seconds ago)
                       ^  ^
                       These two digit = 0x00

2. With 0.5 hour timer
Last IR Received: 18,0C9C0050002000480C9C006000E00101 (2 seconds ago)
                       ^  ^
                       These two digit = 0x09 = 0000 1001
Note: please ignore the second 64 bit code (0C9C006000E00101), I think it is for some compability.
Without it my AC works well with timer , if I only send 0C9C005000200048

3. Turn off timer only
Last IR Received: 24,0C0C005000200048 (7 seconds ago)
                       ^  ^
                       These two digit = 0x00

Others:
With  0.5 hour timer  : Last IR Received: 18,0C{9}C0{0}50002000480C9C006000E00101 (2 seconds ago)
With  1.0 hour timer  : Last IR Received: 18,0C{8}C0{1}50002000580C8C016000C003F1 (2 seconds ago)
With  1.5 hour timer  : Last IR Received: 18,0C{9}C0{1}50002000580C9C016000A005D1 (2 seconds ago)
With  2.0 hour timer  : Last IR Received: 18,0C{8}C0{2}50002000680C8C0260008007C1 (2 seconds ago)
With  2.5 hour timer  : Last IR Received: 18,0C{9}C0{2}50002000680C9C0260006009A1 (2 seconds ago)
With  3.0 hour timer  : Last IR Received: 18,0C{8}C0{3}50002000780C8C036000400B91 (1 second ago)
....                                           { }  { }
With  7.0 hour timer  : Last IR Received: 18,0C{8}C0{7}50002000B80C8C076000401AE1 (2 seconds ago)
With  7.5 hour timer  : Last IR Received: 18,0C{9}C0{7}50002000B80C9C076000201CC1 (2 seconds ago)
....                                           { }  { }
With  9.0 hour timer  : Last IR Received: 18,0C{8}C0{9}50002000D80C8C096000C02191 (1 second ago)
With  9.5 hour timer  : Last IR Received: 18,0C{9}C0{9}50002000D80C9C096000A02371 (2 seconds ago)
....                                           { }  { }
With 10.0 hour timer  : Last IR Received: 18,0C{A}C0{0}50002000480CAC0060008025C1 (4 seconds ago)
With 10.5 hour timer  : Last IR Received: 18,0C{B}C0{0}50002000480CBC0060006027A1 (2 seconds ago)
With 11.0 hour timer  : Last IR Received: 18,0C{A}C0{1}50002000580CAC016000402991 (1 second ago)
...                                            { }  { }
With 16.0 hour timer  : Last IR Received: 18,0C{A}C0{6}50002000A80CAC066000003CB1 (2 seconds ago)
With 16.5 hour timer  : Last IR Received: 18,0C{B}C0{6}50002000A80CBC066000E03D91 (2 seconds ago)
...                                            { }  { }
With 20.0 hour timer  : Last IR Received: 18,0C{C}C0{0}50002000480CCC006000004B61 (4 seconds ago)
...                                            { }  { }
With 24.0 hour timer  : Last IR Received: 18,0C{C}C0{4}50002000880CCC046000005AB1 (2 seconds ago)`

reasly, 2019-07-19:
Please check
`1. turn on the AC without timer
Last IR Received: 24,0C0C005000200048 (2 seconds ago)
                       ^  ^
                       These two digit = 0x00

2. With 0.5 hour timer
Last IR Received: 18,0C9C0050002000480C9C006000E00101 (2 seconds ago)
                       ^  ^
                       These two digit = 0x09 = 0000 1001
Note: please ignore the second 64 bit code (0C9C006000E00101), I think it is for some compability.
Without it my AC works well with timer , if I only send 0C9C005000200048

3. Turn off timer only
Last IR Received: 24,0C0C005000200048 (7 seconds ago)
                       ^  ^
                       These two digit = 0x00

Others:
With  0.5 hour timer  : Last IR Received: 18,0C{9}C0{0}50002000480C9C006000E00101 (2 seconds ago)
With  1.0 hour timer  : Last IR Received: 18,0C{8}C0{1}50002000580C8C016000C003F1 (2 seconds ago)
With  1.5 hour timer  : Last IR Received: 18,0C{9}C0{1}50002000580C9C016000A005D1 (2 seconds ago)
With  2.0 hour timer  : Last IR Received: 18,0C{8}C0{2}50002000680C8C0260008007C1 (2 seconds ago)
With  2.5 hour timer  : Last IR Received: 18,0C{9}C0{2}50002000680C9C0260006009A1 (2 seconds ago)
With  3.0 hour timer  : Last IR Received: 18,0C{8}C0{3}50002000780C8C036000400B91 (1 second ago)
....                                           { }  { }
With  7.0 hour timer  : Last IR Received: 18,0C{8}C0{7}50002000B80C8C076000401AE1 (2 seconds ago)
With  7.5 hour timer  : Last IR Received: 18,0C{9}C0{7}50002000B80C9C076000201CC1 (2 seconds ago)
....                                           { }  { }
With  9.0 hour timer  : Last IR Received: 18,0C{8}C0{9}50002000D80C8C096000C02191 (1 second ago)
With  9.5 hour timer  : Last IR Received: 18,0C{9}C0{9}50002000D80C9C096000A02371 (2 seconds ago)
....                                           { }  { }
With 10.0 hour timer  : Last IR Received: 18,0C{A}C0{0}50002000480CAC0060008025C1 (4 seconds ago)
With 10.5 hour timer  : Last IR Received: 18,0C{B}C0{0}50002000480CBC0060006027A1 (2 seconds ago)
With 11.0 hour timer  : Last IR Received: 18,0C{A}C0{1}50002000580CAC016000402991 (1 second ago)
...                                            { }  { }
With 16.0 hour timer  : Last IR Received: 18,0C{A}C0{6}50002000A80CAC066000003CB1 (2 seconds ago)
With 16.5 hour timer  : Last IR Received: 18,0C{B}C0{6}50002000A80CBC066000E03D91 (2 seconds ago)
...                                            { }  { }
With 20.0 hour timer  : Last IR Received: 18,0C{C}C0{0}50002000480CCC006000004B61 (4 seconds ago)
...                                            { }  { }
With 24.0 hour timer  : Last IR Received: 18,0C{C}C0{4}50002000880CCC046000005AB1 (2 seconds ago)`

reasly, 2019-07-19:
`1. turn on the AC without timer`
`Last IR Received: 24,0C0C005000200048 (2 seconds ago)`
`                       These two digit = 0x00`
`2. With 0.5 hour timer`
`Last IR Received: 18,0C{9}C0{0}50002000480C9C006000E00101 (2 seconds ago)`
`                       These two digit = 0x09 = 0000 1001`
`Note: please ignore the second 64 bit code (0C9C006000E00101), I think it is for some compability.`
`Without it my AC works well with timer , if I only send 0C9C005000200048`
`3. Turn off timer only`
`Last IR Received: 24,0C0C005000200048 (7 seconds ago)`
`                       These two digit = 0x00`
`                       `
`Others:`
`With  0.5 hour timer Received: 18,0C{9}C0{0}50002000480C9C006000E00101 (2 seconds ago)`
`With  1.0 hour timer Received: 18,0C{8}C0{1}50002000580C8C016000C003F1 (2 seconds ago)`
`With  1.5 hour timer Received: 18,0C{9}C0{1}50002000580C9C016000A005D1 (2 seconds ago)`
`With  2.0 hour timer Received: 18,0C{8}C0{2}50002000680C8C0260008007C1 (2 seconds ago)`
`With  2.5 hour timer Received: 18,0C{9}C0{2}50002000680C9C0260006009A1 (2 seconds ago)`
`With  3.0 hour timer Received: 18,0C{8}C0{3}50002000780C8C036000400B91 (1 second ago)`
`....                                { }  { }`
`With  7.0 hour timer Received: 18,0C{8}C0{7}50002000B80C8C076000401AE1 (2 seconds ago)`
`With  7.5 hour timer Received: 18,0C{9}C0{7}50002000B80C9C076000201CC1 (2 seconds ago)`
`....                                { }  { }`
`With  9.0 hour timer Received: 18,0C{8}C0{9}50002000D80C8C096000C02191 (1 second ago)`
`With  9.5 hour timer Received: 18,0C{9}C0{9}50002000D80C9C096000A02371 (2 seconds ago)`
`....                                { }  { }`
`With 10.0 hour timer Received: 18,0C{A}C0{0}50002000480CAC0060008025C1 (4 seconds ago)`
`With 10.5 hour timer Received: 18,0C{B}C0{0}50002000480CBC0060006027A1 (2 seconds ago)`
`With 11.0 hour timer Received: 18,0C{A}C0{1}50002000580CAC016000402991 (1 second ago)`
`...                                 { }  { }`
`With 16.0 hour timer Received: 18,0C{A}C0{6}50002000A80CAC066000003CB1 (2 seconds ago)`
`With 16.5 hour timer Received: 18,0C{B}C0{6}50002000A80CBC066000E03D91 (2 seconds ago)`
`...                                 { }  { }`
`With 20.0 hour timer Received: 18,0C{C}C0{0}50002000480CCC006000004B61 (4 seconds ago)`
`...                                 { }  { }`
`With 24.0 hour timer Received: 18,0C{C}C0{4}50002000880CCC046000005AB1 (2 seconds ago)`

reasly, 2019-07-19:
BTW: all document or code say the code is 64 bit with the byte 7 LSB 4 bit=0x5, I think this is some kind of REMOTE version infomation.  When timer is set, the remote will send two code (bit gap in between), on with 0x5 and one with 0x6. My AC works well if only sending 0x5 code.

Here I paste some  dual code sent waveform (may have some errors) :
`-1,89670D42;178,34976,9048,4446,692,514,692,516,714,1590,694,1616,692,514,692,514,692,512,694,514,692,512,690,514,692,1616,692,1616,690,512,694,512,712,1596,692,1616,690,512,696,512,692,1618,690,514,1898,514,692,514,692,512,1072,148,950,198,734,514,694,514,1080,1226,692,514,692,1616,692,514,690,512,692,1614,718,488,692,19946,692,512,668,538,668,538,690,514,692,514,690,516,690,514,694,514,690,512,692,514,672,534,690,514,692,512,692,1616,688,518,690,516,690,514,694,512,694,514,692,514,690,516,692,512,692,514,692,514,692,514,692,514,690,514,690,1618,692,512,692,514,668,536,692,1614,692,39960,9028,4444,690,516,690,516,666,1638,694,1614,692,516,668,538,690,514,692,514,692,514,692,514,692,1616,690,1618,666,536,690,516,668,1640,690,1616,668,536,666,538,668,1640,690,512,670,538,692,512,692,514,666,538,690,514,690,514,694,514,690,516,668,536,694,1616,688,1616,694,514,668,536,694,1616,690,514,668,19972,692,514,690,514,690,516,688,514,694,512,692,514,692,516,666,536,692,514,692,516,666,538,670,536,692,512,668,538,692,514,690,512,692,514,678,1632,694,512,692,1616,688,1618,690,512,694,1614,690,512,694,1618,664,538,692,512,692,512,694,1618,688,1618,690,514,692,1614,670,140 `

reasly, 2019-07-19:
I would not suggest to use timer enable, becasue I did not test leaving it to disable and leave other bits non-zero.

`t = int_timer_value_in_terms_of_half_hour;`
`if (t==0)  code = 0x00;`
`else  code = t%2+8+(t/2/10*2)+(t/2%10)*16;`

> My analysis so far is 4th LSB (i.e. `0b1000`) is the "is the timer enabled flag?"
>
> The time units are half hours.
>
> ```
> String timer = String((x & 0b110) >> 1);  // Tens of hours. e.g. [0-2]
> timer += [...]

crankyoldgit, 2019-07-19:
@reasly Thanks for that data. It's useful to us. At first glance it confirms my my understanding for the time encoding is correct. I'll look at coding something up soon-ish for it.

crankyoldgit, 2019-07-22:
@reasly Just an update. I haven't forgotten about this. It's just a lower priority at the moment, and I'm juggling a few too many things at present.

crankyoldgit, 2019-07-23:
@reasly I'm just getting back to this. Can you please supply me a couple of suitable `raw` message captures using `IRrecvDumpV2` so I can verify I can accurately decode and reproduce a real message? e.g. Please include what the actual remote says the message should be too.

Looking at the data so far provided, I'm fairly sure this is actually the "Kelvinator" (18) protocol. Not Gree (24).

If you have a look here, the structure is eerily similar to: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Kelvinator.h#L62

crankyoldgit, 2019-07-25:
Friendly ping.

reasly, 2019-07-25:
I only have IR MQTT server, here are some infomation:

1. Power off AC:
Last IR Received: 24,110C005002200090 (2 seconds ago)

2. Power on , Fan : low, VSwing: fix to top, HSwing: off  , Temp 28
Last IR Received: 24,190C005002200010 (1 second ago)

3. Mode switch to AUTO
Last IR Received: 24,08090050002000D0 (16 seconds ago)

4. Cool, Fan : high, VSwing: swing , HSwing : off, Temp 28
Last IR Received: 24,790C005001200010 (6 seconds ago)

5. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23
Last IR Received: 24,3907005041200000 (2 seconds ago)

7. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23
Last IR Received: 24,3907005041200000 (2 seconds ago)

7. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23, light on
Last IR Received: 24,3907205041200000 (4 seconds ago)

8. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23, light on, Show tempreture
Last IR Received: 24,3907205041210000 (4 seconds ago)

9. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23, light on, Show inhouse tempreture
Last IR Received: 24,3907205041220000 (3 seconds ago)

10. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23, light on, Show outdoor tempreture
Last IR Received: 24,3907205041230000 (4 seconds ago)

9. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23, light on, Show outdoor tempreture, Sleep mode
Last IR Received: 24,B907205041230000 (10 seconds ago)

10. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23, light on, Show outdoor tempreture, Sleep mode, Timer : 0.5h
Last IR Received: 18,B997205041230000B997206000E00181 (3 seconds ago)

11. Cool, Fan: high, VSwing: swing, HSwing: fix to middle, Temp 23, light on, Show outdoor tempreture, Sleep mode, Timer : off
Last IR Received: 24,B907205041230000 (5 seconds ago)

As you can see, only when TIMER is set on, the message is in Kelvinator format , and for others it is Gree.

crankyoldgit, 2019-07-26:
I really need to see the raw output of `IRrecvDumpV2` for the `24,` ones. to confirm a few things.

crankyoldgit, 2019-07-31:
A friendly reminder that I need a _full_ capture via `IRrecvDumpV2`.
I suspect the `24` (Gree) messages are matching an `18` (Kelvinator) message that the library is incorrectly disregarding as a Kelvinator message.

I need that info to confirm/reject that theory.

crankyoldgit, 2019-08-09:
Last call for that data from "IRrecvDumpV2", as with out it I can't be sure what is going on here.

crankyoldgit, 2019-08-12:
Marking closed till we hear back from @reasly as there has been no response for 18 days

closed 2019-08-12 by crankyoldgit

crankyoldgit, 2019-08-28:
FYI: The changes mentioned above have been included in v2.6.5 of the library which has just been released.