crankyoldgit/IRremoteESP8266 #834: Unattended signal sent to the AC

ldellus, 2019-07-26:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._

### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar branded BBQ turns on and cooks me some ribs.

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2019-07-27:
You can check the "Info" page on IRMQTTServer and it will tell you the last time it sent an IR message. Obviously check if that matches up with the A/C turning off. If it doesn't. Then perhaps the raw recorded message you have been using uses/has a timer function in it and that's what is triggering the turn off.

Something else to check is if sending the same raw message also turns the A/C off. i.e. it's a toggle message.

10240 is probably overkill. :)
If you want I can probably add basic support for your AC fairly quickly. i.e. sending via a short hex code, instead of raw, if you supply me a raw dump from the latest IRrecvDumpV2, and tell me the Brand & Model numbers for the A/C and the remote.

ldellus, 2019-07-27:
[Carrier AC Raw Data.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/3437923/Carrier.AC.Raw.Data.xlsx)
Thanks for the quick reply. After some debugging, I noticed the following:

1. When I power on the NodeMCU, it turns off the AC. I checked the "info" page and indeed an OFF message was sent.
    "Last MQTT command seen: (topic) 'Carrier/send' (payload) 
    '30,38000,8186,4324,1444,580,470,1634,470,1634,470,1820,....................
    (1 minutes 05 seconds ago)"
After 22 min. it turned off again the AC, and the "info" page contains the same OFF message. The question is, who is sending the MQTT command? I didn't initiate it and neither Homebridge, I checked the Homebridge log, nothing.
    Last MQTT command seen: (topic) 'Carrier/send' (payload) 
    '30,38000,8186,4324,1444,580,470,1634,470,1634,470,1820,....................
    (8 seconds ago)
Even if the AC is off, it sends an unattended OFF message randomly. Just happened now,

2. Good point about the timer, it is a supported function, but the AC turns off randomly, sometimes after 2 min., sometimes after 30 min. I have not recorded the timer raw codes, so I can't see how it could be the problem. Also, as I mentioned above, it happens even if the AC is off.

3. There is a specific raw code for turning off the AC, it's not a toggle.

4. I'm attaching an Excel file with all the raw data I recorded and their corresponding function. The AC is an 18 years old Carrier sub-brand, the controller model is TAC-444, and the remote is TAC-495. I think the encoding is NEC. I didn't find any meaningful information about this AC on the web.

5. I'm not knowledgeable enough to convert the raw data to HEX codes, which is something that would really help me. I tried to emit the HEX codes recorded by IRrecvDumpV2 but it didn't work. If  you can add support for my AC that would be wonderful and I'll be forever grateful.

Cheers,

Laurent

crankyoldgit, 2019-07-27:
I'm pretty sure you've set **retain** on the 'Carrier/send' topic message you are getting on power on.
Thus every time you (re)-connect to the MQTT server for what ever reasons it's going to get that same message again. Basically, go clear the retained MQTT message.

Depending on the version of esp core library you are using, you are probably losing connection and then reconnecting every 20-30 mins. Check the reconnect info on that page too.

TL;DR:
`mosquitto_pub -t ''Carrier/send' -r -n`

You should NOT use retain on that `send` topic.

crankyoldgit, 2019-07-27:
@ldellus I had a quick look at the raw data in the spreadsheet, and it looks "funky" at the very least.
Some of the raw data values are "odd" in that there is an even number of them. Typically there are an odd number of entries in the array.

Anyway. Can I get you to grab the full _text_ from a capture using the latest `IRrecvDumpV2` and put it here please?

I think it's using an encoding where each bit has a total time of 2000 usec. e.g. a mark of ~1600 and a space of ~400, or a mark of ~400 and a space of ~1600. Being either 1 or a 0.

We've got some protocols that use similar, but they are unusual and will take a bit longer to support.

You said it was a "sub-brand" of Carrier. Can you list the Brand as marked please?

ldellus, 2019-07-27:
@crankyoldgit My AC is branded Amcor ADR-853H. I cleared the retained MQTT message. The problem didn't show up again since the reboot and Node Red flows deletion.
I'm using a different IR Dump code running on Arduino Uno because, for some unclear reasons, the IRrecvDumpV2 on NodeMCU is very noisy and I get a lot of false raw data.

Here is a dump with explanations on the state of the AC. Let me know if it helps and if you need more.

Turning on the AC in cooling mode, 27 deg C. By the way, it always sends 2 packets.
--------------------------------------------------
Unknown encoding: DD3094E6 (32 bits)
Raw (101): 8100 -1150 350 -2750 1550 -450 600 -1500 550 -1550 600 -1650 400 -1500 600 -1500 550 -1550 550 -1700 1450 -450 600 -1500 1650 -450 550 -1700 400 -1500 550 -1550 1650 -450 550 -1700 400 -1500 1650 -450 1600 -500 550 -1700 1450 -450 1600 -500 600 -1500
550 -1700 400 -1500 550 -1550 550 -1550 550 -1700 400 -1500 550 -1550 600 -1500 550 -1700 400 -1500 550 -1550 550 -1550 550 -1700 400 -1500 550 -1550 550 -1500 600 -1700 400 -1500 550 -1550 550 -1500 600 -1700 1450 -450 1600 -500 600 -1450 600 -1700

Unknown encoding: E73D69C8 (32 bits)
Raw (101): 8200 -1200 200 -2850 1500 -500 550 -1550 550 -1500 600 -1700 400 -1500 550 -1550 550 -1500 600 -1700 1400 -500 600 -1500 1600 -450 600 -1700 400 -1500 550 -1550 1600 -450 650 -1650 400 -1500 1600 -500 1650 -400 600 -1700 1400 -500 1600 -450 600 -1500
600 -1700 400 -1500 550 -1550 550 -1500 550 -1750 400 -1500 550 -1500 600 -1500 550 -1750 350 -1550 550 -1500 600 -1500 550 -1700 450 -1500 550 -1500 600 -1500 550 -1750 400 -1500 550 -1500 600 -1500 600 -1700 1400 -500 1600 -450 600 -1500 550 -1700

Turning off the AC
--------------------
Unknown encoding: 3F1633F9 (32 bits)
Raw (101): 8150 -4250 1500 -500 550 -1550 550 -1550 550 -1700 400 -1500 550 -1550 550 -1550 550 -1700 1400 -500 550 -1550 1600 -500 550 -1700 400 -1500 600 -1500 1600 -500 550 -1700 400 -1500 1600 -450 1650 -500 550 -1700 1400 -500 1600 -500 550 -1550 550 -1700
400 -1500 550 -1550 550 -1500 550 -1750 400 -1500 550 -1550 550 -1500 600 -1700 400 -1500 550 -1550 550 -1500 550 -1750 400 -1500 550 -1550 550 -1500 550 -1750 400 -1500 550 -1550 550 -1500 550 -1750 400 -1500 550 -1500 1600 -500 1600 -700 400 -1500

Unknown encoding: 87DB51BE (32 bits)
Raw (101): 8100 -4300 1550 -450 600 -1500 600 -1500 600 -1650 450 -1450 600 -1500 600 -1500 600 -1650 1450 -450 600 -1500 1650 -450 550 -1700 400 -1500 600 -1500 1650 -450 550 -1700 450 -1450 1650 -450 1650 -450 600 -1650 1450 -450 1650 -450 550 -1550 600 -1650
400 -1500 600 -1500 600 -1500 600 -1650 450 -1450 600 -1500 600 -1500 600 -1650 400 -1500 600 -1500 600 -1500 600 -1650 400 -1500 600 -1500 600 -1500 600 -1650 400 -1500 600 -1500 600 -1500 600 -1650 400 -1500 600 -1500 1650 -450 1600 -650 400 -1500

crankyoldgit, 2019-07-27:
Sorry, I wanted the "full" output. e.g.
everything from the timestamp & version info up to and including the `uint16_t rawData = {...` line

crankyoldgit, 2019-07-27:
> I'm using a different IR Dump code running on Arduino Uno

Ahhhh.

crankyoldgit, 2019-07-27:
FYI, you can tune away some of the noisy useless signals by tweaking this line in `IRrecvDumpV2` https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino#L94

crankyoldgit, 2019-07-27:
Are you using the very same (physical) IR demodulator between the UNO and the ESP? i.e. If not, it might be a bad unit, or maybe a bad connection.

ldellus, 2019-07-27:
@crankyoldgit Same physical IR demodulator. I'll look at the noise tweaking.

ldellus, 2019-07-27:
@crankyoldgit It works, I had to tweak the parameter and use an external 5V source. See attached file.
[Raw Data.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3438219/Raw.Data.txt)

crankyoldgit, 2019-07-27:
Thanks. I'm halfway through writing a decoder for it.

ldellus, 2019-07-27:
@crankyoldgit you're awesome!

crankyoldgit, 2019-07-27:
@ldellus Please download and test the code in the "amcor" branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/amcor

Let me know how it goes.

NOTE: Don't go crazy using the codes it produces just yet. They are not firm at all.

To confirm I've "possibly" got it, can you please dump the `uint64_t` data/values it produces from your A/C remote for a range of temperatures, starting from the lowest stepping up one increment till you reach the max temp. I suggest the cool mode etc you had for the first entry in the text file.
Keep all other settings off/the same/as simple as possible please.

This will help me work out if I've got the bit ordering correct (otherwise we need to reverse it) and if we have the bits that are 0 & 1 correct (otherwise I will need to invert them).

Now, it's time for me to Zzzzzz.

Fingers crossed I got it right.

ldellus, 2019-07-27:
[Amcor Raw Data Temperature.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3438377/Amcor.Raw.Data.Temperature.txt)
@crankyoldgit Recompiled from the branch and tried sending a command via the "Send a simple IR message" using the new Amcor AC type. It doesn't work. I assume I have to send one of the HEX values from the raw dump. I kept the Bit size default.
I'm attaching a raw dump starting from the lowest temperature to the highest. The AC is in Cooling mode and fan is in Auto mode.  I'll be traveling the whole week so please don't expect a lot of feedback until I'm back home. Thanks a lot.

crankyoldgit, 2019-07-28:
@ldellus I've committed some tweaks to that branch that should improve the reliability of detection for you. When you get back, please download and try it out.

crankyoldgit, 2019-07-28:
Just so I don't have to pull up the text file each time I want this data:
```
24C = 80820C00000C0030 
25C = 80824C00000C0070 
26C = 80822C00000C0008 
27C = 80826C00000C0048 
```

Any chance you happen to know what the max and min temps are for the remote?

crankyoldgit, 2019-07-28:
Analysis so far. 1's are indeed 1's (i.e Bit flipping not needed). However the sequence needs to be reversed. So, don't go crazy capturing data till I say so. i.e. All the previous stuff will need to be thrown out.

crankyoldgit, 2019-07-28:
@ldellus I believe with the latest commit to that branch I've got the bit ordering correct.
IRrecvDumpV2 should now produce a `uint8_t state[8] ={...};` line which should be fine/safe to use going forward.

FYI, the lower 5 bits of `state[2]` (i.e. `state[2] & 0x1F`) should give the temp in degrees C. The last byte `state[7]` appears to contain the checksum. Just some tips etc if you want to start breaking down the detailed encoding. If you work out and can explain what all the bits/bytes mean and why they change (inc the checksum alg) I can add detailed support for your A/C.

ldellus, 2019-07-28:
> Just so I don't have to pull up the text file each time I want this data:
> 
> ```
> 24C = 80820C00000C0030 
> 25C = 80824C00000C0070 
> 26C = 80822C00000C0008 
> 27C = 80826C00000C0048 
> ```
> 
> Any chance you happen to know what the max and min temps are for the remote?

The min temperature is 11 deg C and the remote shows Lo
The max temperature is 32 deg C.

ldellus, 2019-07-28:
> @ldellus I believe with the latest commit to that branch I've got the bit ordering correct.
> IRrecvDumpV2 should now produce a `uint8_t state[8] ={...};` line which should be fine/safe to use going [...]

Thanks, I'll give it a try when I'm back from my trip next week.

crankyoldgit, 2019-08-02:
FYI, those improvements are now in the `master` branch if you want to download and try them.

crankyoldgit, 2019-08-04:
Friendly reminder for you to test it

ldellus, 2019-08-04:
Hi,

Got back from my trip but still jet lagged so expect some nonsense.

See attached Excel with the code I've been able to identify. I'm still missing the checksum formula but I hope to decipher it soon.

[Amcor Codes.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/3465279/Amcor.Codes.xlsx)

crankyoldgit, 2019-08-08:
Hey @ldellus I'm just back from a trip myself. Looking at the data now.
Is there an "auto" mode for operation modes? e.g. cool, heat, vent, dry etc.

You have two values for "Off"
i.e.
`0111180000C0031B` &
`0141240000C00018`

Is or was there a corresponding "on" state from the remote. e.g. for the first one, Low Fan, Cool, 12C? 
Or can you give me a sequence of "on" (recording what ever state the remote was in), then "off" then "on" again. This will help me to try to work out what the minimum difference is to generate an off or an on signal.

also your 'fixed' value of `00003000` doesn't appear to be always that value. e.g. the first of the 'Off's I listed above, and almost all of your 11C entries.

ldellus, 2019-08-08:
Hi @crankyoldgit,

I added more information in the Excel. See attached.
1. You are correct, I missed the Auto state.
2. I added more Off signals, and I believe I identified it to be 0000C00
3. In cooling I have a Low temperature with a strange checksum
4. In heading I don't have a low but a Hi temperature with a strange checksum
5. In all other modes I don't have a low or hi temperatures
6. In Dry mode I have only 1 fan mode, Low

I listed all the mode but the checksum is wrong below the line I marked. I have hard time figuring the algorithm for the checksum.

As usual, thanks a lot for your help.
[Amcor Codes.xlsx](https://github.com/crankyoldgit/IRremoteESP8266/files/3480212/Amcor.Codes.xlsx)

crankyoldgit, 2019-08-08:
Any chance you can convert it to Google Sheets so people like me who don't have Excel can see it easier and we don't have to have multiple versions?

ldellus, 2019-08-08:
Hi @crankyoldgit ,

I'll start with a LibreOffice spreadsheet, see attached, and I'll convert it to Google Sheets later on. I hope this format is OK with you.

Thanks

[Amcor Codes.ods.zip](https://github.com/crankyoldgit/IRremoteESP8266/files/3480777/Amcor.Codes.ods.zip)

ldellus, 2019-08-08:
Hi @crankyoldgit,

Here is the Google Sheets link. Let me know if you have problems accessing it.

https://docs.google.com/spreadsheets/d/1vkgouqBk387_jhK_IfIi6TMd99FV6YHm3v8KoEZKCeU/edit?usp=sharing

Thanks

ldellus, 2019-08-08:
Hi @crankyoldgit,

I update the Googles Sheets data and it seems that the checksum is actually the last 4 bytes. I'll continue to add values during the weekend.

Thanks

crankyoldgit, 2019-08-09:
I can access the spreadsheet fine.

crankyoldgit, 2019-08-09:
What is the `Lo` temp/mode? Do you have a link to the manual at all?

crankyoldgit, 2019-08-09:
Hey @ldellus I've created PR #854 which should add most of the functionality you have so far decoded, except for "lo" mode/temp. I need more detail on when that is available and what it does.
I've also worked out the checksum. It was the sum of all the previous nibbles. i.e. read each 4 bit block and sum it together. The checksum is just the last byte.

Anyway, can you please download & test out the branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/amcor_detailed

Please let me know how it goes etc.

ldellus, 2019-08-09:
@crankyoldgit BTW, building with OTA and password set fails with:
IRMQTTServer.ino: In function 'void handleReset()':
IRMQTTServer.ino:1268:54: error: invalid operands of types 'const char [39]' and 'const char*' to binary 'operator+'
     debug("Basic HTTP authentication failure for " + kUrlWipe);
IRMQTTServer.ino: In function 'void handleReboot()':
IRMQTTServer.ino:1297:54: error: invalid operands of types 'const char [39]' and 'const char*' to binary 'operator+'
     debug("Basic HTTP authentication failure for " + kUrlReboot);

crankyoldgit, 2019-08-09:
@ldellus Eeek. I've created a new issue to fix that.

ldellus, 2019-08-09:
@crankyoldgit The aircon mode works, but is very slow. The HTTP refresh takes 20 to 30 sec. I'm not sure it's related though. I'll try to build and upload to another NodeMCU I have. 
Thanks!

crankyoldgit, 2019-08-09:
@ldellus If you re-download that branch that compile issue should be fixed now.

Re: Being slow.
Check the "Info" page. My guess is the ESP is not connecting to MQTT correctly. You probably need to reset the device and set up the MQTT connection parameters correctly.

crankyoldgit, 2019-08-09:
Added an FAQ for this: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Why_is_IRMQTTServer_very_slow_It_takes_a_while_to_respond_Whats_wrong

ldellus, 2019-08-09:
> @ldellus If you re-download that branch that compile issue should be fixed now.
> 
> Re: Being slow.
> Check the "Info" page. My guess is the ESP is not connecting to MQTT correctly. You probably need to reset the device and set up the MQTT connection parameters correctly.

@crankyoldgit Works fine on another NodeMCU, my problem.

ldellus, 2019-08-09:
@crankyoldgit Hi, here's an update
1. OTA works
2. Aircon works
3. "Lo" is a temperature value, it exists only in "Cool" mode. I guess it tells the AC to go to the lowest temperature possible.
4. "Hi" is a temperature value, it exists only in "Heat" mode. I guess it tells the AC to go to the  highest temperature possible.
5. Thanks for the checksum.
I will now start integrating the AC to my Homebridge MQTTThing plugin via Aircon. I'll keep you posted.
Thanks a lot for everything.

crankyoldgit, 2019-08-10:
I've updated the `IRAmcorAc` class to have `setMax()` and `getMax()` which will turn on "Hi" if in heating mode, and "Lo" if in Cooling mode.

I think that covers everything you've decoded thus far.

ldellus, 2019-08-10:
Thanks, I'll test it. I think this is all I need, the timer function is not needed, I can use the Homekit automation for that.

Question: I'm sending an MQTT aircon command, for example "amcor/ac/cmnd/power", and it works, the AC turns On or Off. I just don't understand why :-) because I'm not sending the protocol number. So, how does the IRMQTTServer knows what protocol to use?

ldellus, 2019-08-10:
@crankyoldgit Hey, I have a lot of disconnects from my MQTT server. I'm not sure it's related but just in case, do you have any tips for me to check?
  Server: 192.168.1.232:1883 (Connected 9 seconds ago)
  Disconnections: 20
  Client id: amcor8810a

Thanks

crankyoldgit, 2019-08-10:
Try using ESP core 2.3.0 or the latest 2.5.x version.

crankyoldgit, 2019-08-10:
> Question: I'm sending an MQTT aircon command, for example "amcor/ac/cmnd/power", and it works, the AC turns On or Off. I just don't understand why :-) because I'm not sending the protocol number. So, how does the IRMQTTServer knows what protocol to use?

IRMQTTServer remembers what A/C was last used by the "Aircon"/climate system and it uses that.

ldellus, 2019-08-10:
> Try using ESP core 2.3.0 or the latest 2.5.x version.

@crankyoldgit Hi,

I think I upgraded my ESP core but I'm not sure. I'm using PlatformIO, and I see the following versions when uploading the firmware, does it means I'm using the right version?
PLATFORM: Espressif 8266 2.2.2 > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES: toolchain-xtensa 2.40802.190218 (4.8.2), tool-esptool 1.413.0 (4.13), tool-mkspiffs 1.200.0 (2.0), tool-esptoolpy 1.20600.0 (2.6.0), f**ramework-arduinoespressif8266 2.20502.0 (2.5.2)**

Thanks!

ldellus, 2019-08-10:
@crankyoldgit Hi,

I have a problem setting a fix IP for the module. See below error message during build.

In file included from .../IRMQTTServer.ino:311:0:
IRMQTTServer.h:62:7: error: 'IPAddress' does not name a type
              const IPAddress kIPAddress = IPAddress(192, 168, 1, 238);
IRMQTTServer.h:63:7: error: 'IPAddress' does not name a type
             const IPAddress kGateway = IPAddress(192, 168, 1, 254);
IRMQTTServer.h:64:7: error: 'IPAddress' does not name a type
            const IPAddress kSubnetMask = IPAddress(255, 255, 255, 0);
.../IRMQTTServer.ino: In function 'void setup_wifi()':
.../IRMQTTServer.ino:1829:36: error: 'kIPAddress' was not declared in this scope
           wifiManager.setSTAStaticIPConfig(kIPAddress, kGateway, kSubnetMask);
.../IRMQTTServer.ino:1829:48: error: 'kGateway' was not declared in this scope
          wifiManager.setSTAStaticIPConfig(kIPAddress, kGateway, kSubnetMask);
.../IRMQTTServer.ino:1829:58: error: 'kSubnetMask' was not declared in this scope
         wifiManager.setSTAStaticIPConfig(kIPAddress, kGateway, kSubnetMask);

Thanks!

crankyoldgit, 2019-08-11:
@ldellus I'll look into it, but unless the problems directly relate to the current Amcor working of the initial issue description/message etc, they should be reported and tracked in a separate/new issue.

crankyoldgit, 2019-08-11:
> f**ramework-arduinoespressif8266 2.20502.0 (2.5.2)**

If you look on the "Info" page on your copy of `IRMQTTServer` it will tell you which ESP core you are using.

crankyoldgit, 2019-08-11:
I've updated the branch, so if you re-download it, it should allow you to compile the code when setting a static IP address. Ref #858

ldellus, 2019-08-11:
> > f**ramework-arduinoespressif8266 2.20502.0 (2.5.2)**
> 
> If you look on the "Info" page on your copy of `IRMQTTServer` it will tell you which ESP core you are using.

@crankyoldgit Hi, I'm using the correct version, ESP8266 Core Version: 2_5_2. I found out that I have far less disconnects with another NodeMCU board. I have 2 versions of the board, one Geekcreit® NodeMcu Lua ESP8266 ESP-12F, which has a lot of disconnects, and one NodeMcu Lua CH340G V3 Based ESP8266, which works better but still have some disconnects.

As usual, thanks!

ldellus, 2019-08-11:
> I've updated the branch, so if you re-download it, it should allow you to compile the code when setting a static IP address. Ref #858

@crankyoldgit Hi, it works, thanks, and I agree, it's probably not a specific issue to this branch and I should have opened a separate issue.

crankyoldgit, 2019-08-11:
> @crankyoldgit Hi, I'm using the correct version, ESP8266 Core Version: 2_5_2. I found out that I have far less disconnects with another NodeMCU board. I have 2 versions of the board, one Geekcreit® [...]

It may just be your wifi/network environment then. Trying ESP8266 core v2.3.0 would be the ultimate test. It's the most stable as far as I know, but it does have known security vulnerabilities.

ldellus, 2019-08-11:
@crankyoldgit I will wait a bit and see how it goes. It looks pretty stable now and it works fine. I'm now focusing on the integration with Homebridge/Homekit.
Thanks a lot for everything, you're awesome!

closed 2019-08-16 by crankyoldgit

crankyoldgit, 2019-08-28:
FYI: The changes mentioned above have been included in v2.6.5 of the library which has just been released.

commit 1fb9f763 (2019-08-10): Amcor: Add `set/getMax()`
* Enables the equiv. of 'Hi' and 'Lo' temp settings.
* Unit tests for the above.

Fixes #834
commit 1e011d76 (2019-08-11): Amcor: Add `set/getMax()`
* Enables the equiv. of 'Hi' and 'Lo' temp settings.
* Unit tests for the above.

Fixes #834
PR #854 (2019-08-16): Initial detailed support for Amcor A/C protocol.
* Add checking of checksum to `decodeAmcor()`
* Add methods to handle power, mode, temp, max, & fan speed.
* Add common a/c support.
* Unit tests for the above.

Fixes #834
commit 77d6c49d (2019-08-16): Initial detailed support for Amcor A/C protocol. (#854)

* Initial detailed support for Amcor A/C protocol.
* Add checking of checksum to `decodeAmcor()`
* Add methods to handle power, mode, temp, & fan speed.
* Add common a/c support.
* Unit tests for the above.

* Amcor: Add `set/getMax()`
* Enables the equiv. of 'Hi' and 'Lo' temp settings.
* Unit tests for the above.

Fixes #834