crankyoldgit/IRremoteESP8266 #835: error: no matching function for call to 'IRsend::sendRaw(unsigned int [259], int, int)

innvoation, 2019-07-27:
Thank you for sharing this project.
I want to use irsend.sendRaw, but there is a problem as below
I hope to provide some suggestions
----------------------------------------------
Arduino: 1.8.9 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (3M SPIFFS), v2 Lower Memory, Serial, None, Only Sketch, 115200"

C:\Users\user\Documents\Arduino\IRsendDemo_F\IRrecvDumpV2\IRrecvDumpV2.ino: In function 'void loop()':

IRrecvDumpV2:18:41: error: no matching function for call to 'IRsend::sendRaw(unsigned int [259], int, int)'

           irsend.sendRaw(Power_ON,259,38);

                                         
C:\Users\user\Documents\Arduino\IRsendDemo_F\IRrecvDumpV2\IRrecvDumpV2.ino:18:41: note: candidate is:

In file included from C:\Users\user\Documents\Arduino\IRsendDemo_F\IRrecvDumpV2\IRrecvDumpV2.ino:4:0:

C:\Users\user\Documents\Arduino\libraries\IRremoteESP8266\src/IRsend.h:117:8: note: void IRsend::sendRaw(uint16_t*, uint16_t, uint16_t)

   void sendRaw(uint16_t buf[], uint16_t len, uint16_t hz);

C:\Users\user\Documents\Arduino\libraries\IRremoteESP8266\src/IRsend.h:117:8: note:   no known conversion for argument 1 from 'unsigned int [259]' to 'uint16_t* {aka short unsigned int*}'

exit status 1
no matching function for call to 'IRsend::sendRaw(unsigned int [259], int, int)'

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.
---------------------------------------
and attach a code

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

unsigned int Power_ON [259]={3314, 1620,  442, 336,  492, 270,  612, 1144,  440, 390,  420, 1216,  440, 390,  494, 316,  462, 398,  442, 1194,  440, 1196,  420, 410,  440, 390,  420, 388,  442, 1216,  442, 1198,  418, 420,  512, 266,  570, 298,  498, 384,  446, 392,  440, 410,  420, 388,  442, 412,  418, 408,  434, 326,  500, 386,  444, 266,  566, 390,  442, 1216,  440, 390,  420, 410,  420, 420,  494, 268,  486, 410,  420, 390,  440, 396,  510, 1142,  420, 392,  440, 410,  420, 412,  448, 336,  472, 1240,  438, 1198,  438, 1198,  490, 1146,  418, 1218,  418, 1216,  442, 1206,  418, 1214,  440, 338,  490, 390,  440, 1196,  440, 388,  440, 344,  488, 340,  492, 356,  482, 338,  490, 340,  490, 346,  538, 318,  460, 1196,  440, 1198,  494, 318,  460, 360,  478, 1194,  494, 268,  510, 388,  494, 270,  508, 388,  442, 390,  442, 1198,  440, 1206,  442, 1192,  442, 342,  488, 388,  442, 388,  442, 388,  496, 314,  464, 388,  496, 274,  514, 384,  442, 332,  498, 1194,  444, 388,  442, 1192,  496, 268,  562, 270,  506, 398,  444, 324,  554, 314,  464, 386,  496, 262,  518, 336,  546, 266,  512, 340,  492, 396,  446, 324,  502, 328,  504, 330,  498, 334,  498, 336,  494, 386,  444, 342,  490, 346,  548, 298,  526, 308,  522, 306,  524, 310,  522, 310,  464, 274,  610, 268,  564, 268,  572, 264,  562, 310,  468, 342,  488, 386,  446, 384,  500, 1138,  444, 386,  446, 394,  446, 328,  500, 1190,  446, 328,  554, 1138,  446, 1190,  446, 326,  504, 1190,  502, 1144,  498};

void setup()
{
  Serial.begin(115200);
}

void loop() {
          irsend.sendRaw(Power_ON,259,38);
          delay(500);
    
      }

crankyoldgit, 2019-07-27:
Change `unsigned int Power_ON [259]={3314, 1620, ...` to `uint16_t Power_ON[259]={3314, 1620, ...`

That should fix your problem.

It could simply be the space between `Power_ON` and `[259]` too.
e.g. unsigned int **Power_ON[259]**={3314, 1620, ...

innvoation, 2019-07-27:
Much thanks and solve this problem.

But still I have an question
The code received by IR is different every time is right?

Thanks again

crankyoldgit, 2019-07-27:
See the FAQ: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Why_is_the_raw_data_for_a_button_or_AC_state_always_different_for_each_capture

crankyoldgit, 2019-07-27:
FYI, the code you have there in raw is a Fujitsu A/C code.

If you look at the output of `IRrecvDumpV2` you will see where I'm getting the `state[]` data from below.
This is the same as your raw `Power_ON` message.
```
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0xC1, 0x01, 0x14, 0x00, 0x00, 0x00, 0x20, 0xDA};
irsend.sendFujitsuAC(state, 16);
```

innvoation, 2019-07-27:
IR Tx is working lights will shine, but A/C no power on or off

Could you please help me confirm the problem? IRrec different value? or other?

Thanks!
-----------------------------------------------------------------------------------------------
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Example of data captured by IRrecvDumpV2.ino
uint8_t state[16] = {0x14, 0x63, 0x00, 0x10, 0x10, 0xFE, 0x09, 0x30, 0xC1, 0x01, 0x14, 0x00, 0x00, 0x00, 0x20, 0xDA};

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {
  Serial.println("a rawData capture from IRrecvDumpV2");
  irsend.sendFujitsuAC(state, 16);
  delay(2000);

}

crankyoldgit, 2019-07-27:
1st question: Have you ever successfully sent an IR message to a device with this circuit/board/config?

What does your circuit diagram look like? Did you use one of the example ones in the wiki? i.e. with a NPN transistor.

innvoation, 2019-07-27:
1.A/C never succeeded ,but modify uint8_t state TV IRsend can be successfully.

2.circuit diagram
[image]

[image]

3. Example is "IRremoteESP8266: IRsendDemo"

crankyoldgit, 2019-07-27:
Try going through these steps:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#sending-an-ir-signal

innvoation, 2019-07-27:
A/C IRsend can be working
I really appreciate your help in resolving the problem.

crankyoldgit, 2019-07-27:
What was the problem?

innvoation, 2019-07-27:
In fact, IRsend can be working, but the TX distance is only one meter
Shorter than I imagined.

closed 2019-07-27 by innvoation

crankyoldgit, 2019-07-27:
Yeah. Those particular boards are pretty poor IR TX units.

crankyoldgit, 2019-07-27:
FWIW, I get about 5-6 meters from one similar to the wiki example: https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-sending

crankyoldgit, 2019-07-27:
You do have to aim well too.

innvoation, 2019-07-27:
Ok, I will try again
Thank you for sharing the information.

mohammadowais3701, 2020-04-17:
**no matching function for call to 'IRsend::sendRaw(unsigned int [149], unsigned int, int&)'
**
Kindly help me for this to solve this error
`
#include <IRremoteESP8266.h>
#include <IRsend.h>

#define IR_SEND_PIN 4
IRsend irsend(IR_SEND_PIN);
int khz=38;
unsigned int ACoff[]={3026, 3044, 3032, 4304, 578, 1692, 550, 582, 550, 1706, 552, 712, 552, 580, 552, 1704, 552, 582, 552, 1838, 550, 582, 580, 566, 550, 582, 550, 714, 552, 582, 550, 582, 552, 582, 550, 714, 550, 582, 552, 582, 578, 1690, 552, 712, 552, 582, 550, 582, 550, 582, 550, 714, 550, 582, 552, 582, 550, 582, 578, 700, 550, 582, 550, 582, 550, 582, 550, 1838, 550, 582, 552, 580, 552, 582, 552, 712, 578, 568, 550, 582, 550, 582, 550, 714, 550, 1704, 552, 1706, 550, 582, 550, 714, 550, 582, 578, 566, 552, 582, 552, 712, 550, 582, 552, 580, 552, 1704, 552, 714, 550, 582, 552, 582, 578, 568, 550, 714, 552, 582, 552, 582, 550, 582, 550, 714, 550, 582, 552, 582, 552, 582, 578, 722, 552, 1706, 550, 582, 552, 1704, 552, 736, 552, 582, 550, 1704, 552, 1704, 552, 582, 580}; // UNKNOWN 9BB5B2B5
unsigned int ACon[]=  {3032, 3046, 3028, 4304, 576, 1692, 552, 582, 550, 1706, 550, 714, 550, 582, 552, 1706, 550, 582, 552, 1836, 552, 582, 578, 568, 552, 580, 552, 714, 550, 582, 550, 582, 552, 582, 550, 1836, 552, 582, 550, 582, 578, 1692, 550, 712, 552, 582, 552, 582, 550, 582, 550, 714, 552, 582, 550, 582, 552, 582, 578, 700, 550, 582, 550, 582, 550, 582, 552, 1838, 550, 582, 552, 582, 552, 582, 550, 714, 578, 568, 550, 582, 550, 582, 550, 714, 552, 1706, 550, 1704, 550, 582, 552, 714, 550, 582, 578, 568, 550, 582, 550, 714, 552, 582, 550, 582, 550, 1706, 550, 714, 550, 582, 550, 582, 578, 568, 550, 714, 550, 582, 550, 582, 552, 582, 550, 714, 550, 582, 552, 582, 552, 582, 578, 722, 552, 1704, 552, 582, 550, 1706, 552, 736, 550, 584, 550, 1706, 550, 1706, 550, 1704, 578};// UNKNOWN 9BB5B2B5
unsigned int Temp16[]={3030, 3042, 3032, 4304, 576, 1694, 550, 582, 552, 1706, 550, 712, 552, 580, 552, 1706, 550, 582, 550, 1836, 550, 582, 578, 568, 552, 582, 552, 1834, 552, 582, 550, 1706, 552, 1706, 548, 714, 552, 580, 552, 582, 578, 1690, 552, 714, 550, 582, 552, 582, 550, 582, 552, 712, 550, 582, 550, 582, 552, 582, 578, 700, 550, 582, 552, 582, 552, 582, 550, 1836, 552, 582, 550, 584, 550, 582, 550, 712, 580, 568, 550, 582, 550, 584, 550, 712, 552, 1704, 552, 1704, 552, 580, 552, 714, 550, 582, 580, 566, 552, 580, 552, 712, 552, 582, 550, 582, 552, 1704, 552, 712, 552, 582, 550, 582, 580, 566, 552, 712, 552, 582, 552, 582, 550, 582, 550, 714, 550, 582, 550, 582, 550, 582, 578, 722, 550, 1706, 552, 582, 550, 1704, 552, 1860, 552, 1704, 552, 1704, 552, 582, 550, 582, 580}; // UNKNOWN F35ACDF0
unsigned int Temp17[] ={3030, 3042, 3032, 4306, 574, 1692, 550, 582, 550, 1704, 552, 712, 552, 582, 552, 1704, 550, 582, 550, 1836, 552, 582, 578, 568, 552, 1704, 552, 714, 550, 582, 550, 1706, 550, 1706, 552, 712, 550, 582, 552, 580, 578, 1688, 554, 712, 552, 582, 550, 582, 552, 582, 550, 714, 550, 582, 552, 582, 550, 582, 580, 698, 550, 582, 552, 582, 550, 582, 550, 1838, 552, 580, 550, 582, 550, 582, 552, 712, 580, 568, 550, 582, 550, 582, 552, 714, 550, 1704, 552, 1706, 552, 580, 552, 712, 550, 582, 578, 568, 550, 582, 550, 714, 550, 582, 552, 582, 550, 1704, 552, 714, 552, 582, 550, 582, 578, 568, 550, 714, 550, 582, 552, 582, 550, 582, 550, 714, 552, 582, 550, 582, 550, 582, 578, 722, 552, 1704, 552, 1706, 552, 582, 550, 738, 552, 1706, 552, 1706, 550, 582, 552, 582, 578}; // UNKNOWN 4046F041
unsigned int Temp18[] = {3030, 3070, 3004, 4304, 576, 1694, 552, 580, 550, 1706, 550, 714, 550, 582, 552, 1706, 550, 580, 552, 1836, 552, 582, 578, 568, 550, 1706, 550, 1838, 550, 580, 552, 1704, 552, 1706, 550, 712, 550, 582, 552, 582, 578, 1692, 552, 714, 550, 582, 550, 582, 550, 582, 552, 714, 550, 582, 550, 582, 550, 582, 578, 698, 552, 582, 550, 582, 550, 582, 552, 1836, 552, 580, 550, 584, 550, 582, 552, 712, 578, 568, 552, 582, 550, 582, 550, 712, 552, 1706, 554, 1702, 552, 580, 550, 714, 552, 582, 578, 568, 550, 582, 552, 712, 550, 582, 550, 582, 552, 1704, 554, 710, 552, 582, 552, 582, 578, 568, 552, 712, 552, 582, 550, 582, 552, 582, 550, 714, 550, 582, 550, 582, 550, 582, 580, 722, 550, 1704, 552, 1704, 552, 582, 550, 1862, 550, 1704, 552, 1704, 552, 582, 550, 582, 578}; // UNKNOWN 1575D184
unsigned int Temp19[] = {3030, 3046, 3028, 4306, 574, 1692, 552, 582, 550, 1706, 550, 714, 552, 582, 550, 1706, 552, 580, 552, 1836, 552, 582, 578, 1690, 552, 582, 550, 714, 550, 582, 550, 1706, 550, 1704, 552, 714, 550, 582, 550, 582, 580, 1690, 552, 712, 552, 582, 550, 582, 550, 582, 550, 714, 550, 582, 550, 582, 552, 582, 578, 698, 552, 582, 552, 582, 552, 582, 550, 1836, 552, 580, 552, 582, 552, 582, 550, 714, 578, 568, 550, 582, 550, 582, 550, 714, 550, 1706, 550, 1706, 550, 582, 552, 714, 550, 582, 580, 568, 550, 584, 550, 714, 550, 582, 550, 582, 550, 1704, 552, 712, 550, 582, 550, 582, 578, 568, 550, 714, 552, 582, 552, 582, 550, 582, 550, 714, 550, 582, 550, 582, 552, 582, 578, 724, 550, 1708, 550, 1704, 552, 1704, 552, 736, 552, 1704, 552, 1706, 550, 582, 550, 582, 578};//Change it for your remote
unsigned int Temp20[] = {3030, 3044, 3030, 4304, 578, 1694, 550, 582, 552, 1708, 548, 714, 550, 582, 550, 1704, 552, 582, 550, 1838, 550, 582, 578, 1692, 550, 582, 550, 1836, 552, 582, 550, 1706, 550, 1704, 552, 712, 550, 582, 550, 582, 580, 1690, 552, 712, 552, 582, 552, 582, 550, 582, 552, 714, 550, 582, 550, 582, 550, 582, 578, 698, 552, 582, 550, 582, 550, 582, 552, 1836, 552 [... cut ...]
   irsend.sendRaw(Temp28, sizeof(Temp28) / sizeof(Temp28[0]), khz); //Send Signal to Turn On the AC 
  delay(2000);
   irsend.sendRaw(Temp29, sizeof(Temp29) / sizeof(Temp29[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);
   irsend.sendRaw(Temp30, sizeof(Temp30) / sizeof(Temp30[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);
   irsend.sendRaw(Temp16, sizeof(Temp16) / sizeof(Temp16[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);
     irsend.sendRaw(ACoff, sizeof(ACoff) / sizeof(ACoff[0]), khz);  //Send Signal to Turn On the AC 
  delay(2000);

    

> `

redPrint7, 2020-04-17:
I answered your question on stackoverflow solution is same as above change unsigned int to uint16_t dor the arrays matching function for call to 'IRsend::sendRaw(unsigned int [149], unsigned int, int&)'

mohammadowais3701, 2020-04-17:
Sorry I can't understand what did you say ,please clear your ans..Thankyou

crankyoldgit, 2020-04-18:
@mohammadowais3701 Please don't hi-jack an existing/closed issue.

> int khz=38;

`uint16_t khz = 38;`

Also you're verging on https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#i-want-to-change-the-example-code-to-make-it-do-something-different-but-i-dont-know-how-please-help-me