crankyoldgit/IRremoteESP8266 #840: RC5 Sending not working

DaKie206, 2019-07-29:
### Version/revision of the library used
2.6.2

### Expected behavior

I want to send a Hex code over RC5 (0x1)

### Actual behavior
I received consistent but senseless codes (0xe7aa2ebc).
It works fine with sendSony()

#### Example code used
`irsend.sendRC5(0x1, 32);`

#### Circuit diagram and hardware used
Simple IR-Led on NodeMCU 1.0

### Has this library/code previously worked as expected for you?
No

crankyoldgit, 2019-07-30:
RC5 is not a 32 bit protocol. Your call of `irsend.sendRC5(0x1, 32);` is forcing the `sendRC5` routine to send it as a 32 bit message. RC5 is typically a [12](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRremoteESP8266.h#L467) bit protocol.

Use `irsend.sendRC5(0x1);` to send a normal RC5 message that contains `0x1`.

DaKie206, 2019-07-30:
It works!
Thank you for your quick answer

closed 2019-07-30 by DaKie206