crankyoldgit/IRremoteESP8266 #841: freeRTOS support

beckmx, 2019-07-30:
I wanted to use the library into a FreeRTOS build, I have noticed some problems, I tried cleaning the code but there are some strong dependencies for arduino, I was wondering if there is anyway the library available for C or as O file without the arduino dependencies

### Version/revision of the library used
master

### Expected behavior
_What steps did you do and what should it have done?_
I created a esp-open-rtos project for C++ but still some dependencies are missing or not available

crankyoldgit, 2019-07-31:
There are a lot of `C++`ism in the code. e.g. classes. They won't back port to standard `C` with out **_monumental_** effort. Thus porting to straight `C` isn't going to or likely happen any time soon. Sorry.

That said, I'm happy to entertain getting it to work under the FreeRTOS framework rather than just the Arduino framework. Most of the areas of incompatibility have been identified with the use of `#ifdef ARDUINO` and `#if UNIT_TEST` through out the core code.
If you want to find out the equivalent ways of doing some of the arduino specific functions/syscalls etc in FreeRTOS, I'm happy to try to integrate that into the library.

TL;DR: No, there is no current way to use this without modification in a non Arduino-Framework. Unless you want to put in some leg work, it is not going to happen anytime soon.

crankyoldgit, 2019-08-12:
Marking closed due to inactivity.

closed 2019-08-12 by crankyoldgit