crankyoldgit/IRremoteESP8266 #842: Implementation request (Battery Voltage)

andreik2900, 2019-07-30:
Hi Boss, since I'm using your wonderfull project as IRremoteESP8266 for my Daiking AC in a little box with battery, I would like to ask you if is possible to implement an MQTT message with the battery voltage level.
Do you think that could be implemented.
Thanks

crankyoldgit, 2019-07-31:
> Do you think that could be implemented.

Yes. But the better questions is "Am I going to implement that?", to which the answer is "Probably not, but I'll keep it in mind". 

Depending what you want to do, you should look at [Tasmota](https://github.com/arendst/Sonoff-Tasmota), [ESPEasy](https://github.com/letscontrolit/ESPEasy), or [OpenMQTTGateway](https://github.com/1technophile/OpenMQTTGateway). They include this library, the ability to send and receive IR messages via it, and probably have the ability to report on the ESP voltage level out of the box. Tasmota for certain. I strongly suggest you give them a try.

That said, you could easily add it yourself to IRMQTTServer. It should be not hard to do.

@jimmys01 Can you confirm if ESPEasy can report it via MQTT?

jimmys01, 2019-07-31:
Haven't done it myself but yes you can. There are 2 ways. One is having the ESP self report with internal methods by enabling build flag `FEATURE_ADC_VCC=true`. For this you have to compile a build yourself since IR library and VCC functionality are not pre-build together to the firmware release list.

The other method is by having a voltage divider and throught the ESP ADC read the value and by knowing the voltage divider resistors values you can deduce the batery voltage. You will need to do this in EasyESP rules.
Keep the voltage divider Resistor values high because it will consume a current and that is not welcomed in battery poweered applications.

Also check this [forum ](https://www.letscontrolit.com/forum/viewtopic.php?f=4&t=6840&p=37887&hilit=voltage+divider#p37887)thread

crankyoldgit, 2019-07-31:
I should also add, there are no real have power saving options in IRMQTTServer's code. It's going to eat your battery in no time. It is not designed with battery use in mind. :-/

andreik2900, 2019-07-31:
It is just a nice feature to implement, up to you. ;-)
If it will drain the battery in short time, can be an option to enable/disable.

crankyoldgit, 2019-08-01:
@andreik2900 I've implemented it in branch `ReportVcc`, See PR #845 
It is turned off by default.

Read the comments as I'm not sure it's going to do what you want but it's the best we can do that fits all cases.

closed 2019-08-04 by crankyoldgit

crankyoldgit, 2019-08-28:
FYI: The changes mentioned above have been included in v2.6.5 of the library which has just been released.

PR #845 (2019-08-04): IRMQTTServer: Add ability to report Vcc at the ESP chip.
* Controlled by `#define REPORT_VCC`
* Turned off by default. Compile time option.
* Costs about 250 bytes of prog space when enabled.

Fixes #842
commit 1359397b (2019-08-04): IRMQTTServer: Add ability to report Vcc at the ESP chip. (#845)

* Controlled by `#define REPORT_VCC`
* Turned off by default. Compile time option.
* Costs about 250 byte of prog space.

Fixes #842