crankyoldgit/IRremoteESP8266 #847: Understanding the output of IRrecvDumpV2

macdroid53, 2019-08-01:
I have IRrecvDumpV2 (from v2.6.4) working and I'm trying to understand the output.
(I'm trying to reverse engineer the IR controls for Dyson fans...)
(I have IRrecvDumpV2 running on one ESP32. On another ESP32 I have IRSendDemo.)

Using the Dyson IR controller, I capture a press of the ON/Off button.
These controllers send a code twice when the On/Off button is pressed.
The output is as follows:

Timestamp : 000065.395
Library   : v2.6.4
`
Encoding  : UNKNOWN
Code      : 9FAA7414 (17 bits)
Raw Timing[33]:
   +  2250, -   826,    +   764, -  1586,    +   772, -   814,    +   766, -   814, 
   +   766, -  1568,    +   772, -   802,    +   768, -   796,    +   764, -   790, 
   +   770, -   848,    +   766, -   820,    +   768, -   816,    +   768, -   810, 
   +   766, -   806,    +   768, -   800,    +   766, -   796,    +   766, -  1554, 
   +   772

uint16_t rawData[33] = {2250, 826,  764, 1586,  772, 814,  766, 814,  766, 1568,  772, 802,  768, 796,  764, 790,  770, 848,  766, 820,  768, 816,  768, 810,  766, 806,  768, 800,  766, 796,  766, 1554,  772};  // UNKNOWN 9FAA7414

Timestamp : 000065.829
Library   : v2.6.4

Encoding  : UNKNOWN
Code      : 9FAA7414 (17 bits)
Raw Timing[33]:
   +  2246, -   830,    +   764, -  1584,    +   772, -   816,    +   768, -   820, 
   +   756, -  1576,    +   766, -   808,    +   764, -   858,    +   704, -   800, 
   +   762, -   854,    +   764, -   824,    +   762, -   822,    +   762, -   874, 
   +   704, -   814,    +   762, -   864,    +   706, -   804,    +   764, -  1556, 
   +   770

uint16_t rawData[33] = {2246, 830,  764, 1584,  772, 816,  768, 820,  756, 1576,  766, 808,  764, 858,  704, 800,  762, 854,  764, 824,  762, 822,  762, 874,  704, 814,  762, 864,  706, 804,  764, 1556,  770};  // UNKNOWN 9FAA7414

`
First question is what is the Code hex value represent? Is this the value of the _code_ the controller is sending? Can it be used to send?

Second, the raw timing data and subsequently the uint16_t array is slightly different for each timestamp. I selected one of the uinit16_t that was captured and I put it in IRSendDemo, modified to send only that uint16_t once every 2000ms. About half the time, capturing with IRrecvDumpV2 gets the wrong Code.

What am I missing?

jimmys01, 2019-08-01:
From the FAQ

> My remote's signal is being reported as an "Unknown" code. What does that mean? How do I send an Unknown code?
> 
> **The decode() method will try to make up a synthetic unique data value for what ever message it received even though it doesn't understand it. The code value is a result of a non-invertible hash function. It is a last-resort attempt to allow you to potentially match the same message when it is sent later.** As the value is a non-invertible hash, it is useless for reproducing the message. The bit count is also a synthetic value; it is just a count of how many high-to-low transitions it detected. Don't rely on these values at all.
> In order to reproduce the message, you are going to have to use the rawData, for example as captured by the IRrecvDumpV2 sketch. You send the raw data using the sendRaw() method as per example code. Please note that the library can't tell what frequency the captured message was transmitted at. You'll have to make an educated guess when using sendRaw(). The most common frequency used is 38kHz. That would be where you should probably start, or use the receiving frequency of the IR receiving module you are using. You can look through the various src/ir_*.cpp files for lines like enableIROut(38) to give yourself an idea of commonly used frequencies, if that doesn't work. 

Short version: If Encoding is UNKNOWN hex code means nothing

macdroid53, 2019-08-01:
Ok, Thanks!

closed 2019-08-01 by macdroid53

crankyoldgit, 2019-08-02:
See also: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Why_is_the_raw_data_for_a_button_or_AC_state_always_different_for_each_capture

happytm, 2019-09-06:
Very helpful explanation in simple terms.

Thanks