crankyoldgit/IRremoteESP8266 #848: IRMQTTServer - Loss of MQTT Server credentials after WiFi Outage

sheppy99, 2019-08-01:
#597 Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.6.3

### Expected behavior
Device retains settings after WiFi failure.

### Actual behavior
Power failure took both of my WiFi access points down, WiFi takes longer to reboot than IRMQTTServer, so presumably it went into access point mode without me noticing
Device reconnected to WiFi network after I power cycled it but it had the default MQTT Server details instead of the ones I had put in. Resetting it back to access point mode with the web interface gave me the screen to put them in again but the details were lost everytime it reconnected to my original WiFi network. It also wouldn't retain a username and password to enable OTA

### Steps to reproduce the behavior
see above

### Has this library/code previously worked as expected for you?
Yes, this version was OK the power failure

### Other useful information
After several tries I gave up and loaded the latest version of IRMQTTServer on to another ESP after editing the MQTT Server settings in IRMQTTServer.ino using a cable. I still have the original ESP with the problem and I'm happy to debug further before I flash it again. All my Tasmota Stuff came back to life without any prodding so I think its not a WiFi issue

darshkpatel, 2019-08-02:
Just to make sure, while flashing the firmware did you keep space for
SPIFFS ?

Allocate atleast 1Mb for SPIFFS to be safe.

crankyoldgit, 2019-08-02:
@sheppy99 You report you were using #597 version of the library, and say it was/is running `v2.6.3`

This is confusing as I _think_ this bug was squashed in #680 which was included with `v2.6.0`
#597 is circa Jan '19, where #680 is circa Apr '19.

What's the "info" page report the version info etc as?

Additionally, [v2.6.0](https://github.com/crankyoldgit/IRremoteESP8266/releases/tag/v2.6.0)'s release notes include this notice:
<hr>

#### IRMQTTServer: Move MQTT server and other parameters to WifiManager. (#680)
WARNING: Previous users may need to fully wipe/reset the
SPIFFS/WifiManager settings by visiting
http://<your_esp8266's_ip_address>/reset prior to or
after update.
<hr>
It fixed a bug where it may not format the SPIFFS on the device, so users would have to force it if upgrading to fix it if it had NEVER previously had a SPIFFS on the device.

I think you may be hitting that bug. Hence I would say please upgrade to the latest released version of the example code and library (v2.6.4), or the current `master` branch. AND then choose the wipe option from the 'admin' menu.
If it then still fails to store MQTT creds etc, then we probably still have an issue, but, I think this was fixed a while ago.

sheppy99, 2019-08-03:
I don't remember ever manually formatting the SPIFFS, so maybe that's it. Does choosing "Erase Flash - All Flash Contents" on the tools menu in Arduino IDE sort it? I had to do that with the replacement before it'd boot. Clicking "Wipe Settings" on the Admin page of the defective device didn't fix it although it did put it back into AP mode.

Details from the info page on the defective one are:

General

Hostname: ir_server
IP address: 192.168.1.110
Booted: 2 minutes 17 seconds ago
Version: v1.3.0
Built: Jul 15 2019 12:59:12
Period Offset: -5us
IR Lib Version: 2.6.3
ESP8266 Core Version: 2_5_2
Free Sketch Space: 3580k
Cpu Freq: 80MHz
IR Send GPIO(s): 4 (default)
Total send requests: 0
Last message sent: FAILED (Never)
IR Recv GPIO: 14
Total IR Received: 0
Last IR Received: None (Never)
Duplicate Wifi networks: Show
Min Wifi signal required: 8%
Serial debugging: Off
MQTT Information

Server: 10.0.0.4:1883 (Disconnected Never)
Disconnections: 0
Client id: ir_server20f959
Command topic(s): ir_server/send, ir_server/send_0
Acknowledgements topic: ir_server/sent
IR Received topic: ir_server/received
Log topic: ir_server/log
LWT topic: ir_server/status
QoS: 1
Last MQTT command seen: (topic) 'None' (payload) 'None' (Never)
Total published: 0
Total received: 0
Climate Information

IR Send GPIO: 4
Last update source: None
Total sent: 0
Last send: Never
State listen period: 5 seconds
State broadcast period: 10 minutes
Last state broadcast: Never
Last discovery sent: Locked
Command topics: ir_server/ac/cmnd/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend)
State topics: ir_server/ac/stat/(protocol|model|power|mode|temp|fanspeed|swingv|swingh|quiet|turbo|light|beep|econo|sleep|filter|clean|use_celsius|resend)

crankyoldgit, 2019-08-04:
> Free Sketch Space: 3580k

This indicates to me that there is no SPIFFS filesystem/partition.
You need to build the firmware with an SPIFFS partition of some size, upload that firmware and perform a wipe etc.
The SPIFFS options are in the Arduino IDE build/target menu.

I see that I didn't explicitly state that in the file comments/instructions so I'll add something that says that. I'll also look into seeing if I can detect this problem and have the program report the issue to a user.

crankyoldgit, 2019-08-04:
Nope. I just can't read. I **did** add instructions for it. See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L41-L42

sheppy99, 2019-08-05:
Thanks, I've formatted it with 1Meg SPIFFS which is the smallest amount, reloaded the sketch and swapped it back into service, its curious how it could remember only the WiFi credentials when previously it had been working. Is there any way of hard coding the preferred AP into the sketch?

crankyoldgit, 2019-08-05:
Wifi settings are stored elsewhere. Not in SPIFFS.

sheppy99, 2019-08-05:
Ah, that explains it all. Curious how it had been working though from putting the settings in and then after the repeated WiFi outages it wouldn't allow me to do that.
In any case its working now so I'll close the issue and reopen if it reoccurs next time I'm moving stuff around. As ever thanks for your prompt attention :-)

closed 2019-08-05 by sheppy99

sheppy99, 2019-08-27:
Updating this in case it helps someone.
Since formatting with SPIFFS and installing the latest version my regular MQTT disconnects have stopped and it seems rock solid now.