crankyoldgit/IRremoteESP8266 #851: IRMQTTServer to Homebridge

pasna, 2019-08-07:
Good morning @crankyoldgit ,
I would like to have your guidance.
I try to integrate IRMQTTServer to Homebridge and Homekit via MQTT, I found that, 
1) I can fully control AC with IR module using Apple Homekit, w/o problem. 
2) I can use real remote to operate AC, the IR module received hex code and show on MQTT as well as decode AC state correctly, but it did not reflecting state ( power on/off , mode and fan speed) to Homekit at all.
BTW IRMQTTServer to HASSio integrated to Homekit works great but only support thermostat characteristic, no fan speed control, no vent swing. 
Below is my config:
Thank you and best regards
> 
            "accessory": "mqttthing",
            "type": "heaterCooler",
            "name": "Living Room AC",
            "url": "http://192.168.0.96:1883",
            "logMqtt": true,
            "topics": {
                "setActive": "living_rm/ac/cmnd/power",
                "getActive": "living_rm/ac/stat/power",
                "getCurrentHeaterCoolerState": "living_rm/ac/stat/mode",
                "setTargetHeaterCoolerState": "living_rm/ac/cmnd/mode",
                "getTargetHeaterCoolerState": "living_rm/ac/cmnd/mode",
                "getCurrentTemperature": {
                    "topic": "living_rm/ac/stat/temp",
                    "apply": "return Math.round ( message );"
                },
                "setCoolingThresholdTemperature": "living_rm/ac/cmnd/temp",
                "setTemperatureDisplayUnits": {
                    "topic": "living_rm/ac/cmnd/use_celsius",
                    "apply": "return Math.round( message * 10 );"
                },
                "getTemperatureDisplayUnits": "living_rm/ac/stat/use_celsius",
                "setSwingMode": "living_rm/ac/cmnd/swingh",
                "getSwingMode": "living_rm/ac/stat/swingh",
                "setRotationSpeed": {
                    "topic": "living_rm/ac/cmnd/fanspeed",
                    "apply": "return [ 'off', 'min', 'min', 'max', 'max', 'max' ][ Math.floor( message / 20 ) ];"
                },
                "getRotationSpeed": {
                    "topic": "living_rm/ac/stat/fanspeed",
                    "apply": "return [ 'off', 'min', 'min', 'max', 'max', 'max' ][ Math.floor( message / 20 ) ];"
                }
            },
            "swingModeValues": [
                "off",
                "auto"
            ],
            "minTemperature": 20,
            "maxTemperature": 30,
            "restrictHeaterCoolerState": [
                1,
                2
            ]

crankyoldgit, 2019-08-08:
I have NO experience with Apple's Homekit at all, so I can't help you specifically.
You'll need raise your config issues with the relevant forums for Home Assistant and Homekit etc.
Configuring those is beyond the scope of this library.

Just checking you've enabled [USE_DECODED_AC_SETTINGS](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.h#L141) for it to decode the A/C remote, yes?

>       "getTargetHeaterCoolerState": "living_rm/ac/cmnd/mode",

your "get"'s should be using the `stat` topics. e.g. `"living_rm/ac/stat/mode"`

> BTW IRMQTTServer to HASSio integrated to Homekit works great but only support thermostat characteristic, no fan speed control, no vent swing.

Again, I can't speak for Homekit as I have no experience with that.
But I can control my A/C via HASSio/Home Assistant and have indirect access to fan speed and swing etc per attached image.

[image]

That's using the example home assistant config in the code comments.

pasna, 2019-08-08:
@crankyoldgit 
Thank you. I will be using Homekit and HA for a while.

pasna, 2019-08-08:
@crankyoldgit sorry to bug you again,
I just notice one more thing is that if I power off the IR module for a while and power it back on but I did not load web service for IRMQTTServer, IR module received infrared command from real remote and display on mqtt monitor but will not decode. State will not change as It is in deep sleep.

crankyoldgit, 2019-08-08:
I'm not sure what you are trying to say. IRMQTTServer doesn't have or use any "deep sleep".

hassbian-ABC, 2019-08-10:
This is a homebridge plugin I wrote, If you find bugs, please submit them to issues or QQ Group: 107927710.
@crankyoldgit @pasna

hassbian-ABC, 2019-08-10:
This is a homebridge plugin I wrote, If you find bugs, please submit them to issues or QQ Group: 107927710.  https://github.com/hassbian-ABC/homebridge-mqtt-acpartner
@crankyoldgit @pasna

crankyoldgit, 2019-08-11:
FYI: Added to https://github.com/crankyoldgit/IRremoteESP8266/wiki/Related-User-Projects

crankyoldgit, 2019-08-12:
> @crankyoldgit sorry to bug you again,
> I just notice one more thing is that if I power off the IR module for a while and power it back on but I did not load web service for IRMQTTServer, IR module [...]

@pasna asking again. Can you please re-clarify what the issue is? I'm not sure what the above means. IRMQTTServer doesn't use and deep sleep mode etc.

pasna, 2019-08-13:
> @pasna asking again. Can you please re-clarify what the issue is? I'm not sure what the above means. IRMQTTServer doesn't use and deep sleep mode etc.

@crankyoldgit  sorry for delay responding. Initially I thought the IRMQTTServer module show sleep -1 on MQTT monitor was equal to sleep mode. I tried to figure what went wrong with my config setting. After resetting my homebridge plugin and reconfig, I can now sync IRMQTTServer with real remote and homekit on 2 functions ie. power on/off and horizontal swing. I will try new pluging from @hassbian-AB.
Thank you.