crankyoldgit/IRremoteESP8266 #852: Web interface for this library

mariusmotea, 2019-08-07:
Hi,

This is not an issue!

As most of the users (including me) are looking for a way to control their AC units using this library but don't find any GUI i want to share what i made as it can be very useful for the guys without any coding skills. If you believed this useful and want to add it in examples i can create a PR (or simply copy my repo).

https://github.com/mariusmotea/esp8266-AC-control

darshkpatel, 2019-08-07:
It would be great if you can send a PR for this :D

crankyoldgit, 2019-08-08:
I've added a link to your project on our wiki page: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Related-User-Projects

crankyoldgit, 2019-08-20:
> It would be great if you can send a PR for this :D

@mariusmotea A friendly reminder to answer this [Q].

Or is there nothing really left to do?

mariusmotea, 2019-08-20:
I made few changes, i will try to integrate also the receiver so the device will update the ac unit status if the original remote is used by caching the IR signal. Once is done i will create PR.

crankyoldgit, 2019-08-20:
@mariusmotea Just FYI, you do know that the [IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) example code does this sort of thing already, just not with as fancy/pretty a web interface as your project has, right? :)

mariusmotea, 2019-08-20:
I did not look into it so far, hope it will be a useful example.

crankyoldgit, 2019-08-23:
> Once is done i will create PR.

I'll close this till we get the PR, and then work from that.

Thanks in advance! :)

closed 2019-08-23 by crankyoldgit