crankyoldgit/IRremoteESP8266 #853: Use the library as a command-line tool

FabrizioRomanoGenovese, 2019-08-07:
I hope this question is not off-topic. 
At the moment I am controlling my IR remotes using 
[broadlink-mqtt](https://github.com/eschava/broadlink-mqtt) and node-red. The setup is simple: Through the node-red dashboard, I click IR keys. This is translated to a MQTT command to `broadlink-mqtt` which takes the IR code store in the relevant file and sends it. 
Example: Clicking on "SoundBar Power On" sends a MQTT message with topic `remotes/soundbar/on` and payload `auto`. `broadlink-mqtt` just reads whats in the file `..../soundbar/on` and transmits the IR code.

This approach works super well for simple remotes, where each key has is own code. For AC it doesn't really work well though, since AC remotes send the whole state to the AC when the key is pressed. 

I was thinking about the following workaround:
- When a key is pressed (e.g. ON), an external script is invoked to generate the IR code corresponding to the changed state;
- The IR code is saved on, say, file `..../ac/tosend`
- An MQTT message with topic `remotes/ac/tosend` and payload `auto` is sent
- `broadlink-mqtt` just transmits the IR code.

So as you can see I'm trying to hack things round by rewriting `..../ac/tosend` each time with the new state when a key is pressed.

The only problem with this is that I need a script to produce the state from a list of parameters (e.g. fan speed, temperature, mode, ...). This requires quite a lot of reverse engineering, and it is my impression that you have already brilliantly worked out this part. In particular, the file
[https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Gree.h]() seems to contain 
what I need.

The question, hence, is this: Is it possible to rework the code in this library as a simple tool to be called from command line? Ideally, I'd like to have something to which I can pass a bunch of parameters, and get as output the raw data I need to write in my `..../ac/tosend` file.

All in all, what I am asking for is a way to run this library on my computer (more precisely a raspberry pi 3) instead of on an arduino. I'd do it myself but I don't know where to start, so if someone could point me to some resources I'd gladly explore things more. 

Finally, if the "creating the state from a list of parameters" thing has been already worked out by some other project, I'd be glad if you could point me to it (I didn't find anything on the internet or in the issues!)

Sorry for the long question, and kudos for the great work!

crankyoldgit, 2019-08-08:
> Is it possible to rework the code in this library as a simple tool to be called from command line?

Yes. All you really need to do is define or use the flag `-DUNIT_TEST` when you compile and it will remove the Arduino specific code. You should be able to use the library to generate codes etc and most of the things you probably want to do, except send an actual signal etc.

For examples how to do that. Check out the [tools](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/) directory and [Makefile](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/Makefile) in there, or similarly the [test](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/test) directory and [Makefile](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/test/Makefile) and various `*_test.cpp` files in there. Practically every part of the library has examples of how to use it in test programs.

Finally, some of the example code ([IRMQTTServer](https://github.com/crankyoldgit/IRremoteESP8266/tree/master/examples/IRMQTTServer) and other A/C related example code) will also give you details of how to do some of what you want to do.

> I'd do it myself but I don't know where to start (I don't know c++)

Sadly, to use this library the way you want, you are really going to have to learn C or C++. There really isn't a way to avoid this.
i.e. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me

There are plenty of arduino/C/C++ tutorials out there. It isn't that hard a language to learn.

crankyoldgit, 2019-08-22:
Closing after two weeks of no activity.

closed 2019-08-22 by crankyoldgit