crankyoldgit/IRremoteESP8266 #859: not working with Interrupts

Saivamsi075, 2019-08-15:
When I using IR  remote as intrrupt in arduino it is working but if I do same with ESP8266 interrupt is not running

This is arduino code
```

#include <IRremote.h>

int RECV_PIN = 3;
void Interrupt();

IRrecv irrecv(RECV_PIN);

decode_results results;

void setup()
{
  Serial.begin(115200);
  // In case the interrupt driver crashes on setup, give a clue
  // to the user what's going on.
  pinMode(3,INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(3), Interrupt,RISING);
  interrupts();
  Serial.println("Enabling IRin");
  irrecv.enableIRIn(); // Start the receiver
  Serial.println("Enabled IRin");
}

void loop() {
  Serial.println(digitalRead(3));
  delay(1000);
}

void Interrupt()
{
  if (irrecv.decode(&results)) {
    Serial.println(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
}
```
I have found that when we call enableIRIn() function, the interrupt pin(output of TSOP1838) is changed and no more working as interrupt.

In the below code I have commented  `irrecv.enableIRIn();`  from `void setup()` and inserted the function in `void ICACHE_RAM_ATTR interrupt()` function.

If I run below code, the interrupt is called at beginning  and it gives output only the string Interrupt from `Serial.print("Interrupt")` in Interrupt() function

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

int RECV_PIN = D5;

IRrecv irrecv(RECV_PIN);

decode_results results;

int a;

void ICACHE_RAM_ATTR Interrupt()
{
  irrecv.enableIRIn();
  Serial.println("Interrupt");
  if (irrecv.decode(&results)) {
    serialPrintUint64(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
}

void setup()
{
  Serial.begin(115200);
  attachInterrupt(D5, Interrupt, RISING);
  Serial.println("Enabling IRin");
  //irrecv.enableIRIn(); // commented***********
  Serial.println("Enabled IRin");
}

void loop() {
  Serial.println(digitalRead(D5));
  delay(1000);
}
```

I have resolved this issue with the below code, but my question is that why the arduino code with `enableIRIn()` is running in arduino but not in ESP8266. I'll be waiting for the reply. 

```
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>
#define intr D4

int RECV_PIN = D5;

IRrecv irrecv(RECV_PIN);

decode_results results;

int a;

void ICACHE_RAM_ATTR Interrupt()
{
  Serial.println(intr);
  if (irrecv.decode(&results)) {
    serialPrintUint64(results.value, HEX);
    irrecv.resume(); // Receive the next value
  }
  detachInterrupt(intr);
  a=1;
}

void setup()
{
  Serial.begin(115200);
  attachInterrupt(intr, Interrupt,RISING);
  Serial.println("Enabling IRin");
  irrecv.enableIRIn(); // Start the receiver
  Serial.println("Enabled IRin");
}

void loop() {
  if(a==1){
  attachInterrupt(intr, Interrupt,RISING);
  a=0;
  }
  Serial.println(digitalRead(D5));
  delay(1000);
}
```

crankyoldgit, 2019-08-16:
While the api interface of [Arduino-IRremote](https://github.com/z3t0/Arduino-IRremote) and this library are similar and often call-for-call compatible, how they work under the hood is quite different.

IRremoteESP8266 uses an interrupt on the receiving pin to detect `CHANGE`s in the signal from the IR demodulator, where as Arduino-IRremote uses a timer interrupt every 50ms to read the value of the receiving pin. (This is a very simplified story of what is going on. It is more complicated than that.)

Thus you can't use an interrupt (like your code is doing) on the `RECV_PIN`. It will interfere with `irrecv()`s use of it.

This library is specifically tuned to the resources, options, & limitations of the ESP family of chips. Hence things "under the hood" vary wildly.

In short, you don't need to mess around with interrupts (see the code examples) with this library. If you remove the interrupt stuff from your supplied code and use `enableIRin()` etc then you'll find it should just work.

As far as I understand it you don't even need to do the interrupt stuff in your first code snippet for the "Arduino-IRremote" library but it's been 2.5 years since I last used it.

In this library. Once `irrecv()` has been properly set up & enabled, all you need to do is check the `.decode()` method to see if there is a new message for you or not, and depending how you configured the `irrecv()` class, you may need to tell it to `.resume()` afterwards. Everything else is done for you in the most efficient manor practical.

**Bonus ESP coding tip(s):**
> `void ICACHE_RAM_ATTR Interrupt()`

1) You're doing WAY to much computation in your interrupt routine. You need to be in and out of the interrupt handler as fast as practical. e.g. set a flag, and then do processing for that flag _outside_ of the interrupt handler. e.g. NEVER do serial output. It's too long and slow. You'll cause problems.
2) All the code called from an interrupt handler needs to be in ICACHE_RAM to reliably work. That means any external functions you call need to be prefaced with that attribute too. Note: ICACHE_RAM is a very scarce resource on the ESP chip line. Adding that attribute forces that code to stay loaded in the cpu's cache. That cache is a small finite amount on the ESP chips. `irrecv.decode()` is huge and slow. Not something to put in an interrupt handler.

Saivamsi075, 2019-08-17:
Thanks for the reply.
Could you please send me the code which is best and efficiently work as IR interrupt for nodemcu if possible.

crankyoldgit, 2019-08-17:
It already does. `irrecv` uses interrupts. Any of the example code that receives will help you.

Saivamsi075, 2019-08-17:
But I must need to check for that command every time which will be delayed if another function is already running.

crankyoldgit, 2019-08-17:
There is no way to set an interrupt for when a message has been fully received. The best you can do, which this library does, is set a timeout interrupt when a message is no longer being detected. See the code in IRrecv.cpp

Saivamsi075, 2019-08-17:
Thank you for the fast reply and for providing us these great libraries which solves the problems with IR receiver when interfaced with ESP8266.

closed 2019-08-17 by Saivamsi075