crankyoldgit/IRremoteESP8266 #863: AC remote store and repeat

venkateshdhara, 2019-08-17:
is it possible to implement ac IR remote store and repeat with IRremote.h library is there any limitation of SRAM in Arduino I couldn't see anyone implemented with Arduino may I know the technical reason behind it?

I have used IR record sketch in Arduino that was working with TV, projectors and all other appliances except Airconditioner, I was thinking there is no enough SRAM is that the reason it couldn't decode ac remote IR?

crankyoldgit, 2019-08-17:
I'm not 100% sure I know what you are asking, but if it is _"can you capture a raw IR message and play it back on an ESP8266/32?"_ Then the answer is yes..
See the example code:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/DumbIRRepeater/DumbIRRepeater.ino
&
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/SmartIRRepeater/SmartIRRepeater.ino

They can help you on your way with what you want to do.

> I have used IR record sketch in Arduino that was working with TV, projectors and all other appliances except Airconditioner, I was thinking there is no enough SRAM is that the reason it couldn't decode ac remote IR?

The https://github.com/z3t0/Arduino-IRremote library is very limited compared to this library. You may find your aircon is already supported.

Try https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRrecvDumpV2/IRrecvDumpV2.ino and it should either decode your a/c if it knows it, or at the very least give you a _raw_ array you can use to replay it. The ESPs has far more space than the tiny typical arduino counterparts. They can easily handle the A/C IR message lengths.

A list of supported protocols and devices can be found at: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/SupportedProtocols.md

crankyoldgit, 2019-08-21:
@venkateshdhara did this answer your question. Can this issue be closed now?

venkateshdhara, 2019-08-22:
@crankyoldgit  

Hi david examples of your library works good 
i would like to know only one thing can i use raw array data decoded by this library and retransmit by 
https://github.com/z3t0/Arduino-IRremote this library using sendraw

crankyoldgit, 2019-08-22:
Yes, the `uint16_t raw[] = {...};` line produced by IRrecvDumpV2 is completely compatible with https://github.com/z3t0/Arduino-IRremote and vice-versa

Ditto for any protocol we decode that Arduino-IRremote has too. e.g. sendNEC/sendSony/sendRC5 etc.

closed 2019-08-22 by crankyoldgit