crankyoldgit/IRremoteESP8266 #864: IRremote decode mismatch between Nano and ESP8266

venkateshdhara, 2019-08-20:
Thank you for doing such a diligent job at maintaining the fork of the IRremote for the ESPP8266 platform.

The original base library is unfortunately falling behind. I am running the IR library from here (https://github.com/z3t0/Arduino-IRremote) on the Nano and your library on ESPP8266. When I do a side by side comparisons, I have many normal remote cases with even a Phillips NEC format remote where the original library has decode errors once every 10 decode cycles and succeeds the other 8 times

[image]

I am working on a project where I am using the Nano and needed to overcome these kinds of error cases. For this, I wanted to port your ESPP8266 library to the Nano. I can do all the work, just wanted to get your guidance on hardware dependencies around porting.

I am happy to contribute my changes back to you so that you can increase the hardware coverage of the library if you'd like.

Could you guide me towards this? I really appreciate your time.

crankyoldgit, 2019-08-20:
FYI, the "Raw" data you have there is very very VERY wrong/corrupted. I'm not sure how you are calculating/getting it, but oh boy is it incorrect.

Leaving that aside, if you want to port IRremoteESP8266 back to the typical AVR/ATMEL/Nano boards etc., you are in for an uphill battle on the "receiving" end.

Both libraries differ when it comes to how it collects the timing information for each "pulse"/"mark" signal from the IR demodulator. This library records the time interval since the last "CHANGE" (via "Change" interrupt) on the GPIO. Where as the "Arduino-IRremote" library uses a repeating timer interrupt (every 50 usecs) to check the value of the GPIO from the demodulator. e.g. is it still the same value, increment the amount of time since the last change. If it has changed, save the time, and move on to the next value. etc.

They are two approaches to do the same thing. The ESP has interrupts available on every GPIO, where as the AVR/ATMEL/etc don't. Hence they have to use repeating timer/polling approach because they only have interrupts on some GPIOs.

The ESP has more memory, and is faster. This library also utilises the extra memory (if allowed) to speed up capture too so it can miss less messages.

Both libraries store the timing data in a relatively similar common structure. e.g. `irparams_t` Once you've assembled the timing data in that structure the rest of the library should port fairly easily.

So, if you port over the base "capture" methods to the nano then the rest of the library should work, if you pare it down enough to fit in the tiny memory of a "nano".
e.g. `enableIRIn()`, `disableIRIn()`, `resume()`, `decode()`
and the code "interrupt" routines of `gpio_intr()`.

crankyoldgit, 2019-08-26:
Need any more info or help?

crankyoldgit, 2019-08-31:
Friendly reminder if you need more help or info. If I don't hear from you after the weekend I'm going to assume you are fine for now, and close this issue.

venkateshdhara, 2019-08-31:
hi david,

sorry for delayed response info provided was enough for us to understand in porting for Arduino NANO kindly close the issue.

crankyoldgit, 2019-08-31:
No worries. Glad to be of help. Thanks for letting me know it can be closed.

closed 2019-08-31 by crankyoldgit