crankyoldgit/IRremoteESP8266 #866: IRMQTTServer What happens if multiple messages arrive at the same time

sheppy99, 2019-08-22:
I've just started using IRMQTTServer with multiple GPIO Emitters to control different devices, what happens if several messages arrive at the same time for different GPIO's? Are they queued up and sent or are some lost?
For example to use my 2 currently configured devices:
ir_server/send_1 7,E0E040BF,32
ir_server/send_0 53,11DA270001CDC3F0280C8004B016240000A9C3A111DA270000082500A0000006600000C19060F6
I've built hardware for 5 channels
It works a charm with my Samsung TV so far albeit I've only tested the Power_ON Command in addition to the Daikin which is on a different GPIO, but so far not at the same time

crankyoldgit, 2019-08-22:
As far as I understand it, they should be "queued", however I'm not sure if order within the queue will be maintained. Essentially, all of this is handled by the PubSubClient library.

**Note**: I mean "queued" in that each topic (with a QoS of `0`) has a single slot. If the ESP doesn't fetch the message before another message is sent to the same topic, it will overwrite it (unless the QoS is set to `1` or more.)
e.g.
`ir_server/send_1 7,E0E040BF,32` will fill the `ir_server/send_1` slot with `7,E0E040BF,32`

If you send `ir_server/send_2 7,E0E040BF,32` at the same time, it will fill the `ir_server/send_2` slot.
As there are two slots (because there are two topics), both of those arriving at the same time shouldn't be a problem. However, no guarantee on the order of which one is acted on first. (As far as I understand it)

**BUT**
if you send `ir_server/send_1 7,E0E040BF,32` and then `ir_server/send_1 7,DEADBEEF,32` at the same time (i.e. before the esp can fetch it), as there is only one "slot" (`ir_server/send_1`), one of those could get lost. (again, depending on the QoS setting)

Aside from that, `IRMQTTServer` will ensure that it is only ever sending one IR message at any moment, regardless of which topic it is from. i.e. It will only ever send one IR message at a time to make sure it doesn't mangle two messages together.

I hope that answers your question.

sheppy99, 2019-08-23:
Thanks very much for the information. As the default setting for IRMQTTServer is QoS 1 that should take care of most of it, and for my scenario it doesn't matter in which order the commands are sent. Unfortunately one of the commands I use is "Mute" sent to my Denon Amp so I'll have to see how that goes. Basically losing a mute command means its ON when it should be OFF and OFF when it should be ON. I'll update this thread as I progress in case it helps anyone else following ON. Its working faultlessly so far sending to the Daikin on one channel and Power ON to the TV on another.
Now I'm adding a third to control the Amp.
I'm very grateful that this is so much easier to work with than LIRC. Learning commands on protocols it knows about is very easy, although it doesn't seem to understand my Mitsubishi Electric Ducted AC

crankyoldgit, 2019-08-23:
I don't think you fully get how QoS works in MQTT. See https://www.hivemq.com/blog/mqtt-essentials-part-6-mqtt-quality-of-service-levels/

You also need to be concerned about sending QoS of the message from your home automation system to the MQTT broker.

Re: Mute.

As all IR message transmission is Optimistic (i.e un-acknowledged) I find that any messages that perform a toggle action are quite prone to failure/unintended states. e.g. As you've just described.
For my previous Amp (Sherwood), I just blasted volume down for ages to "mute" rather than use the mute function, then ramp'd up the volume later. e.g. after changing input source etc. Of course, that could go wrong too. i.e. someone walked in front of the amp at the wrong moment etc.

FYI, if ordering and timing are important, there is a sequencing protocol/message format for IRMQTTServer too. Look in the source/comments etc.

My current amp (yamaha) has the ability to save preset environments (i.e. input source, volume level, etc) and I now use those rather than try to "construct" them on the fly. It's much more reliable for me.

I found the "undocumented" IR commands via the [globalcache control tower](https://irdb.globalcache.com/) db entries for yamaha receivers. I suggest you try it for denon ones to see what might be available.
It' provides globalcache & pronto codes, which you can convert using the progs in the `tools/` directory or sending it and using IRrecvDumpV2 to see what the message was etc.

sheppy99, 2019-08-23:
Thanks for the link, I didn't realise it was "at least once," and I have too much on that broker to make changing it to QoS 2 a realistic proposition, I'll try changing to an unused input instead together with turning it off rather than waiting for the standby saver to do it.
Is it possible to send a continuously repeated command similar to holding the power button on?
Current Power ON Command is `17,221C,15`

crankyoldgit, 2019-08-23:
Depending on the protocol, yes. Just use the `repeat` function/argument.
e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L89

crankyoldgit, 2019-08-26:
This seems resolved now. Marking closed.

closed 2019-08-26 by crankyoldgit