crankyoldgit/IRremoteESP8266 #868: Problem with MQTT discovery message for Home Assistant from IRMQTTServer

alwashe, 2019-08-23:
After sending MQTT discovery message to Home Assistant, it recognizes everything correct but is sending the commands into the wrong topic.

It is sending the AC command to: ```ir_daikin/acac/cmnd/mode```
but it should be: ```ir_daikin/ac/cmnd/mode```

This is the discovery message the IRMQTTServer sends:
```{"~":"ir_daikin/ac","name":"ir_daikin_aircon...}```

When sending the discovery message manually without the first "ac" everything works:
```{"~":"ir_daikin/","name":"ir_daikin_aircon...}```

crankyoldgit, 2019-08-23:
Thanks for the report. I'll look into it.

crankyoldgit, 2019-08-23:
The fix should now be in the `master` branch. If you could download and confirm it is working, that would be fabulous! Otherwise, I'm going to assume it's working until I hear differently.

closed 2019-08-23 by crankyoldgit

alwashe, 2019-08-23:
It is working now, thank you sir.

crankyoldgit, 2019-08-23:
Excellent. Thanks for confirming!

crankyoldgit, 2019-08-28:
FYI: The changes mentioned above have been included in v2.6.5 of the library which has just been released.

PR #869 (2019-08-23): [Bug] IRMQTTServer: Remove duplicate MQTT_CLIMATE from HA discovery
Accidently added the value of MQTT_CLIMATE twice.
Removed duplication.

Fixes #868
commit 8f7ce1fe (2019-08-23): [Bug] IRMQTTServer: Remove duplicate MQTT_CLIMATE from HA discovery (#869)

Accidently added the value of MQTT_CLIMATE twice.
Removed duplication.

Fixes #868