crankyoldgit/IRremoteESP8266 #870: TECO protocol Light on/off

tomiisp, 2019-08-24:
Hi,

I have ALASKA A/C and this use TECO protocol, I have tested that and seems most basic commands works perfect except - A/C start with turned off display and I not found command to turn on/off display in lib.
I have used IRrecvDumpv2 to check and seems LIGHT button is in third (byte 2) byte:
Example cmd:
Light on: 0x250200A09
Light off: 0x250000A09
Then it is in byte 2, and b5 which seems in lib its wrong because TECO lib have for byte 2 this: 
b0-3 = 0b0000
b4-7 = Timer hours (unit, not thenth)

I tested that Timer hours too, 0-9hours is in b0-3 in range 0000-1001 (max up to 9h), b5 = light on/off, in hex byte 2 counts: 0x0-9 (light off) or 0x20-29 (light on).

Just FYI 

/Tomi

tomiisp, 2019-08-24:
There is also other thing, I have on remote 4 buttons without text, seems they do not anything with my A/C but they show icons on remote and change data:

- (big gray button, right of + button)
base state - 0x250000A09
on remote show icon AIR 1 - 0x251000A09
on remote show icon AIR 2 - 0x252000A09

- small gray button, right of - button)
base state - 0x250000A09
on remote show HUMID - 0x250100A09

- small white button on bottom, 1st from right
base state - 0x250000A09
on remote show SAVE - 0x250800A09

- small white button on bottom, 2nd from right
base state - 0x250000A09
on remote show icon of tree with some bubbles around - 0x250400A09

/Tomi

tomiisp, 2019-08-24:
That light cmd I already fixed but of course you can do that better.

ir_Teco.h:
 - add (~line 120)
   void setLight(const bool state);
   bool getLight(void);
- change:
  const uint64_t kTecoPower =      0b00000000000001000000000000000001000;
  (default light on)
- add:
 const uint64_t kTecoLight =      0b00000000000001000000000000000000000;

ir_Teco.cpp:
- add (line ~147)
bool IRTecoAc::getLight(void) { return remote_state & kTecoLight; }
void IRTecoAc::setLight(const bool on) {
  if (on)
    remote_state |= kTecoLight;
  else
    remote_state &= ~kTecoLight;
}
- change (stdAc::state_t IRTecoAc::toCommon(void)):
 result.light = false;
to
result.light = this->getLight();
- add at end (String IRTecoAc::toString(void))
result += addBoolToString(getLight(), F("Light"));

/Tomi

crankyoldgit, 2019-08-25:
Can you please provide the part/model numbers for the A/C remote _and_ the A/C's head unit so I can add it to the supported device list?

I'll work on your suggested changes shortly.

crankyoldgit, 2019-08-25:
```
  byte 1 = Cst 0x50
    b6-7 = "AIR" 0, 1, 2
  byte 2:
    b0 = Save
    b1 = "Tree with bubbles" / Filter??
    b2 = Light/LED.
    b3 = Humid
```

I've added the above. My guess is the "tree" thing is either a Fresh Air mode or a Filter (for pollen?). Any chance you've got a manual? or know what the features of the A/C are/were?

crankyoldgit, 2019-08-25:
I've tried to update what I can based on your info.

Can you please try out the code in PR #870 (https://github.com/crankyoldgit/IRremoteESP8266/tree/alaska) and let me know how it goes for you?

tomiisp, 2019-08-25:
Hi, 
remote and A/C unit itself have mark  ALASKA SAC9010QC
about code, I will test that over today, I let you know if it work

Anyway, thanks for your work :)
/Tomi

tomiisp, 2019-08-25:
Code is working, at least that Light option is ok. Rest I cannot judge, my A/C doesn't support that commands, they are only on remote (and they have missing text/description on remote) and manual also not mention that unmarked buttons and symbols on remote, I think this remote was used also for other A/C units from Alaska where was that functions supported. 

/Tomi

crankyoldgit, 2019-08-25:
Thanks for confirming.

crankyoldgit, 2019-08-26:
As you can't confirm the rest of the functionality, and as that PR is now merged into the `master` branch. I'm marking this closed.

closed 2019-08-26 by crankyoldgit

crankyoldgit, 2019-08-28:
FYI: The changes mentioned above have been included in v2.6.5 of the library which has just been released.