crankyoldgit/IRremoteESP8266 #872: IRMQTTServer takes time to send IR

pasna, 2019-08-25:
Hi @crankyoldgit 
I have IRMQTTServer version 1.32 and 1.34. I found that 1.34 takes time to respond to http webUI somewhat 20 seconds, while version 1.32 just within a blink.

crankyoldgit, 2019-08-25:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#Why_is_IRMQTTServer_very_slow_It_takes_a_while_to_respond_Whats_wrong

pasna, 2019-08-25:
@crankyoldgit 
is this setting correct?
[image]

crankyoldgit, 2019-08-25:
I can't tell as the settings are totally dependent on your environment. Hence check the "info" page or enable serial debugging to see if it is connecting okay

pasna, 2019-08-25:
Thanks, I got them work now. Initially I change Hostname but left MQTT Prefix blank. Now I fill in MQTT Prefix and they all work perfectly.

closed 2019-08-25 by crankyoldgit