crankyoldgit/IRremoteESP8266 #873: I have a remote control DAIKIN ARC480A5 that is not recognized.

1043717432, 2019-08-26:
Use the code intercepted by DUMPV2 as follows:
off
'''Timestamp : 000496.049
                      Library   : v2.6.4

Encoding  : UNKNOWN
                   Code      : 2B9504D3 (160 bits)
                                                  Raw Timing[319]:
                                                                     +   450, -   420,    +   448, -   446,    +   422, -   444,    +   422, -   446,
                                                                                                                                                         +   422, -   446,    +   422, - 25182,    +  3492, -  1718,    +   450, -  1288,
                              +   448, -   422,    +   446, -   448,    +   420, -   446,    +   422, -  1290,
                                                                                                                  +   448, -   422,    +   446, -   446,    +   422, -   446,    +   424, -   420,
                                                                                                                                                                                                      +   448, -  1290,    +   448, -   446,    +   422, -  1288,    +   448, -  1288,
                                                                           +   450, -   420,    +   448, -  1288,    +   448, -  1288,    +   450, -  1288,
                                                                                                                                                               +   448, -  1288,    +   448, -  1290,    +   448, -   446,    +   422, -   446,
                                    +   422, -  1288,    +   450, -   446,    +   422, -   420,    +   446, -   446,
                                                                                                                        +   422, -   422,    +   446, -   446,    +   422, -   420,    +   448, -   422,
                                                                                                                                                                                                            +   446, -   446,    +   422, -   446,    +   422, -   446,    +   422, -   420,
                                                                                 +   446, -   446,    +   422, -   446,    +   422, -   422,    +   446, -   446,
                                                                                                                                                                     +   422, -   422,    +   446, -   446,    +   422, -   446,    +   422, -   446,
                                          +   422, -   446,    +   422, -   446,    +   424, -   444,    +   424, -   446,
                                                                                                                              +   420, -   446,    +   422, -   446,    +   422, -   424,    +   444, -   444,
                                                                                                                                                                                                                 +   422, -   424,    +   444, -  1288,    +   450, -   444,    +   422, -  1288,
                                                                                      +   450, -  1288,    +   450, -   444,    +   422, -   422,    +   446, -   446,
                                                                                                                                                                          +   422, -   446,    +   422, -   446,    +   422, -   446,    +   422, -   422,
                                               +   446, -   420,    +   448, -   444,    +   422, -   446,    +   422, -   446,
                                                                                                                                   +   422, -   420,    +   448, -   446,    +   422, -   446,    +   422, -   446,
        +   422, -   422,    +   446, -  1286,    +   450, -   422,    +   448, -   446,
                                                                                            +   422, -   446,    +   422, -   422,    +   446, -   420,    +   446, -   422,
                                                                                                                                                                                +   446, -   446,    +   422, -   422,    +   446, -   446,    +   422, -   422,
                                                     +   446, -   446,    +   424, -   444,    +   422, -   420,    +   448, -   446,
                                                                                                                                         +   422, -   420,    +   446, -   446,    +   422, -   446,    +   422, -   420,
              +   448, -   444,    +   422, -   422,    +   448, -   444,    +   424, -   420,
                                                                                                  +   446, -   446,    +   422, -   446,    +   422, -   422,    +   446, -   444,
                                                                                                                                                                                      +   422, -   446,    +   422, -   444,    +   422, -   446,    +   422, -   420,
                                                           +   448, -   446,    +   422, -   420,    +   448, -   446,    +   422, -   446,
                                                                                                                                               +   422, -   446,    +   422, -   446,    +   422, -   446,    +   422, -   444,
                    +   422, -  1288,    +   450, -   420,    +   448, -   446,    +   420, -   446,
                                                                                                        +   422, -   446,    +   422, -   446,    +   424, -   420,    +   448, -   444,
                                                                                                                                                                                            +   422, -   422,    +   446, -   446,    +   424, -   420,    +   448, -  1312,
                                                                 +   424, -   420,    +   448, -  1288,    +   448, -   446,    +   422, -   446,
                                                                                                                                                     +   424, -   420,    +   446, -  1288,    +   450, -  1288,    +   450, -   444,
                          +   422, -   446,    +   422, -   422,    +   448, -   444,    +   422, -   420,
                                                                                                              +   448, -   446,    +   422, -  1288,    +   448, -   446,    +   422, -   446,
                                                                                                                                                                                                  +   422, -   444,    +   424, -   444,    +   422, -   446,    +   422, -   446,
                                                                       +   422, -   420,    +   448, -   446,    +   422, -   420,    +   446, -  1290,
                                                                                                                                                           +   448, -  1288,    +   448, -   420,    +   446, -  1288,    +   448, -   420,
                                +   446, -  1288,    +   450, -   444,    +   424, -  1286,    +   450

uint16_t rawData[319] = {450, 420,  448, 446,  422, 444,  422, 446,  422, 446,  422, 25182,  3492, 1718,  450, 1288,  448, 422,  446, 448,  420, 446,  422, 1290,  448, 422,  446, 446,  422, 446,  424, 420,  448, 1290,  448, 446,  422, 1288,  448, 1288,  450, 420,  448, 1288,  448, 1288,  450, 1288,  448, 1288,  448, 1290,  448, 446,  422, 446,  422, 1288,  450, 446,  422, 420,  446, 446,  422, 422,  446, 446,  422, 420,  448, 422,  446, 446,  422, 446,  422, 446,  422, 420,  446, 446,  422, 446,  422, 422,  446, 446,  422, 422,  446, 446,  422, 446,  422, 446,  422, 446,  422, 446,  424, 444,  424, 446,  420, 446,  422, 446,  422, 424,  444, 444,  422, 424,  444, 1288,  450, 444,  422, 1288,  450, 1288,  450, 44 [... cut ...] 2,  480, 390,  454, 414,  478, 390,  454, 414,  452, 414,  454, 414,  454, 414,  456, 414,  478, 388,  482, 386,  454, 1282,  454, 414,  454, 1282,  454, 416,  452, 416,  454, 414,  454, 1282,  456, 1282,  482, 388,  452, 414,  454, 414,  454, 414,  454, 414,  452, 414,  478, 390,  454, 416,  478, 388,  454, 416,  452, 414,  480, 388,  454, 414,  454, 414,  452, 416,  478, 388,  454, 414,  478, 388,  454, 1282,  456, 1284,  478, 388,  454, 1282,  456, 1280,  456, 414,  454};  // UNKNOWN 1D808FCC

1043717432, 2019-08-26:
Temperature 25 on
```
Timestamp : 000729.695
                      Library   : v2.6.4

Encoding  : UNKNOWN
                   Code      : B1D5976B (160 bits)
                                                  Raw Timing[319]:
                                                                     +   454, -   414,    +   482, -   388,    +   480, -   386,    +   456, -   414,
                                                                                                                                                         +   452, -   414,    +   480, - 25150,    +  3498, -  1714,    +   456, -  1282,
                              +   454, -   414,    +   454, -   414,    +   480, -   388,    +   454, -  1280,
                                                                                                                  +   454, -   414,    +   482, -   388,    +   454, -   414,    +   454, -   416,
                                                                                                                                                                                                      +   450, -  1282,    +   456, -   414,    +   452, -  1282,    +   482, -  1256,
                                                                           +   480, -   390,    +   452, -  1284,    +   454, -  1282,    +   454, -  1282,
                                                                                                                                                               +   482, -  1254,    +   456, -  1280,    +   456, -   416,    +   450, -   414,
                                    +   456, -  1282,    +   454, -   414,    +   454, -   414,    +   454, -   414,
                                                                                                                        +   454, -   414,    +   454, -   414,    +   454, -   414,    +   454, -   416,
                                                                                                                                                                                                            +   478, -   388,    +   456, -   438,    +   428, -   416,    +   478, -   388,
                                                                                 +   454, -   414,    +   452, -   414,    +   456, -   440,    +   426, -   416,
                                                                                                                                                                     +   454, -   414,    +   452, -   416,    +   452, -   414,    +   454, -  1280,
                                          +   456, -   440,    +   454, -   386,    +   480, -   388,    +   454, -  1282,
                                                                                                                              +   454, -  1280,    +   482, -   390,    +   454, -   414,    +   454, -   414,
                                                                                                                                                                                                                 +   452, -  1282,    +   456, -   414,    +   478, -   390,    +   478, -  1256,
                                                                                      +   456, -  1282,    +   452, -   414,    +   456, -   414,    +   478, -   388,
                                                                                                                                                                          +   456, -   414,    +   452, -   416,    +   478, -   388,    +   456, -   416,
                                               +   450, -   414,    +   454, -   414,    +   454, -   414,    +   480, -   388,
                                                                                                                                   +   454, -   414,    +   454, -   416,    +   480, -   388,    +   480, -   386,
        +   456, -   414,    +   452, -  1282,    +   482, -   388,    +   454, -   412,
                                                                                            +   454, -   414,    +   452, -   414,    +   454, -   414,    +   454, -   414,
                                                                                                                                                                                +   478, -   390,    +   480, -   388,    +   480, -   388,    +   454, -   416,
                                                     +   452, -   414,    +   454, -   414,    +   454, -   416,    +   452, -   414,
                                                                                                                                         +   454, -   416,    +   452, -   414,    +   480, -   388,    +   480, -   388,
              +   454, -   414,    +   480, -   386,    +   454, -   414,    +   456, -   414,
                                                                                                  +   452, -   414,    +   454, -   412,    +   456, -   414,    +   452, -   414,
                                                                                                                                                                                      +   454, -   414,    +   478, -   388,    +   456, -   414,    +   452, -   414,
[... cut ...]   486, 384,  484, 384,  456, 414,  456, 410,  458, 410,  454, 414,  454, 414,  456, 412,  456, 410,  454, 416,  482, 1252,  458, 412,  482, 1256,  482, 386,  458, 410,  482, 386,  480, 1256,  456, 1280,  484, 386,  456, 412,  456, 410,  456, 412,  480, 388,  454, 1280,  484, 386,  454, 414,  456, 414,  478, 388,  482, 386,  456, 412,  482, 384,  456, 410,  456, 412,  454, 414,  454, 412,  484, 384,  456, 1280,  458, 412,  456, 1280,  456, 1280,  458, 1278,  486, 1252,  484};  // UNKNOWN 95F6AC4B

1043717432, 2019-08-26:
Thank you! Trouble you to join

crankyoldgit, 2019-08-26:
I've added very basic support for the protocol you supplied data for.

Please download and try the code in: https://github.com/crankyoldgit/IRremoteESP8266/tree/Daikin152 and let me know how it goes for you.

1043717432, 2019-08-27:
> I've added very basic support for the protocol you supplied data for.
> 
> Please download and try the code in: https://github.com/crankyoldgit/IRremoteESP8266/tree/Daikin152 and let me know how it goes for you.

Write to https//github.com/crankyoldgit/IRremoteESP8266/tree/Daikin152
after finishing. Arduino writes to IRMQTTServer, but the remote feedback is not visible in the MQTTFX subscription message.

1043717432, 2019-08-27:
> > Write to https//github.com/crankyoldgit/IRremoteESP8266/tree/Daikin152
> > after finishing. Arduino writes to IRMQTTServer, but the remote feedback is not visible in the MQTTFX subscription message.
> 
> I honestly have no idea what you are trying to say here. Can you please rephrase this?
> Have you tried using the `IRrecvDumpV2` program from that branch?

Https//github.com/crankyoldgit/IRremoteESP8266/tree/Daikin152
After I downloaded the above firmware, I wrote it to IRMQTTServer.bin. After connecting WIFI and MQTT, the DAIKIN ARC480A5 remote control could not be fed back during the test. The web air Conditioner Control current Settings - Protocol does not have the DAIKIN152 option.

1043717432, 2019-08-27:
> There is no full/detailed/common a/c support for this protocol yet. **You** need to do the analysis and reverse-engineering to work out which bits in the A/Cs message do what, before I can even begin to write the code for it.
> 
> You can send the codes it receives via the "Send a complex (Air Conditioner) IR message" form on the "Home" page of IRMQTTServer, and select "DAIKIN152" (bottom of the list) and add the hexidecimal code for the message.

Understand what you mean. Is to run the test code on the home page.
I would like to ask the original dumpv2 received code: 95F6AC4B (160 bit) Convert it to a hex code?
How to convert?
In addition, I am not in the air conditioner host, it may be actually operated. I just brought the remote to the office.

crankyoldgit, 2019-08-27:
> Write to https//github.com/crankyoldgit/IRremoteESP8266/tree/Daikin152
> after finishing. Arduino writes to IRMQTTServer, but the remote feedback is not visible in the MQTTFX subscription message.

I honestly have no idea what you are trying to say here. Can you please rephrase this?
Have you tried using the `IRrecvDumpV2` program from that branch?

crankyoldgit, 2019-08-27:
There is no full/detailed/common a/c support for this protocol yet. **You** need to do the analysis and reverse-engineering to work out which bits in the A/Cs message do what, before I can even begin to write the code for it.

You can send the codes it receives via the "Send a complex (Air Conditioner) IR message" form on the "Home" page of IRMQTTServer, and select "DAIKIN152" (bottom of the list) and add the hexidecimal code for the message.

1043717432, 2019-08-27:
> > I would like to ask the original dumpv2 received code: 95F6AC4B (160 bit) Convert it to a hex code?
> > How to convert?
> 
> https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code
> You can't convert/use an `UNKNOWN` hex code to produce a valid message.
> You will need to recapture the message using the `IRrecvDumpV2` program from that branch.
> e.g. It should detect it as a `DAIKIN152` message, which will be 152 bits, not 160.
Hi, I have used daikin152 dumpv2 to get various commands for remote control. Thank you!

Turn on the air conditioner Cooling mode 22°
Timestamp : 000021.332
                      Library   : v2.6.4

Encoding  : DAIKIN152
                     Code      : 11DA2700003128003F000000000000C500006F (152 bits)
                                                                                  Raw Timing[319]:
                                                                                                     +   450, -   420,    +   448, -   446,    +   422, -   420,    +   446, -   422,
                                                                                                                                                                                         +   448, -   420,    +   446, - 25180,    +  3492, -  1718,    +   450, -  1288,
                                                              +   450, -   446,    +   422, -   420,    +   448, -   422,    +   446, -  1288,
                                                                                                                                                  +   450, -   420,    +   446, -   422,    +   446, -   446,    +   422, -   422,
                       +   446, -  1288,    +   448, -   422,    +   448, -  1288,    +   448, -  1290,
                                                                                                           +   446, -   446,    +   422, -  1288,    +   448, -  1288,    +   448, -  1288,
                                                                                                                                                                                               +   448, -  1288,    +   450, -  1286,    +   450, -   422,    +   446, -   422,
                                                                    +   446, -  1288,    +   448, -   422,    +   446, -   446,    +   422, -   422,
                                                                                                                                                        +   446, -   422,    +   448, -   418,    +   446, -   448,    +   422, -   420,
                             +   448, -   422,    +   446, -   420,    +   448, -   420,    +   446, -   420,
                                                                                                                 +   448, -   422,    +   446, -   422,    +   446, -   420,    +   446, -   446,
                                                                                                                                                                                                     +   422, -   422,    +   446, -   420,    +   448, -   420,    +   446, -  1288,
                                                                          +   450, -   420,    +   446, -   422,    +   446, -   420,    +   446, -  1288,
                                                                                                                                                              +   450, -  1288,    +   448, -   446,    +   424, -   422,    +   446, -   420,
                                   +   446, -   424,    +   444, -   420,    +   448, -  1288,    +   448, -   446,
                                                                                                                       +   424, -  1286,    +   448, -   420,    +   446, -   422,    +   446, -   446,
                                                                                                                                                                                                           +   422, -   420,    +   446, -   422,    +   448, -   420,    +   446, -   446,
                                                                                +   422, -   422,    +   446, -   420,    +   446, -   422,    +   448, -  1288,
                                                                                                                                                                    +   450, -  1288,    +   448, -  1288,    +   448, -  1288,    +   450, -  1288,
                                         +   448, -  1290,    +   448, -   422,    +   446, -   446,    +   422, -   446,
                                                                                                                             +   422, -   420,    +   448, -   420,    +   446, -   446,    +   422, -   420,
                                                                                                                                                                                                                 +   448, -   446,    +   424, -   418,    +   448, -   446,    +   422, -   446,
                                                                                      +   422, -   422,    +   448, -   420,    +   448, -   420,    +   446, -   446,
[... cut ...] 50, 416,  454, 414,  454, 416,  450, 418,  452, 414,  452, 416,  450, 418,  452, 1282,  454, 416,  452, 1282,  454, 1282,  454, 414,  452, 1284,  454, 1282,  452, 1282,  454};  // DAIKIN152
                                                                                                                                                                               uint8_t state[19] = {0x11, 0xDA, 0x27, 0x00, 0x00, 0x31, 0x36, 0x00, 0xAF, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC5, 0x00, 0x00, 0xED};

crankyoldgit, 2019-08-27:
> I would like to ask the original dumpv2 received code: 95F6AC4B (160 bit) Convert it to a hex code?
> How to convert?

https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code
You can't convert/use an `UNKNOWN` code to produce a valid message.
You will need to recapture the message using the `IRrecvDumpV2` program from that branch.
e.g. It should detect it as a `DAIKIN152` message, which will be 152 bits, not 160.

1043717432, 2019-08-27:
> Excellent. It (basic decoding) appears to be working! Thanks for the confirmation.
> 
> Good luck with breaking down and analysing what bits do what.

Thank you~ you answered in time. Thank you for sharing your results! Thank you!

crankyoldgit, 2019-08-27:
Excellent. It (basic decoding) appears to be working! Thanks for the confirmation.

Good luck with breaking down and analysing what bits do what.

closed 2019-08-28 by crankyoldgit

crankyoldgit, 2019-08-28:
FYI: The changes mentioned above have been included in v2.6.5 of the library which has just been released.

1043717432, 2019-08-29:
> https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#My_remotes_signal_is_being_reported_as_an_Unknown_code_What_does_that_mean_How_do_I_send_an_Unknown_code
> You can't [...]

Help, how do I get the firmware to recognize my daikinARC480A45 A/C remote control?
After downloading version 2.6.5, after brushing into IRMQTTServer, the firmware cannot recognize the ARC480A45 A/C remote control.

crankyoldgit, 2019-08-29:
What does the "info" page say all the versions are?
i.e. Use the remote on IRMQTTServer and then cut and paste the entire info page text here.

1043717432, 2019-08-30:
> What does the "info" page say all the versions are?
> i.e. Use the remote on IRMQTTServer and then cut and paste the entire info page text here.

[image]
[image]
Only the home page has

crankyoldgit, 2019-08-30:
Repeating what I said earlier:

> There is no full/detailed/common a/c support for this protocol yet. You need to do the analysis and reverse-engineering to work out which bits in the A/Cs message do what, before I can even begin to write the code for it.

In order for the "aircon" page/interface/etc to work for a protocol, it needs "detailed" support. There is no "detailed" support for DAIKIN152 yet. If you want "detailed" support (i.e. controlling modes and temperature), then the first step is **YOU** analysing & decoding the hex values that the library now produces to work out how to control all the features you need. Then and ONLY then, can "detailed" support for that protocol. No one else will probably decode it for you. You are the only one at present with a vested interest in it.

TL;DR: There is no _detailed_ support for this protocol currently. With out it, you can't use the "aircon" features of this library.

crankyoldgit, 2019-08-30:
and FYI, that wasn't the info page.

1043717432, 2019-08-31:
Ok! Thank you!

1043717432, 2019-09-18:
> and FYI, that wasn't the info page.

Hi! Let you wait too long. I have been busy recently and have no time to deal with it.
Today, I tested the code at home one day, DAKIN air conditioner, perfectly supporting DAKIN152 transmission code, but the air conditioner panel light did not light up. Looking forward to your firmware!

1043717432, 2019-09-18:
Using the receiving code grabbed by dumpv2 above, it is successful to control the air conditioner with DAKIN152. The wind speed and temperature control are normal.

crankyoldgit, 2019-09-18:
Thanks for letting us know it is working.

FYI, here are some tips to help **you** work out what all the bits in the message do. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

1043717432, 2019-09-18:
Can you send me a form for reference?

crankyoldgit, 2019-09-18:
> Can you send me a form for reference?

I am not sure what you are asking.

1043717432, 2019-09-18:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

> > Can you send me a form for reference?
> 
> I am not sure what you are asking.

https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol 
I have checked this address,
Can send a sample excel form

crankyoldgit, 2019-09-18:
> https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol
> I have checked this address,
> Can send a sample excel form

Check again. I included some links to example a little while ago.
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#example-spreadsheets

1043717432, 2019-09-20:
> Check again. I included some links to example a little while ago.
> https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#example-spreadsheets

Excuse me, input remote control code, is all refrigeration mode, temperature 30-16, including low air volume-maximum air volume, each line of code has to be obtained?

The workload is heavy. There's a lot of code.

crankyoldgit, 2019-09-20:
You only need to collect enough information to be sure you understand the particular control/setting you are focusing on. e.g. To a level where you can predict what each byte should be for a specific setting.

FYI, for `DAIKIN152` I've already worked out the Integrity Check/Checksum for you. This is because it is similar to other Daikin protocols. You can be pretty sure (but not 100%) that we have the correct bit ordering too based on that.

> The workload is heavy. There's a lot of code.

Yes. It's a lot of work. It is not uncommon to collect 30-50 messages to decode a protocol.

1043717432, 2019-09-20:
> You only need to collect enough information to be sure you understand the particular control/setting you are focusing on. e.g. To a level where you can predict what each byte should be for a [...]

Well. Thank you!! It's been collected!
[DAIKIN-ARC480A5.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3634534/DAIKIN-ARC480A5.txt)

1043717432, 2019-09-20:
Thank you for your hard work. Where is missing, please tell me!

crankyoldgit, 2019-09-20:
> Thank you for your hard work. Where is missing, please tell me!

Your analysis of what bits and bytes correspond to each setting is missing.

As far as I can tell, you are past the stage of
[Raw Data should no longer be needed now](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#raw-data-should-no-longer-be-needed-now)
& and are now entering the [Analysing the data](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data) phase.

There is nothing in that txt file that I can see describing how to break down the `state[]` data into individual controls, and what their specific values do.

1043717432, 2019-09-20:
Oh, My God!! I can't. How to operate it?

crankyoldgit, 2019-09-20:
> How to operate it?

The rest of that document. i.e. https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#analysing-the-data gives you tips on how to do it.

A fair number of people have worked out their own AirCon codes work. You can do it. You've had most of the hard parts already done for you.

1043717432, 2019-09-20:
I'm far from home and can't operate the remote control in front of the air conditioner. Only take the remote control to the office to learn code, can you fulfill your requirements?

crankyoldgit, 2019-09-20:
Once you have collected all the data, which it seems you have, you don't need to be at home or have the remote handy. It is now a data analysis problem.

You now seem to have enough data to start the analysis/reverse engineering at the very least.
You now have to use your brain to work out how it all works. Once you have done that, I'll typically help you code it up.

1043717432, 2019-09-21:
> Once you have collected all the data, which it seems you have, you don't need to be at home or have the remote handy. It is now a data analysis problem.
>
> You now seem to have enough data to start [...]

Could you tell me in detail how to start analysis/reverse engineering, what tools to use and how to start? If there were photo textbooks, I think I would better accomplish the task, which is too difficult for me. Thank you!

crankyoldgit, 2019-09-21:
Add those documents suggest, and the example spreadsheets. Put each part of a state[] in it's own cell in a spreadsheet. And stack two similar state codes one under the other, and look for differences.

It's not that hard.

I don't know of any tools but @AnalysIR might sell one. https://www.analysir.com/blog/get-analysir/
is their software. I've never used it, so I can't recommend it. I've only heard of it.

1043717432, 2019-10-04:
Sorry, I don't know where to start!

I don't know how to make analysis and comparison.

crankyoldgit, 2019-10-05:
I've tried adding a worked example to the document.
i.e. How to go about starting a reverse engineering of a trivial code. i.e. How to approach the analysis.

https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#example-analysis

If you can't follow that, or working out how to do it from that, then I don't know if I can help you any further.

1043717432, 2019-10-10:
> I've tried adding a worked example to the document.
> i.e. How to go about starting a reverse engineering of a trivial code. i.e. How to approach the analysis. [...]

Thank you. I try to study hard. I have no experience in this area. But I will try hard. Thank!

crankyoldgit, 2019-10-30:
FYI, the changes mention above have been included in the v2.7.0 release of the library.

commit f99b1d66 (2019-08-28): Basic support for Daikin152 A/C protocol. (#874)

* Basic support (send & capture) for Daikin152 protcol (ARC480A5 remote)
* Unit tests to cover decoding a real message & self-decoding.

Fixes #873