crankyoldgit/IRremoteESP8266 #876: IRremoteESP8266 MQTT and ArduinoJson 6

oxedgar, 2019-08-28:
Hello,

Can you upgrade your code for IRMQTTserver to work with ArduinoJson6 pls?
I use 2.6.5 library

Thank you

crankyoldgit, 2019-08-29:
I'll look at what it will take to change.
Why does it need to be upgraded? What does v6 offer that v5 doesnt?

oxedgar, 2019-08-29:
ArduinoJson V6 is more easy to use and the library is rewrited for better memory allocation. Some bugs are removed on this version and is the latest. In my case, i cannot compile all your exemples where V5 is used because i have maked other sketchs with V6 version! Wath do you think about this?

crankyoldgit, 2019-08-30:
I think I've converted it to use ArduinoJson V6. i.e. it compiles and still seems to work.

Please try out the code in the following branch and let me know if it works okay: https://github.com/crankyoldgit/IRremoteESP8266/tree/ArduinoJson6

> In my case, i cannot compile all your exemples where V5 is used because i have maked other sketchs with V6 version! Wath do you think about this?

Honestly, that sounds like a weak reason to move. If you used PlatformIO rather than the ArduinoIDE you could handle different environments and libraries easier.

I'm still undecided if I should move away from v5 as v6 costs about 1.3+ kbytes more space.

TL;DR: Try it out, but it may not get merged in

crankyoldgit, 2019-08-31:
@oxedgar It will be merged, just waiting on your feedback before I do so.

closed 2019-09-05 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

oxedgar, 2019-09-23:
good thankyou for your good job

PR #878 (2019-09-05): IRMQTTServer: Move from ArduinoJson v5 to v6
Note: This costs about ~1.3k of prog space.

Fixes #876

@sillyfrog Your thoughts wanted in addition to a code review:
I'm of two minds about this, moving to v6 means more modern code (v5 is still maintained/patched etc) and it's what users are likely to install by default, but it comes with a cost of increased space.
There is no "need" to make these changes.

The json version won't affect EasyESP but asking @jimmys01 for his thoughts too in cause they have had this similar dilemma too
commit a5b1624b (2019-09-05): IRMQTTServer: Move from ArduinoJson v5 to v6 (#878)

* IRMQTTServer: Move from ArduinoJson v5 to v6
Note: This costs about ~1.3k of prog space.
* Update travis build config to use ArduinoJson v6.

Fixes #876