crankyoldgit/IRremoteESP8266 #879: Problem with MQTT discovery message for Home Assistant from IRMQTTServer, it is not retained

alwashe, 2019-08-30:
After restarting Homeassistant there is no AC Device anymore.
I have to hit the Button for Send a Climate MQTT discovery message to Home Assistant again.
It would solve this issue if the MQTT discovery message is retained.
Other devices (for example zigbee2mqtt) send retained discovery messages.

crankyoldgit, 2019-08-31:
@alwashe Can you please download and try the code in the branch `Issue879` and let me know if it works as you want? https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue879

alwashe, 2019-08-31:
yes, it works. thank you.
Is it intended that you have to hit discovery? would it not be better if the deiscovery message would be send on a reboot or first start?

crankyoldgit, 2019-08-31:
Thanks for the confirmation.

I originally had it broadcasting it every X minutes, and that caused it to play up on my HA setup, so I decided to just make it an optional/manual step. This means that if people don't want to use discovery, the don't have to recompile the code etc.

So, yes, it is intended to be only hit once .. kind-of. :)

closed 2019-08-31 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

PR #881 (2019-08-31): IRMQTTServer: Use retain for discovery message.
* Use MQTT retain for the climate discovery message.
* Better code disabling when MQTT discovery message is not required.

Fixes #879
commit 932710b8 (2019-08-31): IRMQTTServer: Use retain for discovery message. (#881)

* Use MQTT retain for the climate discovery message.
* Better code disabling when MQTT discovery message is not required.

Fixes #879