crankyoldgit/IRremoteESP8266 #882: TECO Timer on/off

tomiisp, 2019-09-02:
Hi,

just another question, have you on plan to implement Timer On and Timer Off functions ? I see there bit mapping but functions in code seems missing.
If its needed I can scan and provide real data for this commands.

Best regards
/Tomi

crankyoldgit, 2019-09-02:
I can add them if you proved the data

tomiisp, 2019-09-02:
Hi,

here is the data, seems it use same bits for hour counter but difference is in state of A/C, Timer Off can be activated only if A/C is already running, Timer On can be used only if A/C is off.

Base state - A/C off (cool,26C,autofan,swing on,light on)

- Timer On (A/C must be Off to accept this commands)

	base - 0x250200A41
	0.5h - 0x250209A41
	1.0h - 0x250218A41
	1.5h - 0x250219A41
	2.0h - 0x250228A41
	2.5h - 0x250229A41
	3.0h - 0x250238A41
	3.5h - 0x250239A41
	4.0h - 0x250248A41
	4.5h - 0x250249A41
	5.0h - 0x250258A41
	5.5h - 0x250259A41
	6.0h - 0x250268A41
	6.5h - 0x250269A41
	7.0h - 0x250278A41
	7.5h - 0x250279A41
	8.0h - 0x250288A41
	8.5h - 0x250289A41
	9.0h - 0x250298A41
	9.5h - 0x250299A41
	10.0h - 0x25020AA41
	10.5h - 0x25020BA41
	11.0h - 0x25021AA41
	11.5h - 0x25021BA41
	12.0h - 0x25022AA41
	12.5h - 0x25022BA41
	13.0h - 0x25023AA41
	13.5h - 0x25023BA41
	14.0h - 0x25024AA41
	14.5h - 0x25024BA41
	15.0h - 0x25025AA41
	15.5h - 0x25025BA41
	16.0h - 0x25026AA41
	16.5h - 0x25026BA41
	17.0h - 0x25027AA41
	17.5h - 0x25027BA41
	18.0h - 0x25028AA41
	18.5h - 0x25028BA41
	19.0h - 0x25029AA41
	19.5h - 0x25029BA41
	20.0h - 0x25020CA41
	20.5h - 0x25020DA41
	21.0h - 0x25021CA41
	21.5h - 0x25021DA41
	22.0h - 0x25022CA41
	22.5h - 0x25022DA41
	23.0h - 0x25023CA41
	23.5h - 0x25023DA41
	24.0h - 0x25024CA41
	back to - 0x250200A41

-----------------------------------------------------------------
Base state - A/C on (cool,26C,autofan,swing on,light on)
- Timer Off (A/C must be On to accept this commands)

	base - 0x250200A49
	0.5h - 0x250209A49
	1.0h - 0x250218A49
	1.5h - 0x250219A49
	2.0h - 0x250228A49
	....
	12.0h - 0x25022AA49
	....
	20.0h - 0x25020CA49
	....
	24.0h - 0x25024CA49
	back to - 0x250200A49	

/Tomi

crankyoldgit, 2019-09-02:
Can you elaborate more on what it means by timer on and off.
are there two timers? Can both be set at the same time.
Are the times relative to the current time. e.g. if you set an on time(r) for 1.0h is that turn on in 1 hour, or turn on at 1am?

tomiisp, 2019-09-02:
Hi,

it is simple:
- if A/C is OFF you can set(use) Timer On, if you set this timer to 2.0h then A/C turn ON after 2.0h delay (with settings which are in message). 
- if A/C is ON you can set(use) Timer Off, if you set this timer to 3.0h then A/C turn OFF after 3.0h delay.

This timers start counting after you sent command with timer setting to A/C, there is no real time relation.
Only one timer can be set/used,  I think there is only one timer (one time counter), it choose if do On or Off depend on state of A/C set in message.

On remote there are two buttons, Timer OFF button do something/works only if A/C is ON, second button Timer ON works only if A/C is OFF.

A/C in OFF state, Timer ON button:

[image]

---------------------------------------------------

A/C in ON state, Timer OFF button:

[image]

/Tomi

crankyoldgit, 2019-09-02:
@tomiisp Can you please download and try out the code in PR #883 (https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue882) and let me know if it works as you expect.

tomiisp, 2019-09-02:
Timer OFF seem works, this code works fine:

```
  ac.on(); 
  ac.setTemp(26);
  ac.setFan(1);
  ac.setMode(kTecoCool);
  ac.setLight(1);
  ac.setTimer(30);
  ac.send();
```
  
  output of printState():
  TECO A/C remote is in the following state:
  Power: On, Mode: 1 (COOL), Temp: 26C, Fan: 1 (Low), Sleep: Off, Swing: Off, Light: On, Humid: Off, Save: Off, Timer: 00:30
  
  this is equivalent to Timer OFF 0.5h = A/C is ON and it turn OFF after 30minutes, this is ok

----------------------------------------------------------------
but Timer ON with this code seems not work, A/C  recognize command (it makes "beep") but after 30minutes it doesnt turn ON: 
```
  ac.off(); 
  ac.setTemp(27);
  ac.setFan(1);
  ac.setMode(kTecoCool);
  ac.setLight(1);
  ac.setTimer(30);
  ac.send();
```
but I suspect one thing, maybe commands must be in sync, for example: 
- A/C was ON and 26C/Cool and it was turned off at this state, my code send again OFF command but temperature was changed to 27C. 
I think last OFF command and new OFF command with activated timer must match in parameters (in OFF state temperature cannot be changed on remote).
Actually I testing this part but of course it takes some time because each test combination takes at least 30minutes of waiting.

/Tomi

tomiisp, 2019-09-02:
Ok, seems problem was between chair and keyboard :) I use this test sequence for Timer ON:

```
  // Turn ON A/C
  ac.on(); 
  ac.setTemp(27);
  ac.setFan(1);
  ac.setMode(kTecoCool);
  ac.setLight(1);
  ac.setTimer(0);
  ac.send();
  printState();

  delay (5000);

  // Turn OFF A/C
  ac.off();
  ac.send();
  printState();

  delay (5000);

  //Send again Turn OFF cmd with timer set
  ac.off();
  ac.setTemp(26);
  ac.setTimer(30);
  ac.send();
  printState();
```

here I turn on A/C at 27C/Cool, then simple turn it off and then send turn off command but with changed temperature and activated Turn ON timer.

It works fine, seems it works in both scenarios. Of course I do some more test but actually I think code ts in work/confirmed state.

/Tomi

tomiisp, 2019-09-02:
Just one note, its is possible to set setLight() option to 1 as default in code ? Now it is 0 and if I forget to set it in code at begin, display runs in that "night" mode.

/Tomi

crankyoldgit, 2019-09-03:
> Just one note, its is possible to set setLight() option to 1 as default in code (as part of ac.begin) ? Now it is 0 and if I forget to set it in code at begin, display runs in that "night" mode as default.

You've got several options:

1) Remember to add `setLight(true);` in your code. **_;-)_**
2) Add: `setLight(true);` to the [stateReset()](https://github.com/crankyoldgit/IRremoteESP8266/issues/882#issuecomment-527228151) code. **or** change `remote_state = kTecoReset;` to `remote_state = kTecoReset | kTecoLight;`
3) Change:
`const uint64_t kTecoReset =      0b01001010000000000000010000000000000;`
to
`const uint64_t kTecoReset =      0b01001010000001000000010000000000000;`
at
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Teco.h#L45

closed 2019-09-03 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

PR #883 (2019-09-03): Teco: Add timer support.
Fixes #882
commit fb61af1d (2019-09-03): Teco: Add timer support. (#883)

Fixes #882