crankyoldgit/IRremoteESP8266 #885: Update CI pipeline to use PlatformIO build tools

augustozanellato, 2019-09-02:
Using platformio command to build the library on TravisCI should be a lot less a mess than installing a full fledged Arduino environment every time a CI build is done

crankyoldgit, 2019-09-02:
Please submit it in a separate PR. Yes, I'm fine with moving to using platformio in Travis

augustozanellato, 2019-09-04:
Should I also move the testing infrastructure to PlatformIO unit testing?

crankyoldgit, 2019-09-04:
Not yet. But we can look at it after this.

crankyoldgit, 2019-09-10:
I saw in your repo you had made some efforts in doing this. How's it going?

augustozanellato, 2019-09-10:
Still working on that on my local copy

crankyoldgit, 2019-09-10:
No worries. I was just checking in. Thanks very much for offering to do it and for what you've tried thus far.

crankyoldgit, 2019-09-30:
How goes it? Just a reminder/incentive, your eventual PR is worth a quarter of a t-shirt if you get it in this month. See #932

augustozanellato, 2019-09-30:
I’m just full with school stuff atm, probably around the end of this week I could take an in depth look into it

crankyoldgit, 2019-09-30:
School comes first. No rush or pressure. Just trying to make sure your effort might help score you a shirt. :)

augustozanellato, 2019-10-01:
The PR is almost ready, the only problem is with the [Web-AC-control](https://travis-ci.com/augustozanellato/IRremoteESP8266/jobs/240842686#L6658) example as you can see here, but it seems to be a bug in the platformio file for that example

closed 2019-10-02 by crankyoldgit

crankyoldgit, 2019-10-02:
Ref #932

crankyoldgit, 2019-10-30:
FYI, the changes mention above have been included in the v2.7.0 release of the library.

PR #936 (2019-10-02): Port CI pipeline to PlatformIO
Fixes #885
commit 22baa618 (2019-10-02): Port CI pipeline to PlatformIO (#936)

Fixes #885