crankyoldgit/IRremoteESP8266 #887: Trying to understand a feature in IR repeater code

happytm, 2019-09-06:
### Version/revision of the library used
I have been using this library with latest espeasy firmware.
### Expected behavior
I have a question for usability of IR repeater.

### Actual behavior
I Have not tried it as I am trying to understand a  feature.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

N/A

### Steps to reproduce the behavior

N/A

crankyoldgit, 2019-09-06:
What is your question?

happytm, 2019-09-06:
Great useful software and also great clean management of this repo.I can not believe low ratio you have for open/closed issues and pull request. Also plentiful comments in code is rarely seen in other repos. It make very easy for non programmers like me.

I want to send MAC ID of my esp8266 with IR LED connected to it  to another esp8266 device with IR receiver connected to it.  I looked in to smart repeater code and it looks like it is easiest for me to modify out of all other examples. I am looking for simplest solution for very simple task of sending Mac ID periodically over IR protocol.

Is it even possible ?

Please help.

Thanks

happytm, 2019-09-06:
Sorry dinner time came in between 2 comments.

happytm, 2019-09-06:
I am going to use following code. Is this OK for sender unit.

Is this OK to use?

On receiver node I am planning to use smart repeater code. Will it work as it is on receiver side or I need to modify the code there ?

Thanks.

`#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 4;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Example of data captured by IRrecvDumpV2.ino
uint16_t rawData[67] = {9000, 4500, 650, 550, 650, 1650, 600, 550, 650, 550,
                        600, 1650, 650, 550, 600, 1650, 650, 1650, 650, 1650,
                        600, 550, 650, 1650, 650, 1650, 650, 550, 600, 1650,
                        650, 1650, 650, 550, 650, 550, 650, 1650, 650, 550,
                        650, 550, 650, 550, 600, 550, 650, 550, 650, 550,
                        650, 1650, 600, 550, 650, 1650, 650, 1650, 650, 1650,
                        650, 1650, 650, 1650, 650, 1650, 600};
// Example Samsung A/C state captured from IRrecvDumpV2.ino
uint8_t samsungState[kSamsungAcStateLength] = {0xb4, 0xe6, 0x52, 0x44, 0x86, 0xad, 0xb4, 0xe6, 0x52, 0x44, 0x86, 0xad };

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {
  //Serial.println("NEC");
  //irsend.sendNEC(0x00FFE01FUL);
  //delay(2000);
  //Serial.println("Sony");
  //irsend.sendSony(0xa90, 12, 2);  // 12 bits & 2 repeats
  //delay(2000);
  //Serial.println("a rawData capture from IRrecvDumpV2");
  //irsend.sendRaw(rawData, 67, 38);  // Send a raw data capture at 38kHz.
  //delay(2000);
  Serial.println("Sending MAC adress to Master Node");
  irsend.sendSamsungAC(samsungState);
  for(uint8_t u=0; u<6; u++)
   {
      Serial.print("   MAC ID :  ");
      Serial.print(samsungState[u]);
   }

  
  delay(2000);
}`

crankyoldgit, 2019-09-06:
Often Aircon protocols contain a checksum that gets calculated before being sent, which may modify the code you are sending, and/or they often have other integrity checks in the protocol.

A MAC address is 48 bits long, so I'd pick a protocol that is _at least_ 48 bits long.
`MIDEA` happens to be 48 bits long and has no checksum or integrity checks in the data value. So it should be safe to use.
e.g.
On your "sender" ESP, you should be able to add something like this to the `IRsendDemo.ino` code:
```
uint64_t hwaddr = 0x1234567890AB;  // 48bits
irsend.sendMidea(hwaddr);
```

And for the other "receiver" ESP module, use code something like `IRrecvDemo.ino` but with:

```
IRrecv irrecv(kRecvPin, 250);  // Increase the capture buffer to 250 entries. Should be overkill for Midea.
```
and
```
  if (irrecv.decode(&results)) {
    if (result.decode_type == MIDEA) {
      serialPrintUint64(results.value, HEX);  // Should be your 48bit MAC Address.
      Serial.println("");
    }
    irrecv.resume();  // Receive the next value
  }
```

You should be able to able to disable decoding of all other protocols except `MIDEA` in `IRremoteESP8266.h` if that's all you want to use this library for. It will make it faster and reduce code the code space required.

Let me know how you go.

happytm, 2019-09-06:
Thank you for quick solution. The code above I was using was working fine and showing correct values for mac id bytes in serial monitor. I don't know whether it will end up same on receiver side or not. I tried your new protocol and it compiles fine as it is but I want to build a mac id using some variables in my code like:

uint64_t hwaddr = {temperature, humidity, pressure, battery, lux, device};  // 48bits  but now there is compiling error like this:

scalar object 'hwaddr' requires one element in initializer

Is there any solution to fix this error ? 

I am trying to build remote sensors sending sensor data over ir to one master node. 

Thanks

crankyoldgit, 2019-09-06:
You are trying to assign an [array](http://www.cplusplus.com/doc/tutorial/arrays/) to an integer. That's what the error message is about. It's not a library specific question per se. It's fairly basic programming problem. There are many ways to solve or address that problem. Too many to discuss here.

Doing a quick google search for: ["c++ convert array to int"](https://www.google.com/search?q=c%2B%2B+convert+array+to+int)
yields
https://stackoverflow.com/questions/11201015/converting-an-integer-array-into-a-number

FYI, you're starting to stray into [this territory](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me).

happytm, 2019-09-06:
I agree with you that I am starting to stray. For now I will stay in lane and  I am going to test samsung default code for few days with receiver hardware and see if there is any data errors. 

Thanks again for help.

happytm, 2019-09-09:
I tried midea protocol exactly as you suggested earlier using Pin 16 of Wemos d1 mini to send IR and on receiving side using pin 2 of Wemos d1 mini. I get following sample in serial console on receiving Wemos:

7109F1EF
9765D0AE
AE2388BB
A4468BF0
77DF9F52
6C6C8538
8CF4E78C
28819C92
EAD32682
732B064A
2AFF441E
326CE819
63D0CD55
F09EEED2
C4EEBB09
E2809C58
E63957E0
D8E674B6
FF35492F
2E37EB5

my question is why it is not same code received for constant value sent ? does pin selection matters ?

Thanks

crankyoldgit, 2019-09-10:
> my question is why it is not same code received for constant value sent ?

Are you sure you are sending a constant value?

Can you link/paste the code you are running on both ends?

> does pin selection matters ?

It shouldn't.

happytm, 2019-09-10:
I think it was because of delay settings in sender code I changed from 200000 to 500.But following are both codes. What is the minimum delay required in sender code?

Sender:
`#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

const uint16_t kIrLed = 16;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).

IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

// Example of data captured by IRrecvDumpV2.ino
uint16_t rawData[67] = {9000, 4500, 650, 550, 650, 1650, 600, 550, 650, 550,
                        600, 1650, 650, 550, 600, 1650, 650, 1650, 650, 1650,
                        600, 550, 650, 1650, 650, 1650, 650, 550, 600, 1650,
                        650, 1650, 650, 550, 650, 550, 650, 1650, 650, 550,
                        650, 550, 650, 550, 600, 550, 650, 550, 650, 550,
                        650, 1650, 600, 550, 650, 1650, 650, 1650, 650, 1650,
                        650, 1650, 650, 1650, 650, 1650, 600};
// Example Samsung A/C state captured from IRrecvDumpV2.ino
//uint8_t samsungState[kSamsungAcStateLength] = {0xb4, 0xe6, 0x52, 0x44, 0x86, 0xad, 0xb4, 0xe6, 0x52, 0x44, 0x86, 0xad };
uint64_t hwaddr = 0x1234567890AB;  // 48bits

void setup() {
  irsend.begin();
#if ESP8266
  Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
#else  // ESP8266
  Serial.begin(115200, SERIAL_8N1);
#endif  // ESP8266
}

void loop() {
  //Serial.println("NEC");
  //irsend.sendNEC(0x00FFE01FUL);
  //delay(2000);
  //Serial.println("Sony");
  //irsend.sendSony(0xa90, 12, 2);  // 12 bits & 2 repeats
  //delay(2000);
  //Serial.println("a rawData capture from IRrecvDumpV2");
  //irsend.sendRaw(rawData, 67, 38);  // Send a raw data capture at 38kHz.
  //delay(2000);
  Serial.println("Sending MAC adress to Master Node");
  //irsend.sendSamsungAC(samsungState);
  irsend.sendMidea(hwaddr);
  for(uint8_t u=0; u<6; u++)
   {
      Serial.print("   MAC ID :  ");
//      Serial.print(samsungState[u]);
//        Serial.print(hwaddr[u]);
   }

  
  delay(500);
}`

Receiver:

`/*
 * IRremoteESP8266: IRrecvDemo - demonstrates receiving IR codes with IRrecv
 * This is very simple teaching code to show you how to use the library.
 * If you are trying to decode your Infra-Red remote(s) for later replay,
 * use the IRrecvDumpV2.ino example code instead of this.
 * An IR detector/demodulator must be connected to the input kRecvPin.
 * Copyright 2009 Ken Shirriff, http://arcfn.com
 * Example circuit diagram:
 *  https://github.com/crankyoldgit/IRremoteESP8266/wiki#ir-receiving
 * Changes:
 *   Version 0.2 June, 2017
 *     Changed GPIO pin to the same as other examples.
 *     Used our own method for printing a uint64_t.
 *     Changed the baud rate to 115200.
 *   Version 0.1 Sept, 2015
 *     Based on Ken Shirriff's IrsendDemo Version 0.1 July, 2009
 */

#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRrecv.h>
#include <IRutils.h>

// An IR detector/demodulator is connected to GPIO pin 14(D5 on a NodeMCU
// board).
const uint16_t kRecvPin = 2;

IRrecv irrecv(kRecvPin, 250);

decode_results results;

void setup() {
  Serial.begin(115200);
  irrecv.enableIRIn();  // Start the receiver
  while (!Serial)  // Wait for the serial connection to be establised.
    delay(50);
  Serial.println();
  Serial.print("IRrecvDemo is now running and waiting for IR message on Pin ");
  Serial.println(kRecvPin);
}

void loop() {
  if (irrecv.decode(&results)) {
    // print() & println() can't handle printing long longs. (uint64_t)
    serialPrintUint64(results.value, HEX);
    Serial.println("");
    irrecv.resume();  // Receive the next value
  }
  delay(100);
}`

Thanks

happytm, 2019-09-10:
With delay setting of 2000 with above code I get following at receiver's serial console:
Lot better than before.

9C6008CD
AF0245A
AF0245A
9C6008CD
9C6008CD
AF0245A
AF0245A
9C6008CD
9C6008CD
9C6008CD
AF0245A
AF0245A
AF0245A
9C6008CD
AF0245A
AF0245A
AF0245A

Thanks

crankyoldgit, 2019-09-10:
That's not the code I told you to use in the receiver side. See: https://github.com/crankyoldgit/IRremoteESP8266/issues/887#issuecomment-528745806

You need something like this in your loop:
> ```
>   if (irrecv.decode(&results)) {
>     if (result.decode_type == MIDEA) {
>       serialPrintUint64(results.value, HEX);  // Should be your 48bit MAC Address.
>       Serial.println("");
>     }
>     irrecv.resume();  // Receive the next value
>   }
> ```

e.g.
```
void loop() {
  if (irrecv.decode(&results)) {
     if (result.decode_type == MIDEA) {
       serialPrintUint64(results.value, HEX);  // Should be your 48bit MAC Address.
       Serial.println("");
    }
    irrecv.resume();  // Receive the next value
  }
  delay(100);
}
```

crankyoldgit, 2019-09-10:
Using the `IRrecvDumpV2.ino` example code on the receiver will also give you a more detailed idea of what is going on.

happytm, 2019-09-10:
This is my dumpV2 log and it still not clear to me why there are different values sent ?

`Timestamp : 000014.605
Library   : v2.6.3

Encoding  : UNKNOWN
Code      : AF0245A (100 bits)
Raw Timing[199]:
   +  4424, -  4546,    +   518, -   612,    +   520, -   612,    +   518, -   610, 
   +   494, -  1754,    +   496, -   636,    +   518, -   608,    +   492, -  1758, 
   +   544, -   588,    +   494, -   638,    +   518, -   608,    +   496, -  1756, 
   +   492, -  1754,    +   494, -   638,    +   494, -  1756,    +   492, -   636, 
   +   520, -   612,    +   494, -   638,    +   516, -  1732,    +   494, -   636, 
   +   492, -  1760,    +   514, -   612,    +   492, -  1788,    +   462, -  1758, 
   +   494, -   638,    +   490, -   666,    +   466, -  1782,    +   466, -  1784, 
   +   466, -  1758,    +   492, -  1758,    +   492, -   638,    +   516, -   612, 
   +   516, -   614,    +   494, -  1760,    +   490, -   636,    +   518, -   640, 
   +   492, -  1732,    +   516, -   638,    +   494, -   636,    +   468, -   662, 
   +   466, -   638,    +   518, -  1730,    +   522, -   612,    +   490, -  1756, 
   +   516, -   612,    +   494, -  1756,    +   518, -   610,    +   518, -  1760, 
   +   464, -  1784,    +   466, -  5704,    +  4394, -  4574,    +   492, -  1758, 
   +   466, -  1756,    +   516, -  1760,    +   464, -   664,    +   492, -  1732, 
   +   516, -  1734,    +   492, -   662,    +   466, -  1786,    +   490, -  1786, 
   +   464, -  1732,    +   490, -   664,    +   492, -   612,    +   490, -  1784, 
   +   490, -   644,    +   460, -  1758,    +   492, -  1758,    +   518, -  1760, 
   +   490, -   610,    +   496, -  1782,    +   492, -   636,    +   466, -  1784, 
   +   466, -   638,    +   516, -   638,    +   438, -  1818,    +   458, -  1788, 
   +   464, -   638,    +   492, -   638,    +   494, -   638,    +   490, -   664, 
   +   492, -  1734,    +   490, -  1782,    +   462, -  1764,    +   490, -   666, 
   +   488, -  1758,    +   466, -  1758,    +   492, -   638,    +   492, -  1784, 
   +   490, -  1760,    +   464, -  1786,    +   466, -  1786,    +   464, -   640, 
   +   490, -  1758,    +   492, -   638,    +   516, -  1734,    +   490, -   638, 
   +   492, -  1758,    +   490, -   666,    +   464, -   640,    +   488

uint16_t rawData[199] = {4424, 4546,  518, 612,  520, 612,  518, 610,  494, 1754,  496, 636,  518, 608,  492, 1758,  544, 588,  494, 638,  518, 608,  496, 1756,  492, 1754,  494, 638,  494, 1756,  492, 636,  520, 612,  494, 638,  516, 1732,  494, 636,  492, 1760,  514, 612,  492, 1788,  462, 1758,  494, 638,  490, 666,  466, 1782,  466, 1784,  466, 1758,  492, 1758,  492, 638,  516, 612,  516, 614,  494, 1760,  490, 636,  518, 640,  492, 1732,  516, 638,  494, 636,  468, 662,  466, 638,  518, 1730,  522, 612,  490, 1756,  516, 612,  494, 1756,  518, 610,  518, 1760,  464, 1784,  466, 5704,  4394, 4574,  492, 1758,  466, 1756,  516, 1760,  464, 664,  492, 1732,  516, 1734,  492, 662,  466, 1786,  490, 1786,  464, 1732,  490, 664,  492, 612,  490, 1784,  490, 644,  460, 1758,  492, 1758,  518, 1760,  490, 610,  496, 1782,  492, 636,  466, 1784,  466, 638,  516, 638,  438, 1818,  458, 1788,  464, 638,  492, 638,  494, 638,  490, 664,  492, 1734,  490, 1782,  462, 1764,  490, 666,  488, 1758,  466, 1758,  492, 638,  492, 1784,  490, 1760,  464, 1786,  466, 1786,  464, 640,  490, 1758,  492, 638,  516, 1734,  490, 638,  492, 1758,  490, 666,  464, 640,  488};  // UNKNOWN AF0245A

Timestamp : 000044.799
Library   : v2.6.3

Encoding  : UNKNOWN
Code      : 9C6008CD (100 bits)
Raw Timing[199]:
   +  4448, -  4546,    +   468, -   634,    +   518, -   612,    +   518, -   610, 
   +   496, -  1728,    +   546, -   610,    +   494, -   638,    +   518, -  1730, 
   +   516, -   614,    +   494, -   636,    +   494, -   636,    +   494, -  1782, 
   +   464, -  1758,    +   518, -   612,    +   494, -  1754,    +   520, -   610, 
   +   494, -   636,    +   494, -   662,    +   494, -  1730,    +   494, -   638, 
   +   490, -  1784,    +   466, -   636,    +   518, -  1758,    +   516, -  1732, 
   +   490, -   612,    +   492, -   664,    +   466, -  1758,    +   492, -  1758, 
   +   518, -  1732,    +   492, -  1756,    +   492, -   636,    +   518, -   616, 
   +   576, -   550,    +   496, -  1758,    +   490, -   638,    +   518, -   638, 
   +   466, -  1782,    +   466, -   638,    +   492, -   638,    +   516, -   612, 
   +   514, -   616,    +   516, -  1734,    +   492, -   664,    +   466, -  1758, 
   +   516, -   612,    +   492, -  1758,    +   516, -   612,    +   492, -  1758, 
   +   492, -  1784,    +   466, -  5704,    +  4396, -  4546,    +   492, -  1784, 
   +   466, -  1782,    +   492, -  1734,    +   518, -   608,    +   494, -  1756, 
   +   492, -  1760,    +   492, -   662,    +   468, -  1782,    +   468, -  1758, 
   +   492, -  1784,    +   490, -   638,    +   464, -   640,    +   494, -  1754, 
   +   494, -   638,    +   518, -  1756,    +   464, -  1786,    +   490, -  1736, 
   +   490, -   664,    +   466, -  1784,    +   488, -   614,    +   494, -  1756, 
   +   494, -   662,    +   468, -   662,    +   466, -  1760,    +   490, -  1784, 
   +   492, -   640,    +   464, -   690,    +   464, -   640,    +   466, -   638, 
   +   518, -  1734,    +   490, -  1784,    +   464, -  1786,    +   462, -   694, 
   +   440, -  1784,    +   466, -  1786,    +   464, -   666,    +   464, -  1786, 
[... cut ...]   490, 1758,  492, 1758,  492, 636,  466, 664,  518, 1760,  464, 638,  492, 1782,  492, 1734,  492, 1758,  516, 612,  492, 1786,  462, 664,  466, 1786,  464, 664,  466, 636,  518, 1734,  518, 1730,  520, 636,  466, 664,  490, 614,  490, 666,  466, 1784,  464, 1784,  466, 1758,  492, 638,  492, 1758,  494, 1782,  466, 640,  490, 1784,  490, 1732,  492, 1784,  466, 1760,  490, 692,  462, 1760,  492, 638,  464, 1786,  464, 664,  464, 1786,  464, 664,  490, 640,  464};  // UNKNOWN AF0245A

`

Thanks

crankyoldgit, 2019-09-10:
FYI, it's ``` on a line of it's own to start and end the "code" formatting of a block of text.

happytm, 2019-09-10:
Sorry. Did not know that. Thanks for hint.

crankyoldgit, 2019-09-10:
I do apologise. I missed this line: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Midea.cpp#L472

MIDEA has a checksum in it, so you can't just send arbitrary data.
Thus, you can't use `sendMidea()`. Sorry for sending you down that wrong path.

I'll see if I can find you another protocol that will suit you.

happytm, 2019-09-10:
Thank you.
All I need to do is send one 6 byte packet which is self generated. 

Thanks.

crankyoldgit, 2019-09-11:
I had a look over all the protocols last night. None of the "simple" ones are large enough to do what you want.
You'd ether have to massage the data to pass what ever checks a protocol had, or send it via multiple messages.

I think your earlier samsung approach you had is unfortunately the better solution. :-/

If you wanted to "modify" the library, I could easily tell you things to change to get it handle the MIDEA case or to use some other protocol to do this by tweaking some code, but if you update the library, you'd have to make those changes again etc.

happytm, 2019-09-11:
I think I would like to make changes to library and I will keep copy of it in another folder if I accidentally upgrade the library I will still have old copy somewhere.I would like to use midea or samsung if possible.

Thanks

crankyoldgit, 2019-09-11:
If you change this line: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRrecv.h#L402

to:
```
                   bool strict = false);
```

e.g.
```
#if DECODE_MIDEA
  bool decodeMidea(decode_results *results, uint16_t nbits = kMideaBits,
                   bool strict = true);
#endif
```

Then the previous MIDEA suggestions should work.

I'll write a test case for it to confirm it in a sec.

happytm, 2019-09-11:
I just uploaded my sketch to see what I am trying to do here. On receiving side I am going to use OpenMQTTGateway which uses your library.Link to my code below:

https://github.com/happytm/Triple-Sender/blob/master/Triple%20Sender/Sender/Sender.ino/Sender/Sender.ino

Everything else is working except IR data are not received reliably. 
Thanks

crankyoldgit, 2019-09-11:
I've added a test case (See #904) and confirmed it should work if `strict` is disabled on Midea.

If you want it to work with some project that uses this library, you are going to have to ensure it is using a version of this library with that flag set to `false` like I've described.

I can't and won't provide direct support for someone else's project per second last point of: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide#things-we-are-unlikely-to-help-you-with

happytm, 2019-09-11:
I wanted to confirm that change in library solved the problem. Now it is showing same hex code every time without errors.

I really appreciate your efforts to resolve the issue.

Thanks.

crankyoldgit, 2019-09-11:
> I wanted to confirm that change in library solved the problem. Now it is showing same hex code every time without errors.
> 
> I really appreciate your efforts to resolve the issue.

Excellent. Can I mark this issue closed now?

happytm, 2019-09-11:
Yes. Thank you.

crankyoldgit, 2019-09-11:
Good luck!

closed 2019-09-11 by crankyoldgit

happytm, 2019-09-13:
In case anyone wants to use custom message I used following code to send custom message over IR:

````#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>

int value1 = 1;
int value2 = 10;
int value3 = 20;
int value4 = 30;
int value5 = 40;
int value6 = 50; 

uint8_t msg[] = {value1,value2,value3,value4,value5,value6}; 

const uint16_t kIrLed = 16;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRsend irsend(kIrLed);  // Set the GPIO to be used to sending the message.

//uint64_t irData = 0x1234567890AB; // 48bits 
uint64_t irData = 
    (uint64_t)(uint8_t (msg[5])) | 
    (uint64_t)(uint8_t (msg[4])) << 8 |          
    (uint64_t)(uint8_t (msg[3])) << 16 |          
    (uint64_t)(uint8_t (msg[2])) << 24 |          
    (uint64_t)(uint8_t (msg[1])) << 32 |          
    (uint64_t)(uint8_t (msg[0])) << 40;    

void setup() {
Serial.begin(115200, SERIAL_8N1, SERIAL_TX_ONLY);
Serial.println();

irsend.begin();

}

void loop() {

//===========================IR loop===============================
  
  Serial.println("Starting IR data sender");
  Serial.println("Sending custom data over IR protocol to Master Node");
  Serial.println();
   
   irsend.sendMidea(irData); 
   Serial.print("Message sent");
   Serial.println();
   
}