crankyoldgit/IRremoteESP8266 #888: Add unsupported Mitsubishi Electric Ducted AC

sheppy99, 2019-09-06:
### Version/revision of the library used
V2.6.5
### Expected behavior
Loaded IRrecvDumpV2 to analyze unknown protocol
#### Output of raw data from IRrecvDumpV2.ino (if applicable)
Encoding  : UNKNOWN
Code      : 66B4490E (138 bits)
Raw Timing[275]:
   +  3324, -  1474,    +   520, -  1110,    +   492, -  1110,    +   524, -   314, 
   +   498, -   318,    +   466, -   336,    +   474, -  1124,    +   514, -   322, 
   +   464, -   338,    +   472, -  1124,    +   516, -  1112,    +   482, -   342, 
   +   480, -  1118,    +   488, -   338,    +   466, -   344,    +   480, -  1124, 
   +   480, -  1124,    +   510, -   328,    +   484, -  1114,    +   480, -  1132, 
   +   510, -   330,    +   456, -   344,    +   464, -  1134,    +   506, -   334, 
   +   452, -   346,    +   462, -  1136,    +   504, -   336,    +   450, -   348, 
   +   460, -   350,    +   472, -   338,    +   474, -  1124,    +   472, -   352, 
   +   472, -   340,    +   474, -   342,    +   446, -   354,    +   454, -   354, 
   +   468, -   344,    +   470, -   344,    +   442, -   356,    +   450, -   358, 
   +   466, -   346,    +   466, -   348,    +   440, -   360,    +   448, -   360, 
   +   462, -   350,    +   464, -   352,    +   438, -   360,    +   434, -  1162, 
   +   490, -   350,    +   438, -  1148,    +   486, -   350,    +   464, -   352, 
   +   436, -   362,    +   432, -   376,    +   462, -   352,    +   462, -  1138, 
   +   448, -   376,    +   460, -  1142,    +   462, -  1150,    +   484, -  1140, 
   +   462, -   360,    +   446, -  1152,    +   492, -  1132,    +   460, -   362, 
   +   466, -   348,    +   466, -   348,    +   438, -   360,    +   446, -  1152, 
   +   492, -   348,    +   436, -   360,    +   434, -   374,    +   462, -   350, 
   +   464, -   350,    +   436, -   362,    +   434, -   376,    +   460, -   352, 
   +   462, -   352,    +   434, -   364,    +   432, -   378,    +   458, -   354, 
   +   460, -   356,    +   434, -   364,    +   430, -   380,    +   456, -   356, 
   +   458, -   356,    +   432, -   366,    +   428, -   382,    +   454, -   358, 
   +   456, -   358,    +   430, -  1158,    +   476, -  1146,    +   458, -  1156, 
   +   478, -  1150,    +   454, -  1154,    +   480, -  1142,    +   460, -   362, 
   +   434, -  1164,    +   488, -   352,    +   436, -  1148,    +   488, -  1140, 
   +   462, -  1144,    +   490, -  1136,    +   466, -  1142,    +   492, -   346, 
   +   466, -  1132,    +   462, -   360,    +   466, -   348,    +   466, -   350, 
   +   438, -  1152,    +   482, -   348,    +   464, -   350,    +   438, -  1144, 
   +   490, -  1142,    +   462, -  1148,    +   486, -  1138,    +   466, -   354, 
   +   450, -  1146,    +   496, -  1132,    +   460, -  1150,    +   494, -  1130, 
   +   464, -  1146,    +   498, -  1130,    +   464, -  1144,    +   498, -  1130, 
   +   462, -  1144,    +   500, -  1126,    +   468, -  1142,    +   502, -  1122, 
   +   470, -  1142,    +   502, -  1130,    +   464, -  1140,    +   504, -  1124, 
   +   468, -  1140,    +   504, -  1122,    +   472, -  1142,    +   502, -  1122, 
   +   472, -  1138,    +   506

uint16_t rawData[275] = {3324, 1474,  520, 1110,  492, 1110,  524, 314,  498, 318,  466, 336,  474, 1124,  514, 322,  464, 338,  472, 1124,  516, 1112,  482, 342,  480, 1118,  488, 338,  466, 344,  480, 1124,  480, 1124,  510, 328,  484, 1114,  480, 1132,  510, 330,  456, 344,  464, 1134,  506, 334,  452, 346,  462, 1136,  504, 336,  450, 348,  460, 350,  472, 338,  474, 1124,  472, 352,  472, 340,  474, 342,  446, 354,  454, 354,  468, 344,  470, 344,  442, 356,  450, 358,  466, 346,  466, 348,  440, 360,  448, 360,  462, 350,  464, 352,  438, 360,  434, 1162,  490, 350,  438, 1148,  486, 350,  464, 352,  436, 362,  432, 376,  462, 352,  462, 1138,  448, 376,  460, 1142,  462, 1150,  484, 1140,  462, 360,  446, 1152,  492, 1132,  460, 362,  466, 348,  466, 348,  438, 360,  446, 1152,  492, 348,  436, 360,  434, 374,  462, 350,  464, 350,  436, 362,  434, 376,  460, 352,  462, 352,  434, 364,  432, 378,  458, 354,  460, 356,  434, 364,  430, 380,  456, 356,  458, 356,  432, 366,  428, 382,  454, 358,  456, 358,  430, 1158,  476, 1146,  458, 1156,  478, 1150,  454, 1154,  480, 1142,  460, 362,  434, 1164,  488, 352,  436, 1148,  488, 1140,  462, 1144,  490, 1136,  466, 1142,  492, 346,  466, 1132,  462, 360,  466, 348,  466, 350,  438, 1152,  482, 348,  464, 350,  438, 1144,  490, 1142,  462, 1148,  486, 1138,  466, 354,  450, 1146,  496, 1132,  460, 1150,  494, 1130,  464, 1146,  498, 1130,  464, 1144,  498, 1130,  462, 1144,  500, 1126,  468, 1142,  502, 1122,  470, 1142,  502, 1130,  464, 1140,  504, 1124,  468, 1140,  504, 1122,  472, 1142,  502, 1122,  472, 1138,  506};  // UNKNOWN 66B4490E

### Steps to reproduce the behavior
Send via Remote ON, 20C, Cooling, MaxFan

crankyoldgit, 2019-09-07:
Hey Kevin / @sheppy99 , you know the drill by now.
Please download and try out the Issue888 branch (https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue888) associated with PR #890 

It's just basic support. So if you wanted detailed support, as per other times, you'll need to break-down/analyse the protocol data like last time.

Enjoy `MITSUBISHI136` ;-) Let me know what works/doesn't etc.

sheppy99, 2019-09-07:
I've just compiled it onto an ESP and basic support seems to be working on the receive side, although it'll be next week before I can point it at the AC and check it works. Once I know if it can calculate its own checksum - question asked on the common a/c framework thread - I'll code it into a rule and change the IR wires over.
Thanks for the prompt response, I'll have to head over the ditch again and buy you another beer ;-)

crankyoldgit, 2019-09-07:
MMMmmm beer.

crankyoldgit, 2019-09-07:
Oh, and thanks for confirming receiving is working. It's now merged into the `master` branch.

closed 2019-09-07 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

commit 20f7ea9e (2019-09-07): Basic support for Mitsubishi136 A/C protocol. (#890)

* Basic support for Mitsubishi136 protocol.
* `sendMitsubishi136()` & `decodeMitsubishi136()` only.
* No detailed support as yet.
* Supporting unit tests and relevant supporting code changes.

Fixes #888