crankyoldgit/IRremoteESP8266 #891: not working with mitsuibishi Electric - model no: MS-GK24VA

jenish-jain, 2019-09-07:
_(Please use this template for reporting issues. You can delete what ever is not relevant. Giving us this information will help us help you faster. Please also read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions) & [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide). Your problem may already have an answer there.)_

### Version/revision of the library used
_Typically located in the `library.json` & `src/IRremoteESP8266.h` files in the root directory of the library.
e.g. v2.0.0, or 'master' as at 1st of June, 2017. etc._

### Expected behavior
_What steps did you do and what should it have done?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar branded BBQ turns on and cooks me some ribs.

### Actual behavior
_What steps did you do, and what did or didn't actually happen?_

e.g.
1. Initialise the IRsend class.
2. IRsend.sendFoobar(0xdeadbeef);
3. Foobar BBQ went into Cow(er)-saving mode and fried me a couple of eggs instead.

#### Output of raw data from IRrecvDumpV2.ino (if applicable)
_Include some raw dumps of what the device saw._

### Steps to reproduce the behavior
_What can we do to (pref. reliably) repeat what is happening?_

#### Example code used
_Include all relevant code snippets or links to the actual code files. Tip: [How to quote your code so it is still readable](https://github.com/adam-p/markdown-here/wiki/Markdown-Cheatsheet#code)._

#### Circuit diagram and hardware used (if applicable)
_Link to an image of the circuit diagram used. Part number of the IR receiver module etc. ESP8266 or ESP32 board type._

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes/No._

### Has this library/code previously worked as expected for you?
_Yes/No. If "Yes", which version last worked for you?_

### Other useful information
_More information is always welcome. Be verbose._

crankyoldgit, 2019-09-07:
more info please.

jenish-jain, 2019-09-07:
i have tried sending your mitsuibishi.h library example after making the necessary changes to turn on and send temp values to a mitsuibishi electric  Model no: MS-GK24VA air conditioner but it is not sending commands , i have tested the circuit fr a whirlpool AC and thus there are no faults with the circuit.

crankyoldgit, 2019-09-07:
You've provided no code or data for me to evaluate. Without it it is hard to be able to help you.

A good first step would be to follow the tips/steps/examples in the Issue template. e.g.
> **Output of raw data from IRrecvDumpV2.ino (if applicable)**
Include some raw dumps of what the device saw.

...
> **Example code used**
Include all relevant code snippets or links to the actual code files. Tip: How to quote your code so it is still readable.

jenish-jain, 2019-09-07:
``
/*
    for Radhey HVAC engineers Iot device to control Air Conditioner's temperature [ slave code ac1]
    developed by nutrient code
    for queries contact jenish jain at jenishjain@nutrientcode.com or administrator@nutrientcode.com
*/
#ifndef UNIT_TEST
#include <Arduino.h>
#endif
//#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <ir_Mitsubishi.h>
//#include <ir_Whirlpool.h>

#include <ArduinoJson.h>
#include <ESP8266WiFi.h>
#include <ESP8266HTTPClient.h>
WiFiClient client;

const char* host = "radheyhvac.com";
#define usid 164000029

#define chipId ESP.getChipId()
/*
  needed libraries for managing different wifi credentials
*/
#include <DNSServer.h>
#include <ESP8266WebServer.h>
#include <WiFiManager.h>

//for LED status
#include <Ticker.h>
#define conn_led 1   // connected on GPIO01 | D
Ticker ticker;
int flag1, flag2;
int temp;
int S_temp;
int R_temp;
int ac_status_one = 2; // 1 for on 2 for off
int ac_status_two = 2; // 1 for on 2 for off
int delayTime = 300000; // 5 minutes delay = 5*60*1000 microseconds
int sensorValue = 0;  // variable to store the value coming from the sensor

String Mode = "Online";
int count = 0;
String url;

const uint16_t kIrLed = 2;  // ESP8266 GPIO pin to use. Recommended: 4 (D2).
IRMitsubishiAC ac(kIrLed);  // Set the GPIO used for sending messages.
//IRWhirlpoolAc ac(kIrLed);   // Set the GPIO used for sending messages.

void tick()
{
  //toggle state
  int state = digitalRead(conn_led); // get the current state of GPIO1 pin
  digitalWrite(conn_led, !state);    // set pin to the opposite state
}

void configModeCallback(WiFiManager *myWiFiManager)
{
  Serial.println("Entered config mode");
  Serial.println(WiFi.softAPIP());
  //if you used auto generated SSID, print it
  Serial.println(myWiFiManager->getConfigPortalSSID());
  //entered config mode, make led toggle faster
  ticker.attach(0.2, tick);
}

void printState() {
  // Display the settings.
[... 557 lines cut ...]
  String Current_Mode = Mode;
  Serial.println(Current_Mode);
  checkMode();
  Serial.println(Mode);
  if ( !(Current_Mode.equals(Mode))) {
    Serial.println("Switching to  ");
    Serial.println(Mode);  
    ESP.reset();
  }
  if (Current_Mode.equals(Mode)) {
    return;
  }
}

void Timerdelay( int time_val) {

  int StartTime = millis();

  while ( millis() <= StartTime + time_val) {
    Serial.println("executing delay");
    FetchFreshValues();
  }
}``

this is the code i am using

crankyoldgit, 2019-09-07:
Does `IRrecvDumpV2.ino` recognise the protocol correctly? What is the output for a similar message on the actual remote?

jenish-jain, 2019-09-07:
I have checked it with a similar kind of ac but since this particular one is not in my reach I am making this project for a remote site.

crankyoldgit, 2019-09-07:
It may be the case that protocol/model is not yet supported. `IRrecvDumpV2` is the ultimate tool to use to see if it is understood/supported at all.

jenish-jain, 2019-09-07:
In case the protocol/model is not supported what all changes do I have to make to make it work or which information do you require to provide support?

crankyoldgit, 2019-09-07:
https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

jenish-jain, 2019-09-14:
Timestamp : 000464.638
Library   : v2.6.5

Encoding  : UNKNOWN
Code      : 23F3C6B3 (289 bits)
Raw Timing[577]:
   +  3364, -  1790,    +   386, -  1340,    +   362, -  1370,    +   364, -   530, 
   +   336, -  1372,    +   388, -  1344,    +   362, -   508,    +   360, -   530, 
   +   338, -  1362,    +   362, -  1360,    +   362, -   530,    +   358, -  1342, 
   +   362, -   528,    +   336, -   528,    +   338, -  1364,    +   362, -  1362, 
   +   362, -   530,    +   338, -  1360,    +   362, -  1360,    +   360, -   532, 
   +   336, -   528,    +   338, -  1362,    +   362, -   502,    +   388, -   514, 
   +   336, -  1366,    +   362, -   528,    +   338, -   528,    +   336, -   528, 
   +   336, -   506,    +   360, -   530,    +   338, -   526,    +   360, -   516, 
   +   336, -   532,    +   336, -   542,    +   338, -   530,    +   336, -   528, 
   +   338, -   526,    +   338, -   530,    +   338, -   530,    +   336, -   528, 
   +   336, -   532,    +   336, -   528,    +   338, -   532,    +   336, -   532, 
   +   338, -   526,    +   364, -  1344,    +   362, -   530,    +   336, -   528, 
   +   338, -   530,    +   338, -   504,    +   360, -   526,    +   338, -   526, 
   +   360, -  1340,    +   362, -   528,    +   340, -   526,    +   338, -   524, 
   +   336, -   528,    +   336, -   532,    +   336, -   526,    +   338, -  1362, 
   +   362, -   524,    +   338, -   528,    +   360, -   508,    +   338, -   528, 
   +   336, -   528,    +   336, -  1364,    +   362, -   530,    +   360, -   510, 
   +   364, -  1344,    +   364, -  1360,    +   362, -   532,    +   338, -   530, 
   +   338, -  1362,    +   362, -   532,    +   338, -   530,    +   360, -   512, 
   +   338, -   526,    +   336, -   530,    +   336, -  1364,    +   362, -   508, 
   +   358, -   528,    +   336, -   526,    +   360, -   508,    +   338, -   528, 
   +   336, -   528,    +   338, -   530,    +   338, -   528,    +   338, -   528, 
   +   338, -   526,    +   338, -   528,    +   336, -   528,    +   338, -   532, 
   +   338, -   526,    +   336, -   526,    +   336, -   526,    +   362, -   506, 
   +   336, -   532,    +   336, -   528,    +   360, -   508,    +   336, -   528, 
   +   336, -   534,    +   336, -   526,    +   336, -   528,    +   336, -   528, 
   +   338, -  1374,    +   360, -   532,    +   360, -   512,    +   336, -   526, 
   +   338, -   532,    +   336, -   526,    +   338, -   526,    +   338, -   530, 
   +   338, -   526,    +   338, -   530,    +   338, -   536,    +   338, -   530, 
   +   338, -   528,    +   338, -   530,    +   360, -   510,    +   338, -   530, 
   +   336, -   530,    +   338, -   528,    +   336, -   524,    +   336, -   532, 
   +   338, -   534,    +   366, -   510,    +   338, -   526,    +   362, -   506, 
   +   338, -   528,    +   336, -   528,    +   338, -   530,    +   336, -   526, 
   +   336, -   532,    +   336, -   530,    +   336, -   526,    +   362, -   506, 
   +   336, -   536,    +   362, -   506,    +   338, -   528,    +   336, -   530, 
   +   336, -   530,    +   336, -  1364,    +   386, -  1338,    +   362, - 17130, 
   +  3486, -  1796,    +   360, -  1366,    +   386, -  1340,    +   362, -   526, 
   +   336, -   528,    +   336, -   530,    +   336, -  1362,    +   362, -   524, 
   +   338, -   524,    +   338, -  1366,    +   384, -  1340,    +   386, -   502, 
   +   338, -  1362,    +   362, -   528,    +   338, -   534,    +   336, -  1362, 
   +   362, -  1360,    +   362, -   530,    +   338, -  1364,    +   386, -  1336, 
   +   362, -   532,    +   338, -   528,    +   338, -  1358,    +   362, -   530, 
   +   336, -   528,    +   336, -  1364,    +   362, -   528,    +   362, -   506, 
   +   336, -   528,    +   340, -   530,    +   338, -   526,    +   338, -   526, 
   +   336, -   528,    +   360, -   508,    +   336, -   532,    +   336, -   530, 
   +   338, -   526,    +   338, -   528,    +   338, -  1550,    +   190, -   526, 
   +   338, -   526,    +   338, -   528,    +   338, -   526,    +   360, -   510, 
   +   338, -   526,    +   336, -  1358,    +   364, -   524,    +   336, -   528, 
   +   338, -   530,    +   336, -   530,    +   336, -   532,    +   338, -   524, 
   +   338, -  1370,    +   362, -   530,    +   338, -   526,    +   338, -   526, 
   +   360, -   508,    +   336, -   504,    +   360, -   530,    +   338, -  1364, 
   +   362, -   530,    +   338, -   528,    +   338, -   526,    +   362, -   506, 
   +   336, -   524,    +   338, -  1366,    +   362, -   528,    +   338, -   524, 
   +   336, -  1362,    +   362, -  1362,    +   362, -   530,    +   338, -   526, 
   +   362, -  1340,    +   360, -   530,    +   336, -   530,    +   362, -   514, 
   +   336, -   508,    +   358, -   526,    +   338, -  1358,    +   362, -   528, 
   +   338, -   530,    +   338, -   526,    +   336, -   526,    +   338, -   532, 
   +   338, -   526,    +   362, -   510,    +   338, -   530,    +   338, -   528, 
   +   338, -   532,    +   338, -   526,    +   338, -   528,    +   338, -   532, 
   +   338, -   526,    +   338, -   530,    +   336, -   530,    +   334, -   530, 
   +   336, -   532,    +   336, -   528,    +   338, -   530,    +   338, -   526, 
   +   338, -   536,    +   336, -   528,    +   338, -   528,    +   336, -   524, 
   +   338, -   526,    +   336, -   534,    +   338, -   526,    +   336, -   530, 
[... cut ...]  526,  338, 528,  338, 530,  336, 528,  338, 538,  336, 532,  360, 518,  354, 512,  338, 528,  338, 526,  338, 530,  338, 530,  336, 528,  338, 530,  338, 528,  338, 530,  338, 532,  336, 528,  364, 512,  338, 530,  338, 526,  336, 526,  336, 530,  338, 528,  338, 530,  338, 526,  336, 532,  338, 1362,  362, 1362,  362};  // UNKNOWN 23F3C6B3

this is the hex code i am receiving , it is detecting wrong values i am sending 22 degrees but it says 16 and many time when i press button it says unknown

jenish-jain, 2019-09-14:
Timestamp : 000674.953
Library   : v2.6.5

Encoding  : MITSUBISHI_AC
Code      : 000000000000000000000000000000000000 (144 bits)
Mesg Desc.: Power: Off, Mode: 0 (UNKNOWN), Temp: 16C, Fan: 0 (Auto), Vane: AUTO, Wide Vane: 0, Time: 00:00, On timer: 00:00, Off timer: 00:00, Timer: -
Raw Timing[583]:
   +  3342, -  1814,    +   396, -  1326,    +   342, -  1384,    +   398, -   468, 
   +   340, -   528,    +   342, -   526,    +   342, -  1382,    +   398, -   468, 
   +   422, -   452,    +   396, -  1330,    +   364, -  1354,    +   342, -   526, 
   +   396, -  1324,    +   342, -   528,    +   430, -   448,    +   396, -  1326, 
   +   396, -  1326,    +   396, -   468,    +   340, -  1384,    +   342, -  1386, 
   +   396, -   472,    +   342, -   524,    +   340, -  1384,    +   396, -   470, 
   +   390, -   472,    +   396, -  1326,    +   368, -   496,    +   366, -   496, 
   +   396, -   470,    +   362, -   504,    +   398, -   468,    +   340, -   526, 
   +   340, -   526,    +   342, -   526,    +   366, -   498,    +   396, -   468, 
   +   394, -   470,    +   342, -   526,    +   396, -   466,    +   366, -   498, 
   +   396, -   466,    +   394, -   470,    +   386, -   484,    +   396, -   468, 
   +   394, -   470,    +   396, -   468,    +   340, -  1382,    +   366, -   498, 
   +   342, -   524,    +   342, -   526,    +   396, -   470,    +   396, -   468, 
   +   340, -  1382,    +   368, -  1354,    +   392, -   474,    +   396, -   466, 
   +   396, -   468,    +   340, -  1384,    +   396, -  1324,    +   396, -   468, 
   +   396, -   472,    +   364, -   500,    +   340, -   530,    +   396, -   470, 
   +   342, -   524,    +   342, -   526,    +   394, -  1328,    +   396, -  1328, 
   +   340, -   524,    +   396, -  1326,    +   396, -  1328,    +   396, -   470, 
   +   342, -   530,    +   398, -   472,    +   340, -  1384,    +   264, -   720, 
   +   248, -   510,    +   398, -   466,    +   342, -   528,    +   340, -  1382, 
   +   396, -   476,    +   342, -   526,    +   396, -   468,    +   340, -   526, 
   +   340, -   530,    +   338, -   526,    +   340, -   526,    +   398, -   468, 
   +   396, -   468,    +   398, -   472,    +   396, -   470,    +   340, -   526, 
   +   342, -   524,    +   340, -   532,    +   396, -   468,    +   340, -   526, 
   +   368, -   494,    +   398, -   466,    +   396, -   472,    +   394, -   468, 
   +   366, -   500,    +   340, -   528,    +   396, -   466,    +   340, -   540, 
   +   342, -   526,    +   396, -   472,    +   342, -   526,    +   394, -   470, 
   +   364, -   506,    +   422, -   454,    +   394, -   470,    +   396, -   470, 
   +   396, -   468,    +   396, -   468,    +   340, -   526,    +   364, -   504, 
   +   366, -   500,    +   394, -   468,    +   342, -   528,    +   340, -   528, 
   +   342, -   526,    +   340, -   524,    +   398, -   468,    +   340, -   526, 
   +   340, -   528,    +   396, -  1328,    +   340, -   524,    +   342, -   526, 
   +   396, -   466,    +   342, -   526,    +   340, -   528,    +   340, -   528, 
   +   396, -   468,    +   340, -   526,    +   394, -   472,    +   342, -   526, 
   +   396, -   468,    +   340, -   528,    +   394, -   472,    +   342, -   528, 
   +   396, -  1326,    +   396, -  1326,    +   342, -   526,    +   396, -  1330, 
   +   396, -  1324,    +   398, - 17078,    +  3534, -  1764,    +   398, -  1328, 
   +   396, -  1326,    +   342, -   528,    +   366, -   508,    +   342, -   526, 
   +   396, -  1324,    +   424, -   452,    +   396, -   468,    +   342, -  1382, 
   +   398, -  1326,    +   396, -   468,    +   396, -  1326,    +   342, -   528, 
   +   428, -   450,    +   396, -  1326,    +   398, -  1326,    +   396, -   468, 
   +   340, -  1384,    +   342, -  1384,    +   368, -   500,    +   340, -   526, 
   +   340, -  1412,    +   196, -   640,    +   366, -   496,    +   340, -  1388, 
   +   396, -   468,    +   394, -   468,    +   396, -   470,    +   396, -   466, 
   +   396, -   470,    +   340, -   526,    +   340, -   526,    +   340, -   528, 
   +   396, -   468,    +   396, -   468,    +   396, -   468,    +   340, -   526, 
   +   368, -   496,    +   366, -   498,    +   398, -   468,    +   396, -   468, 
   +   384, -   484,    +   366, -   498,    +   396, -   468,    +   396, -   468, 
   +   340, -  1384,    +   396, -   468,    +   340, -   524,    +   342, -   526, 
   +   398, -   468,    +   396, -   468,    +   340, -  1384,    +   396, -  1324, 
   +   398, -   468,    +   396, -   466,    +   398, -   466,    +   340, -  1384, 
   +   398, -  1324,    +   396, -   468,    +   396, -   472,    +   394, -   470, 
   +   364, -   508,    +   396, -   470,    +   340, -   526,    +   340, -   528, 
   +   366, -  1356,    +   398, -  1328,    +   340, -   524,    +   396, -  1324, 
   +   398, -  1326,    +   398, -   468,    +   342, -   530,    +   396, -   472, 
   +   342, -  1384,    +   426, -   450,    +   396, -   470,    +   396, -   466, 
   +   342, -   528,    +   382, -  1340,    +   396, -   474,    +   340, -   528, 
   +   366, -   496,    +   340, -   526,    +   340, -   530,    +   340, -   526, 
   +   340, -   526,    +   396, -   470,    +   396, -   470,    +   398, -   470, 
   +   394, -   472,    +   340, -   526,    +   340, -   524,    +   340, -   532, 
   +   396, -   468,    +   340, -   526,    +   366, -   496,    +   396, -   468, 
[... cut ...]  340, 524,  342, 528,  396, 1324,  342, 526,  396, 468,  340, 526,  342, 526,  396, 472,  396, 470,  340, 526,  396, 468,  366, 498,  396, 466,  396, 468,  340, 528,  342, 524,  396, 468,  374, 1346,  342, 1386,  342, 530,  394, 1330,  366, 1356,  368};  // MITSUBISHI_AC
uint8_t state[18] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};

this was when i sent 19 degrees, from the original remote but the code is detecting as 16 degrees

crankyoldgit, 2019-09-14:
FYI, please use ``` before and after those text dumps. It makes them format better.

jenish-jain, 2019-09-14:
okay will do that in further replies, do you need any more values from me or will these do? to make the support

crankyoldgit, 2019-09-14:
`Code      : 000000000000000000000000000000000000 (144 bits)`
`uint8_t state[18] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};`

These indicate something is not working in the docode routine. I'll look at it soon.

jenish-jain, 2019-09-14:
ok

crankyoldgit, 2019-09-14:
What remote/device etc is the above signals capture from? It looks to me like the library is making a false match against the `MITSUBISHI_AC` protocol.

jenish-jain, 2019-09-14:
i am using a mitsuibishi remote for the ac model no:  MS-GK24VA  the type on the remote printed is KM14A
0179213

crankyoldgit, 2019-09-14:
Thanks. 

Looking at the data by hand (and via some coded tests) there are a number of very short marks and spaces in the message. e.g. Too short for the protocol to be detected properly by the library.
e.g. `+   144,` & `+   196,`. All of those +values in the message should be almost always either `3400` or `430`-ish. That is, with in a hundred or so.

In fact, this protocol (`MITSUBISHI_AC`) was submitted by a user of the library, and it's full of problems now I've had a deeper look into it.

I've been able to hack the existing code to "kind-of" match the message, but your last message is failing it's checksum verification.

The "too short" sections in the message could be cause by a number of things. Flat-ish batteries & the remote being too far or too close to the receiver.

Can I please get you to check the batteries and try again with the remote about 1 metre from the IR receiver? e.g. If I'm to fix this and get it to match for you, you are going to need a better signal and I'm going to need a better capture from you.

jenish-jain, 2019-09-14:
i have replaced the batteries but still getting the same output 00000.. and detected as MITSUBISHI_AC

crankyoldgit, 2019-09-14:
Yes, it will until till I fix it. Is the data better. e.g. Did you check all the `+values` to see if they are in range. If so, post the result here.

crankyoldgit, 2019-09-14:
> Can I please get you to check the batteries and try again with the remote about 1 metre from the IR receiver? e.g. If I'm to fix this and get it to match for you, you are going to need a better signal **and** I'm going to need a better capture from you.

Note the "and" in my previous request.

jenish-jain, 2019-09-14:
```
Timestamp : 000815.516
Library   : v2.6.5

Encoding  : MITSUBISHI_AC
Code      : 000000000000000000000000000000000000 (144 bits)
Mesg Desc.: Power: Off, Mode: 0 (UNKNOWN), Temp: 16C, Fan: 0 (Auto), Vane: AUTO, Wide Vane: 0, Time: 00:00, On timer: 00:00, Off timer: 00:00, Timer: -
Raw Timing[583]:
   +  3418, -  1742,    +   386, -  1342,    +   398, -  1324,    +   366, -   498, 
   +   342, -   524,    +   384, -   482,    +   366, -  1354,    +   340, -   528, 
   +   366, -   500,    +   396, -  1328,    +   366, -  1354,    +   340, -   522, 
   +   340, -  1384,    +   342, -   528,    +   366, -   496,    +   340, -  1382, 
   +   398, -  1324,    +   386, -   482,    +   338, -  1386,    +   420, -  1320, 
   +   340, -   526,    +   366, -   500,    +   396, -  1324,    +   398, -   466, 
   +   384, -   480,    +   340, -  1382,    +   340, -   530,    +   340, -   526, 
   +   380, -   486,    +   366, -   500,    +   384, -   480,    +   424, -   452, 
   +   398, -   466,    +   380, -   488,    +   366, -   500,    +   396, -   470, 
   +   340, -   526,    +   366, -   496,    +   366, -   502,    +   396, -   468, 
   +   342, -   522,    +   384, -   482,    +   342, -   530,    +   386, -   482, 
   +   340, -   524,    +   396, -   468,    +   366, -   500,    +   382, -   486, 
   +   366, -   500,    +   366, -   502,    +   396, -   468,    +   366, -   502, 
   +   366, -  1356,    +   340, -  1380,    +   382, -   484,    +   386, -   482, 
   +   342, -   526,    +   362, -   506,    +   340, -   526,    +   338, -   526, 
   +   340, -  1388,    +   366, -   500,    +   380, -   486,    +   366, -   500, 
   +   366, -   498,    +   380, -   488,    +   416, -  1308,    +   412, -  1316, 
   +   368, -   500,    +   384, -  1338,    +   396, -  1324,    +   382, -   488, 
   +   368, -   498,    +   380, -   488,    +   340, -   524,    +   366, -   502, 
   +   384, -   480,    +   418, -   452,    +   396, -   468,    +   340, -  1382, 
   +   366, -   498,    +   366, -   500,    +   366, -   496,    +   340, -   528, 
   +   366, -   506,    +   342, -   528,    +   340, -   526,    +   340, -   524, 
   +   412, -   458,    +   340, -   528,    +   366, -   502,    +   340, -   526, 
   +   338, -   528,    +   396, -   466,    +   396, -   466,    +   366, -   496, 
   +   366, -   500,    +   366, -   502,    +   366, -   498,    +   366, -   500, 
   +   396, -   470,    +   396, -   470,    +   386, -   484,    +   366, -   498, 
   +   382, -   496,    +   396, -   470,    +   368, -   498,    +   366, -   500, 
   +   396, -   474,    +   342, -   524,    +   342, -   524,    +   366, -   500, 
   +   396, -   470,    +   366, -   502,    +   366, -   498,    +   380, -   488, 
   +   340, -   522,    +   412, -   460,    +   396, -   468,    +   396, -   468, 
   +   366, -   496,    +   340, -   522,    +   366, -   504,    +   396, -   466, 
   +   396, -   470,    +   340, -   526,    +   396, -   468,    +   412, -   470, 
   +   396, -   470,    +   366, -   502,    +   384, -   482,    +   366, -   498, 
   +   418, -   452,    +   424, -   450,    +   366, -   496,    +   342, -   524, 
   +   340, -   524,    +   366, -  1356,    +   396, -  1326,    +   366, -   496, 
   +   340, -  1382,    +   396, -   470,    +   366, -  1356,    +   342, -   526, 
   +   396, -  1322,    +   386, - 17100,    +  3524, -  1772,    +   366, -  1358, 
   +   396, -  1326,    +   364, -   506,    +   366, -   500,    +   384, -   482, 
   +   396, -  1324,    +   340, -   526,    +   340, -   524,    +   342, -  1380, 
   +   342, -  1382,    +   342, -   526,    +   338, -  1382,    +   386, -   484, 
   +   428, -   450,    +   364, -  1356,    +   366, -  1358,    +   366, -   498, 
   +   412, -  1312,    +   382, -  1346,    +   368, -   500,    +   384, -   482, 
   +   398, -  1326,    +   366, -   500,    +   396, -   466,    +   412, -  1314, 
   +   342, -   526,    +   380, -   490,    +   340, -   526,    +   384, -   484, 
   +   396, -   466,    +   366, -   498,    +   340, -   522,    +   342, -   524, 
   +   382, -   488,    +   366, -   494,    +   340, -   524,    +   366, -   496, 
   +   352, -   520,    +   340, -   522,    +   380, -   486,    +   366, -   498, 
   +   340, -   526,    +   340, -   524,    +   382, -   488,    +   366, -   498, 
   +   396, -   470,    +   342, -   524,    +   340, -   524,    +   366, -   500, 
   +   366, -   498,    +   366, -   498,    +   414, -  1312,    +   366, -  1354, 
   +   362, -   508,    +   340, -   524,    +   340, -   528,    +   422, -   454, 
   +   422, -   452,    +   396, -   468,    +   384, -  1340,    +   366, -   502, 
   +   412, -   460,    +   426, -   450,    +   396, -   466,    +   382, -   486, 
   +   366, -  1358,    +   382, -  1344,    +   414, -   458,    +   366, -  1356, 
   +   382, -  1342,    +   386, -   482,    +   366, -   494,    +   386, -   482, 
   +   342, -   524,    +   342, -   524,    +   380, -   484,    +   366, -   500, 
   +   384, -   480,    +   428, -  1306,    +   366, -   502,    +   396, -   472, 
   +   340, -   526,    +   366, -   496,    +   420, -   456,    +   380, -   486, 
   +   366, -   498,    +   366, -   496,    +   398, -   466,    +   340, -   524, 
   +   382, -   490,    +   366, -   494,    +   342, -   524,    +   396, -   466, 
   +   380, -   490,    +   340, -   524,    +   396, -   470,    +   394, -   478, 
[... cut ...]  460,  426, 450,  396, 468,  380, 1344,  340, 1380,  414, 460,  380, 1342,  386, 482,  366, 1354,  340, 526,  340, 1386,  396};  // MITSUBISHI_AC
uint8_t state[18] = {0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00};

```

after replacing the batteries the values are still out of range if am checking it right i am sending you a sample to have a look at that. if you want some more samples or some other details with the output let me know.

jenish-jain, 2019-09-14:
and these are from 1 meter from the receiver.

crankyoldgit, 2019-09-14:
Thanks, that's a much better raw signal.

jenish-jain, 2019-09-14:
what else do you need now, should I send some other set of signals with their settings as displayed on the remote?

crankyoldgit, 2019-09-14:
Not yet.

crankyoldgit, 2019-09-14:
If you download and try out the code in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue891 / PR #914 it should hopefully detect your remote.

Please let me know how it goes.

jenish-jain, 2019-09-14:
ok letting you know just in few minutes.

crankyoldgit, 2019-09-14:
I won't be checking until tomorrow

jenish-jain, 2019-09-14:
ok

jenish-jain, 2019-09-14:
I have checked with almost all the features of the original remote and it is giving me obvious results as expected and shown on the remote, thanks a lot, you are doing a great job.

closed 2019-09-15 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

PR #914 (2019-09-15): MitsubishiAC: Improve decoding.
* Tweak tolerance and excess parameters.
* Fix a couple of potential code gotchas.
* Add unit test case to ensure it matches user's data.

Fixes #891
commit 707ffbe0 (2019-09-15): MitsubishiAC: Improve decoding. (#914)

* Tweak tolerance and excess parameters.
* Fix a couple of potential code gotchas.
* Add unit test case to ensure it matches user's data.
* Update compatible model info

Fixes #891