crankyoldgit/IRremoteESP8266 #892: Possible Typo in IRMQTTServer.h

sheppy99, 2019-09-09:
I've just found a possible Typo on line 71 of IRMQTTServer.h

https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.h#L71

HIDE_DUPLIATE_NETWORKS

is it missing a "C" or is WiFiManager missing a "C"?

HIDE_DUPLICATE_NETWORKS

Everything compiles so maybe it just spits an error somewhere that flies past the compilation window whilst being ignored

crankyoldgit, 2019-09-09:
Typo fixed. At least I was consistent, so if you had changed it's value, (true/false) it would work as expected.

closed 2019-09-09 by crankyoldgit

sheppy99, 2019-09-09:
All good, I only found it whilst looking for anything apart from JSON that I might want to change and figured it was the sort of thing that came back to bite

commit 5017afe6 (2019-09-09): Fix a typo.

Fixes #892