crankyoldgit/IRremoteESP8266 #893: Can't compile IRMQTTServer with #define MQTT_CLIMATE_JSON true

sheppy99, 2019-09-09:
It won't compile with #define MQTT_CLIMATE_JSON true, compiles OK with #define MQTT_CLIMATE_JSON false.

Errors:
` void debug(const char *str) {

      ^

D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino: In function 'stdAc::state_t jsonToState(stdAc::state_t, String)':

IRMQTTServer:2678:53: error: no matching function for call to 'deserializeJson(ArduinoJson6120_000001::DynamicJsonDocument&, const String&, const uint16_t&)'

   if (deserializeJson(json, str, kJsonAcStateMaxSize)) {

                                                     ^

D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:2678:53: note: candidates are:

In file included from D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:33:0,

                 from D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:314:

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:415:22: note: ArduinoJson6120_000001::DeserializationError ArduinoJson6120_000001::deserializeJson(ArduinoJson6120_000001::JsonDocument&, TInput&, ArduinoJson6120_000001::NestingLimit) [with TInput = const String]

 DeserializationError deserializeJson(

                      ^

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:415:22: note:   no known conversion for argument 3 from 'const uint16_t {aka const short unsigned int}' to 'ArduinoJson6120_000001::NestingLimit'

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:408:22: note: template<class TInput> ArduinoJson6120_000001::DeserializationError ArduinoJson6120_000001::deserializeJson(ArduinoJson6120_000001::JsonDocument&, TInput*, size_t, ArduinoJson6120_000001::NestingLimit)

 DeserializationError deserializeJson(

                      ^

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:408:22: note:   template argument deduction/substitution failed:

D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:2678:53: note:   mismatched types 'TInput*' and 'String'

   if (deserializeJson(json, str, kJsonAcStateMaxSize)) {

                                                     ^

In file included from D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:33:0,

                 from D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:314:

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:401:22: note: template<class TInput> ArduinoJson6120_000001::DeserializationError ArduinoJson6120_000001::deserializeJson(ArduinoJson6120_000001::JsonDocument&, TInput*, ArduinoJson6120_000001::NestingLimit)

 DeserializationError deserializeJson(

                      ^

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:401:22: note:   template argument deduction/substitution failed:

D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:2678:53: note:   mismatched types 'TInput*' and 'String'

   if (deserializeJson(json, str, kJsonAcStateMaxSize)) {

                                                     ^

In file included from D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.hpp:33:0,

                 from D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson.h:9,

                 from D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:314:

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:394:22: note: ArduinoJson6120_000001::DeserializationError ArduinoJson6120_000001::deserializeJson(ArduinoJson6120_000001::JsonDocument&, const TInput&, ArduinoJson6120_000001::NestingLimit) [with TInput = String]

 DeserializationError deserializeJson(

                      ^

D:\My Documents\Arduino\libraries\ArduinoJson\src/ArduinoJson/Json/JsonDeserializer.hpp:394:22: note:   no known conversion for argument 3 from 'const uint16_t {aka const short unsigned int}' to 'ArduinoJson6120_000001::NestingLimit'
`

crankyoldgit, 2019-09-09:
Fixed. Please download & try the `master` branch again.

closed 2019-09-09 by crankyoldgit

sheppy99, 2019-09-09:
Its compiled and now I have JSON
Presumably any value can be sent as JSON and the rest will be remembered from the previous state? How often does it write to flash with this setup?

crankyoldgit, 2019-09-09:
> Presumably any value can be sent as JSON and the rest will be remembered from the previous state?

Yes, for climate stuff only of course.

> How often does it write to flash with this setup?

Never.

sheppy99, 2019-09-09:
Perfect, thanks!

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

commit e12e1941 (2019-09-09): Fix compilation error when `MQTT_CLIMATE_JSON` is true

Ref #878
Fixes #893