crankyoldgit/IRremoteESP8266 #894: Update for Fujitsu AR-RY4 Remote

sheppy99, 2019-09-09:
I've just been testing the common ac framework with a Fujitsu AR-RY4 remote which came with my Hi-Wall Units, all the common functions are decoded correctly, apart from "Air Clean" and an additional function "Coil Dry."
Air Clean seems to be nibble 29, Air Clean ON = 8, Air Clean OFF = 0
Coil Dry seems to be nibble 19, Coil Dry Selected = 8
Coil Dry is selected with the unit ON and it goes into a Mold Guard type mode to dry the internal unit. As mine has a habit of growing black stuff I'm keen on getting this one, although I could send a full byte string instead if its a lot of work.

crankyoldgit, 2019-09-09:
Can you give me the full `state[]` codes for both settings, one with it off, and one with it on. This is so I can add it to the unit tests so I know I got the fixes correct.

sheppy99, 2019-09-09:
Take a look at the sheet, I added another tab to it

crankyoldgit, 2019-09-09:
For the people watching at home, that's https://docs.google.com/spreadsheets/d/1f8EGfIbBUo2B-CzUFdrgKQprWakoYNKM80IKZN4KXQE/edit#gid=646887633&range=A27:B30

crankyoldgit, 2019-09-09:
BTW, what Fujitsu model did you select/use in the library?

sheppy99, 2019-09-09:
I've got 2, 
AGTV14LAC Floor Console which doesn't have Air Cleaner Mode, and is used with AR-RAC1E remote, ASTB09LBC Wall Mount which does have Air Cleaner Mode and is used with AR-RY4 remote

crankyoldgit, 2019-09-09:
I'm referring to the "model" setting on the climate/common a/c stuff.

crankyoldgit, 2019-09-09:
Hey, your spread sheet seems to have "air clean" (which I'm going to call "filter") when the Coil-dry/"mold" thing is on. Is clean/filter required to be on if "mold" is on? Or are they completely independent?

sheppy99, 2019-09-09:
"air clean" is what is says on the remote, its an extra level of air cleaning using plasma technology that runs when the unit is on. The Coil Dry function from what I can work out puts the fan on for a timed period at a low speed to dry the internal unit, it may also do something clever with the compressor. Its a manually selected mode that ends with the unit being OFF. They are both independent, I tried all combinations on the sheet to ensure I wasn't looking at the wrong nibble.
On Daikin2 the Mold Guard is a different function which is set once and left alone. It does a similar coil drying function to the Fujitsu Coil Dry whenever the unit is turned OFF and it was operating in one of the cooling modes.
Essentially Air Clean and Coil Dry are independent and coil Dry is also different to Mold Guard on Daikin2

sheppy99, 2019-09-09:
> I'm referring to the "model" setting on the climate/common a/c stuff.
Both remotes are decoded as model 1

crankyoldgit, 2019-09-09:
Thanks.

Okay, I'm going with "Air Clean" as "filter" and "Coil Dry" as "clean" in the common a/c mapping system.

sheppy99, 2019-09-09:
That makes sense although filter in Daikin2 refers to auto filter cleaning? I guess there’s so many different things that to give everything a distinct name is both impractical and a pain to keep track of

crankyoldgit, 2019-09-09:
> AGTV14LAC Floor Console which doesn't have Air Cleaner Mode, and is used with AR-RAC1E remote

Which model is that one detected as? and does it work so far with the library? _e.g. Can I/how should I record it in the supported model section(s)._

sheppy99, 2019-09-09:
Both models are detected as FUJITSU_AC MODEL1. The remotes are sort of interchangeable and look almost the same until you send an air clean command to the floor console which makes it go into a sulk until it’s power cycled as it doesn’t have that option. I’m assuming that’s the command that broke it when I got the controls mixed up after the units moved house.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

crankyoldgit, 2019-09-09:
Hmm .. we should try to avoid that if we can.

Can you set both remotes to exactly the same settings and capture them please? (i.e. their `state[]`s is fine) So I can see if there is some difference that can be detected between the two a/c models.
i.e. So I create a special model that only has the clean/filter setting so we don't cause things to ass-plode!

crankyoldgit, 2019-09-09:
Hey @sheppy99 can you please try out branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/fuji_clean_mold / PR #895 
It hopefully will support clean and filter **IF** you set it to model 5 / ARRY4.

Let me know how it goes.

sheppy99, 2019-09-09:
Good Morning,
I've just checked the manual for the Daikin and also the Fujitsu's so I can hopefully define the terms correctly
**Daikin2 Options**
1)Auto Filter Cleaning - when ON it automatically cleans the units filter - not sure why you'd ever turn this OFF!
2)Mold Proof - when ON it dries the internal unit automatically as the unit is turned OFF
3)Flash Streamer Air Purifying Operation - When ON "Flash Streamer discharge decomposes mold and allergic substances to clean the air in the room"
4)Fresh Air Supply Ventilation - "The air in the room is cleaned, taking in fresh air from outside"
**Fujitsu Options - Floor Console AR-RAC1E Remote**
1)coil dry - "The Indoor unit can be dried by pressing the COIL DRY button
on the Remote Control Unit so as to avoid going moldy
and restrain the breed of bacterium." This is a different function to anything on Daikin2.
With Daikin2 Mold Guard does this automatically when needed whereas Fujitsu coil dry is selected manually instead of pressing the OFF button.
**Fujitsu Options - Floor Console AR-RAC1E Remote**
1)coil dry - same function on both Fujitsu units - see above
2)air clean - "If you wish to eliminate dirt, dust, cigarette smoke, pollen or just simply purify the air of the room, use the air cleaning mode." This is equivalent to Daikin2's Flash Streamer Air Purifying Operation.

sheppy99, 2019-09-09:
Moving on to testing, I've set a separate IRMQTTServer with a receiver next to an IR LED on my main IRMQTTServer, I'll send it a sample of packets and see what the differences are between generated IR and  original remote IR. I'll document this in the sheet on the Fujitsu Tab

sheppy99, 2019-09-10:
Using "model":"ARRY4" in JSON gives no acknowledgement 
Using "model":ARRY4 gives a full JSON acknowledgement containing "model":-1
Using "model":5 gives a full JSON acknowledgement containing "model":5
Sending with model 5 allows correct air cleaning both ON and OFF
With the standard remote nibble 29 is set to 8, whereas MQTTIRServer sets it to 8 - do you know what this nibble does?

crankyoldgit, 2019-09-10:
> Using "model":"ARRY4" in JSON gives no acknowledgement
> Using "model":ARRY4 gives a full JSON acknowledgement containing "model":-1
> Using "model":5 gives a full JSON acknowledgement containing "model":5
> Sending with model 5 allows correct air cleaning both ON and OFF

Excellent. I'll push a fix to the branch in sec.

> With the standard remote nibble 29 is set to 8, whereas MQTTIRServer sets it to 8 - do you know what this nibble does?

According to your spreadsheet, it's "Air Clean".

crankyoldgit, 2019-09-10:
> Excellent. I'll push a fix to the branch in sec.

Pushed. Download the branch and test away.

sheppy99, 2019-09-10:
Thanks, will test shortly.
On a different note will it take long to add full common support to the Mitsubishi Electric Ducted protocol, and would you like me to add an issue for it? The Fujitsu Model 1 and the Ducted are on the same channel so I need to work out how I'm going to get heating this evening :-)

crankyoldgit, 2019-09-10:
> On a different note will it take long to add full common support to the Mitsubishi Electric Ducted protocol, and would you like me to add an issue for it?

How long is a piece of string? :) Yes, please.

> so I need to work out how I'm going to get heating this evening :-)

Hey, I was up to 2am fixing the JSON thing. Give me a second. ;-)

sheppy99, 2019-09-10:
hahahaha, seriously no worries as I can just make a quick rule to spit out raw codes, it'll probably take me all of 10 minutes as I'll need maybe 30 combinations, it'll be quicker than modifying it to use the framework - issue coming up.
"model":"arry4" now produces "model":"5" in the stats, and the output behaves correctly.
I've just noticed some weirdness with the Fujitsu test around nibble 19
This should only turn to 8 when Coil Dry is selected which I haven't done, I'm getting:
Both Original Remotes nibble 19 = 0 which is what I'd expect
FUJITSU_AC MODEL 1 nibble 19 = 1
FUJITSU_AC MODEL 5 nibble 19 = 9
Do you plan to add coil dry to the framework?

sheppy99, 2019-09-10:
I've coded the look up table and its working, so no worries on the framework

crankyoldgit, 2019-09-10:
> I've just noticed some weirdness with the Fujitsu test around nibble 19
> This should only turn to 8 when Coil Dry is selected which I haven't done, I'm getting:
> Both Original Remotes nibble 19 = 0 which is what I'd expect
> IRMQTTServer FUJITSU_AC MODEL 1 nibble 19 = 1
> IRMQTTServer FUJITSU_AC MODEL 5 nibble 19 = 9

Hmm. Not sure what the LSB bit of that nibble is for .. lets take a look at the code. Nibble 19 is the lower half of `remote_state[9]`.

Lines that reference that byte are:
`remote_state[9] = _mode | 0 << 4;  // timer off`
`remote_state[9] |= (_clean << 3)`
`setMode(remote_state[9] & 0b00000111);`
& `getClean()`'s `return remote_state[9] & 0b00001000;`

Only _mode_ and _clean_ appear to have anything to do with that nibble (LSB half of the byte).
The "bottom" three bits (0b0111) are the Operatin Mode. The top one is "Clean". (0b1000)

The difference between 0 (0b0000) & 1 (0b0001) and 8 (0b1000) & 9 (0b1001) is the lowest bit.
e.g. 0b0001. Which happens to be COOL mode.
If you set it to "Auto" mode .. (which is 0b0000) .. you should get you 0 & 8 back in nibble 19.

crankyoldgit, 2019-09-10:
> Do you plan to add coil dry to the framework?

"coil dry" is mapped to "clean".

sheppy99, 2019-09-10:
What’s air clean mapped to?

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

crankyoldgit, 2019-09-10:
> What’s air clean mapped to?

See earlier comment: https://github.com/crankyoldgit/IRremoteESP8266/issues/894#issuecomment-529306242

> Okay, I'm going with "Air Clean" as "filter" and "Coil Dry" as "clean" in the common a/c mapping system.

sheppy99, 2019-09-10:
I've just tested this.
1)nibble 20 is set at 3 when sent from IRMQTTServer and its set at 1 from the original remotes - any idea what that one does? It doesn't seem to affect the operation though.
2)nibble 19 works as expected
3)nibble 17 is set zero when the AC is already ON and 1 when it is turning ON. I discovered this in the past when sending it recorded commands from the original remote. Every successful command that wants the unit to turn ON when its OFF needs to be recorded in that way so I just sent every command as if the AC was previously OFF. The Library does this and now I know why :-)

crankyoldgit, 2019-09-11:
> 1)nibble 20 is set at 3 when sent from IRMQTTServer and its set at 1 from the original remotes - any idea what that one does? It doesn't seem to affect the operation though.

Nibble 20 should be the upper half of `remote_state[10]`.
`remote_state[10]` is used for fanspeed & swing mode. The upper half is the swing mode. i.e. nibble 20.

```
const uint8_t kFujitsuAcSwingOff = 0x00;
const uint8_t kFujitsuAcSwingVert = 0x01;
const uint8_t kFujitsuAcSwingHoriz = 0x02;
const uint8_t kFujitsuAcSwingBoth = 0x03;
```

So `3` is both horiz & vertical swing. If the` ARRY4` doesn't have a horizontal swing, let me know I can make it use `1` instead  i.e. Vertical swing.

> 2)nibble 19 works as expected

Excellent.

> 3)nibble 17 is set zero when the AC is already ON and 1 when it is turning ON. I discovered this in the past when sending it recorded commands from the original remote. Every successful command that [...]

I'm confused with this. Is there a problem I need to fix in this?

sheppy99, 2019-09-11:
>  If the ARRY4 doesn't have a horizontal swing, let me know I can make it use 1 instead i.e. Vertical swing.
Correct, vertical swing only on RAC1E (model 1) and ARRY4 (model5) remotes plus a toggle button that sends a single shortened packet everytime its pressed to make the vanes move one increment.

> I'm confused with this. Is there a problem I need to fix in this?
No fix required, its just an anomaly that I discovered when I originally set it up without the knowledge of your excellent library

crankyoldgit, 2019-09-11:
Okay, I'll make an update for the swing later.

crankyoldgit, 2019-09-11:
Okay. Marking this issue closed for now.

closed 2019-09-11 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

commit 36c11a71 (2019-09-09): [Fujitsu] Add support for ARRY4 remote.
* New model ARRY4 (5)
* Add support for Clean and Filter for that model.
* Unit tests for the above.
* Code style cleanup.

Note: AR-RY4 appears to be the same as AR-RAH2E but has clean & filter 
settings. Thus auto detection won't always get it correct.

Fixes #894