crankyoldgit/IRremoteESP8266 #897: Add common ac framework to MITSUBISHI136 Protocol

sheppy99, 2019-09-10:
The MITSUBISHI136 protocol is tested and working on my Mitsubishi Electric Ducted system, can we add common ac framework support to it please

crankyoldgit, 2019-09-10:
Hey @sheppy99 / Kevin, same deal ... Can you please download and try out https://github.com/crankyoldgit/IRremoteESP8266/tree/Mitsub136Detailed ?

Feedback welcome etc etc.
That branch should have all the latest changes in to too (i.e. ARRY4 etc)

sheppy99, 2019-09-10:
Sweet - tomorrow mornings job!

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

sheppy99, 2019-09-10:
Just testing this now, a couple of things so far.
1)The remote has 4 fan speed settings which are decoded as low,low,medium,high. I'd suggest for compatibility with later units that these are set as the usual min,low,medium,high. My unit treats min and low as the same in any case. My suggested mappings are
1 = min
3 = low
5 = med
7 = high
I also get a minor error when compiling:
`D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:472:6: warning: unused parameter 'str' [-Wunused-parameter]

 void debug(const char *str) {`
just in case its relevant to anything in future.

I'll test it live later together with the Fujitsu revisions

crankyoldgit, 2019-09-11:
> I also get a minor error when compiling:
> `D:\My Documents\Arduino\IRMQTTServer\IRMQTTServer.ino:472:6: warning: unused parameter 'str' [-Wunused-parameter]
> 
> void debug(const char *str) {`
> just in case its relevant to anything in future.

Nope. That's just because debug messages are now turned off by default to save code space.

> 1)The remote has 4 fan speed settings which are decoded as low,low,medium,high. I'd suggest for compatibility with later units that these are set as the usual min,low,medium,high. My unit treats min and low as the same in any case. My suggested mappings are
1 = min
3 = low
5 = med
7 = high

Currently I have four levels of fan speed (The numbers don't match your numbering, don't worry about that)
```
const uint8_t kMitsubishi136FanMin =          0b00;
const uint8_t kMitsubishi136FanLow =          0b01;
const uint8_t kMitsubishi136FanMed =          0b10;
const uint8_t kMitsubishi136FanMax =          0b11;
const uint8_t kMitsubishi136FanQuiet = kMitsubishi136FanMin;
```

As far as I can tell from the code it should be being decoded as "min,low,medium,max", not "low,low,medium,high".

Can you please go into more detail over what exactly is reporting the "low" message? IRMQTTServer, or IRrecvDumpV2? etc. And what the messages/states are that are causing the incorrect reading?

> A live test works, do you need me to do a byte by byte confirmation against the remote or shall we call it done? I'm using it live so any problems should become obvious when I use the heating this evening

We can call it "done" for now and open a new issue etc. Obviously I want things to be as correct as possible, so if you want to a byte by byte comparison, I will welcome it. Your call.

sheppy99, 2019-09-11:
My setup is IRMQTTServer > IR LED > IR Receiver on another IRMQTTServer > MQTT.fx so I'm reading the result from what another copy of IRMQTTServer publishes.
HOWEVER ITS WORKING CORRECTLY NOW - go figure

crankyoldgit, 2019-09-11:
> Would you like me to open an issue for the multi channel common ac framework challenge?

Sure. Why not. I'll merge & close out this PR and issue. Then I'll try to merge the other branch to the new "master", by that time, you should have created the issue. ;-)

closed 2019-09-11 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

commit de0a9302 (2019-09-10): Mitsubishi136: Add detailed a/c support.
* Power, Temp, Mode, Fan, SwingV & Quiet
* Add common a/c support.
* Unit tests for the above.
* Minor code cleanup in Mitsubishi too.

Fixes #897
PR #898 (2019-09-11): Mitsubishi136: Add detailed a/c support.
* Power, Temp, Mode, Fan, SwingV & Quiet
* Add common a/c support.
* Unit tests for the above.
* Minor code cleanup in Mitsubishi too.

Fixes #897
commit 6158e7f3 (2019-09-11): Mitsubishi136: Add detailed a/c support. (#898)

* Power, Temp, Mode, Fan, SwingV & Quiet
* Add common a/c support.
* Unit tests for the above.
* Minor code cleanup in Mitsubishi too.

Fixes #897