crankyoldgit/IRremoteESP8266 #899: IR LED doesn't light up if state.protocol start from DAIKIN or COOLIX

k2biru, 2019-09-10:
### Version/revision of the library used
master as at 10 Sept, 2019.

### Expected behavior
* Trying `CommonAcControl.ino` 

### Actual behavior
* IR LED doesn't light up (when `ac.sendAc(state, &prev);`),  if `state.protocol` start from DAIKIN or COOLIX 
* If `state.protocol` start form KELVINATOR, IR LED light up  even though the `state.protocol` is DAIKIN or COOLIX

### Steps to reproduce the behavior
* Using `CommonAcControl.ino` example, change (line 20) `kIrLed = 14`

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Other useful information
* Using generic ESP8266 1M [TYWE3S](https://fccid.io/2AFNL-TYWE3S/User-Manual/Users-Manual-3525098)
sorry for my bad english

crankyoldgit, 2019-09-11:
> * IR LED doesn't light up (when `ac.sendAc(state, &prev);`),  if `state.protocol` start from DAIKIN or COOLIX
> * If `state.protocol` start form KELVINATOR, IR LED light up  even though the `state.protocol` is DAIKIN or COOLIX

I am confused by what you are saying here. Can you please try to describe it another way? Or maybe include the Serial Console output and add comments to it explain is or isn't going on?

> sorry for my bad english

No need to apologise. Your "bad" English is better than my "Non existent" _your_ native language. :)

k2biru, 2019-09-11:
I mean, on example `CommonAcControl.ino` and run it. 
The first and second AC protocol (DAIKIN and COOLIX) doesn't light up IR LED. But after the third AC protocol (KELVINATOR), IR LED light up as expected.

Or, try this code. 
* This code deesn't light up IR LED (GPIO14).
 
```
/* Copyright 2019 David Conran
*
* This example code demonstrates how to use the "Common" IRac class to control
* various air conditions. The IRac class does not support all the features
* for every protocol. Some have more detailed support that what the "Common"
* interface offers, and some only have a limited subset of the "Common" options.
*
* This example code will:
* o Try to turn on, then off every fully supported A/C protocol we know of.
* o It will try to put the A/C unit into Cooling mode at 25C, with a medium
*   fan speed, and no fan swinging.
* Note: Some protocols support multiple models, only the first model is tried.
*
*/
#include <Arduino.h>
#include <IRremoteESP8266.h>
#include <IRac.h>
#include <IRutils.h>

const uint16_t kIrLed = 14;
IRac ac(kIrLed); 
stdAc::state_t state; 
stdAc::state_t prev; 

void setup() {
  Serial.begin(115200);
  delay(200);

  state.protocol = decode_type_t::DAIKIN;  // Set a protocol to use.
  state.model = 1;  // Some A/C's have different models. Let's try using just 1.
  state.mode = stdAc::opmode_t::kCool;  // Run in cool mode initially.
  state.celsius = true;  // Use Celsius for units of temp. False = Fahrenheit
  state.degrees = 25;  // 25 degrees.
  state.fanspeed = stdAc::fanspeed_t::kMedium;  // Start with the fan at medium.
  state.swingv = stdAc::swingv_t::kOff;  // Don't swing the fan up or down.
  state.swingh = stdAc::swingh_t::kOff;  // Don't swing the fan left or right.
  state.light = false;  // Turn off any LED/Lights/Display that we can.
  state.beep = false;  // Turn off any beep from the A/C if we can.
  state.econo = false;  // Turn off any economy modes if we can.
  state.filter = false;  // Turn off any Ion/Mold/Health filters if we can.
  state.turbo = false;  // Don't use any turbo/powerful/etc modes.
  state.quiet = false;  // Don't use any quiet/silent/etc modes.
  state.sleep = -1;  // Don't set any sleep time or modes.
  state.clean = false;  // Turn off any Cleaning options if we can.
  state.clock = -1;  // Don't set any current time if we can avoid it.
  state.power = false;  // Initially start with the unit off.

  prev = state;
}

void loop() {
    decode_type_t protocol = decode_type_t::DAIKIN;
    if (ac.isProtocolSupported(protocol)) {
      state.protocol = protocol;
      Serial.println("Protocol " + String(protocol) + " / " +
                     typeToString(protocol));
      state.power = true;
      ac.sendAc(state, &prev);
      Serial.println("Sent a message to turn ON the A/C unit.");
      prev = state;
      delay(1000);
    }
    decode_type_t protocol = decode_type_t::COOLIX;
    if (ac.isProtocolSupported(protocol)) {
      state.protocol = protocol;
      Serial.println("Protocol " + String(protocol) + " / " +
                     typeToString(protocol));
      state.power = true;
      ac.sendAc(state, &prev);
      Serial.println("Sent a message to turn ON the A/C unit.");
      prev = state;
      delay(1000);
    }
}

```
Thanks

crankyoldgit, 2019-09-11:
Thanks. What is the Serial output of your modified code?

k2biru, 2019-09-11:
> Thanks. What is the Serial output of your modified code?
Here my Serial output
``` 
␀rl␀l��|␀�l�|␂␌␌␌�␌l�␌b|��␂�␒�r�bl␌b��nn�lnn���␌b␜p�lb␎lrlp�n�␐␂␌␌�␌l␌��␌␌␌b␌n�|␂l�␌␌�b��nn�␀l��l`␂�␒␒nn␌l`␂␎␂nr���n␌␌b␌�␎l␎r��n␌␌b␌�␎l�␂�␂ll�␒�l`␂��n�␂Protocol 16 / DAIKIN
Sent a message to turn ON the A/C unit.
Protocol 15 / COOLIX
Sent a message to turn ON the A/C unit.
Protocol 18 / KELVINATOR
Sent a message to turn ON the A/C unit.
Protocol 15 / COOLIX
Sent a message to turn ON the A/C unit.
Protocol 16 / DAIKIN
Sent a message to turn ON the A/C unit.
Protocol 15 / COOLIX
Sent a message to turn ON the A/C unit.
```

crankyoldgit, 2019-09-11:
Thanks.

If I recall correctly, it (`ac.sendAc()`) doesn't notice a difference in the states if **only** the protocol changes. To force it to change, I suggest you do as the original example does, i.e. transition from "off" to "on" and then "on" to "off". Power status is definitely something that is detected as a change in `sendAc()`

k2biru, 2019-09-11:
Thanks.
> If I recall correctly, it (ac.sendAc()) doesn't notice a difference in the states if only the protocol changes. To force it to change, I suggest you do as the original example does, i.e. transition [...]

 if like that. When i using the example provided `CommonAcControl.ino`, the first and second AC protocol (DAIKIN and COOLIX) doesn't light up IR LED.

https://github.com/crankyoldgit/IRremoteESP8266/issues/899#issuecomment-530204546
> I mean, on example `CommonAcControl.ino` and run it.
The first and second AC protocol (DAIKIN and COOLIX) doesn't light up IR LED. But after the third AC protocol (KELVINATOR), IR LED light up as expected.

crankyoldgit, 2019-09-11:
Okay. I'll take a deeper look shortly.

k2biru, 2019-09-11:
I found something interesting.
If i add `pinMode(kIrLed,OUTPUT);` on `setup()`. The problem gone. 

Should i add `pinMode();` **manually**, or the library already add it automatically.

crankyoldgit, 2019-09-11:
Well spotted/worked out!! That's a bug. The library should do that for you.

If you download the code in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue899 / PR #905 and compile using that, it should do that for you now.

Please let me if it fixes/doesn't fix the problem!

closed 2019-09-11 by crankyoldgit

k2biru, 2019-09-11:
I think you forget to add `ac->begin();` for AMCOR protocol
[src/IRac.cpp:154](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L154)

crankyoldgit, 2019-09-11:
@k2biru Thanks. So I did.  Fixed.

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

commit 6ba94bec (2019-09-11): [Bug] Ensure `begin()` is called for every supported common a/c. (#905)

Some protocols didn't call `begin()`. Fixed.
Moved all the calls to inside their respective procedures.

Fixes #899
commit d6750b94 (2019-09-11): Add missing `begin()` for Amcor().

Ref #905 
Fixes #899