crankyoldgit/IRremoteESP8266 #908: Incorrect functions In common ac framework

sheppy99, 2019-09-12:
I've tested the common ac framework against my Daikin2 Remote and looking at the MQTT JSON output on `ir_server/ac/stat/json` It works the AC, but a few functions are incorrect.
I've put the results in the google sheet on a new tab called "framework." For each error I've added a set of strings to the "test sheet" tab selected function (column A), byte/nibble controlling it (column B) the correct value (column C) and the full string of the message (column D)

crankyoldgit, 2019-09-12:
For the record, that's: https://docs.google.com/spreadsheets/d/1f8EGfIbBUo2B-CzUFdrgKQprWakoYNKM80IKZN4KXQE/edit#gid=1822437752

crankyoldgit, 2019-09-12:
Any chance you could stack/format the correct and incorrect codes above each other? Makes it easier for me to see the difference.
e.g.

Correct | Comment | Byte | Correct Value | raw
-- | -- | -- | -- | --
Correct |Auto | 28L | A | 53,11DA2700014942B0280C8004B016240000AAC35D11DA270000093C00**A**0000006600000C190600E
Wrong|  |  |  | 53,11DA2700014A42B0280C8004B016240000AAC35E11DA270000093C00**B**0000006600000C190601E

sheppy99, 2019-09-12:
Will do in future, I was doing it from a decoding perspective, do you need me to add them for any of the current strings  or just move to that going forward?

crankyoldgit, 2019-09-12:
I was waiting till you told me you found all the errors etc before I started in earnest

sheppy99, 2019-09-12:
That's all of them, there's only the intelligent eye which IIRC we removed, and the default is off which is how I run it. At some stage I need to see if there is a solution to it moving the vanes when it switches from cooling to heating, in the middle of the night as the noise wakes me up, but it does that when the relevant IR bits which aren't changed between modes.

crankyoldgit, 2019-09-12:
Blah blah blah ... https://github.com/crankyoldgit/IRremoteESP8266/tree/daikin2_changes

Doesn't contain the multi stuff ... blah blah blah.

crankyoldgit, 2019-09-13:
> The question is more about whether I can send a packet and have it send and decode it at the same time. Has the ESP8266 got enough power to do that?

Can it do it, yes. There is an option in the to enable it. The default is it stops listening for the brief time it's sending. e.g. ~1/10th of second or so. The better questions is "Should you?". It depends if the receiver module can see the TX IR LEDs and if you have it configured to re-transmit incoming IR a/c messages.

crankyoldgit, 2019-09-15:
In all the other issue traffic, I'm not sure if you tested this or not, and if so, what the feedback was?

sheppy99, 2019-09-15:
I haven’t yet, although I have just recompiled it to 7 channels and had a fight getting the framework to work again. This time around it was ac_3 playing up. Multiple changing of GPIO’s around and several clears of retained messages finally got it all to work so DAIKIN2 is next for testing. Is it a lot of work to add a “delete all retained messages and reboot” option?

crankyoldgit, 2019-09-15:
Log a feature request. :)

sheppy99, 2019-09-15:
done! This will be fantastic when its finished!!!

crankyoldgit, 2019-09-15:
I must admit, I'm concerned what might be causing the corruption.
As you've got it (multi) all working. I might back out your "personal" settings and merge the "multi" branch to master, and then you can have the dakin2 stuff and multi at the same time.

sheppy99, 2019-09-15:
My guess is its down to swapping channels around with different AC's. and possibly that the broker hasn't cleared all the messages before its read them again. If it does it again I'll unplug IRMQTTServer, clear the messages and then plug it in again. Its downstairs so I haven't done that yet.
I've now got a spare channel pointing at a receiver so I can do the Daikin2 test easier.
I'll head out to Jaycar now if you want to do the merge and I'll change my settings on compile when I get back if you want to merge it

crankyoldgit, 2019-09-15:
Branch https://github.com/crankyoldgit/IRremoteESP8266/tree/daikin2_changes has been updated to the current `master` branch (i.e. the Multi stuff)

sheppy99, 2019-09-15:
I'm wondering if I've got the correct branch - take a look at the Fan tests and Swing Vertical on the spreadsheet. I used the examples from the sheet and sent the full hex string to the receiver and its decoding isn't correct.

crankyoldgit, 2019-09-15:
I don't think I saw the "vertical" stuff before,  or if I did, I missed it. Sorry.

Also, to be fair. I'm very confused by the structure of the "Test Sheet". Especially Cols A-D
If you could be clear or verbose as to what the issues are, that would be grand.

sheppy99, 2019-09-15:
Sure, can you tell me what states are available in the framework for :
swingv and fanspeed please

crankyoldgit, 2019-09-15:
See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRsend.h#L44-L113

crankyoldgit, 2019-09-15:
Another update to the branch. Most of the feedback/comments are in the spreadsheet now.
I've added unit tests to confirm my theories/comments in the spreadsheet's comments.
e.g. I think you've recorded something incorrectly for the anomalous fan speed setting. i.e. It reads as `medium` to me, not `auto`.

sheppy99, 2019-09-15:
All tests good now apart from swingv which is OK apart from setting swing or auto on the remote both decoding as auto. Seem to remember you were coding auto on the remote as off in framework?
auto on the remote means it does whatever it thinks is the best for its current mode. As the flap motors on my unit make more noise than the fan I don't use either of those modes

crankyoldgit, 2019-09-15:
> Seem to remember you were coding auto on the remote as off in framework?

Yep. I missed part of what was required to change it. My bad.

I just updated the branch and it should fix that as of now. Let me know how it goes.
(also the branch now has the option to enable clearing of the retained settings, you will need to enable that at compile time for it to show up.)

sheppy99, 2019-09-15:
Clear messages works, but the vertical swing is proving a bit tricky to tie down. positions 1-6 work and both swing and auto on the remote come up as auto in the framework BUT after reading the manual it seems there could be more to it. Extra modes "Breeze" "Circulate" and "Auto" have other options hidden away in the setup menu to do with the magic eye. I'll try and tie it down later this week

crankyoldgit, 2019-09-16:
I've changed the swingv values for auto & swing to what I think they should be.
Download and try it out please.
 "Breeze" "Circulate" are not going to be supported under common. If they are received via IR, they will be interpreted as "native" `auto`, which per your request, is mapped to "common"s `off`.

closed 2019-09-16 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

commit 8b00cc79 (2019-09-16): Daikin2: Corrections for common a/c interface. (#910)

* Update SwingH common support.
  -  Allow finer grained control over horizontal swing.
* Add SwingV swing option
* Unit tests to verify/confirm the findings & assumptions of @sheppy99

Fixes #908