crankyoldgit/IRremoteESP8266 #911: Tested Fujitsu AR-DL10 remote : works as ARDB1 for A/C ASU30C1

tavdog, 2019-09-13:
Power on and off functions work.  Mode and fan speeds work.  Swing function does not work.
Feel free to close, just wanted to add my test result.

crankyoldgit, 2019-09-13:
Thanks for the feedback.
Can you capture the commands via `IRrecvDumpV2` at all? If so, we may be able to add support for that model.
Also, does it have horizontal swing at all?

crankyoldgit, 2019-09-13:
FYI, you might want to try `ARJW2` (4) as it is similar.

tavdog, 2019-09-13:
Yes they do have horizontal swing.  I don't have an IR rx module so I can't get a dump at the moment.  I'll try the ARJW2 and post my report.

crankyoldgit, 2019-09-13:
FYI ... https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Fujitsu.h#L95
`ARJW2,        // (4) AR-JW2  (Same as ARDB1 but with horiz control)`

crankyoldgit, 2019-09-14:
> I'll try the ARJW2 and post my report.

@tavdog How did it go?

tavdog, 2019-09-14:
Tried with ARJW2. Everything works but the swing control.

closed 2019-09-14 by tavdog

crankyoldgit, 2019-09-14:
Thanks for the update. If you get a IR receiver module, let us know and we can try to add the swing control for your model.