crankyoldgit/IRremoteESP8266 #912: Inconsistent string variable definition

k2biru, 2019-09-13:
### Version/revision of the library used
'master' as at 13st of Sept, 2019.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
Yes

### Information
* On file `IRac.cpp` , i found some functions like `stdAc::opmode_t IRac::strToOpmode(const char *str, const stdAc::opmode_t def) ` that convert string to enum. [IRac.cpp:1404](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L1404)
* On file `IRac.cpp` , also i found some functions like `String IRac::opmodeToString(const stdAc::opmode_t mode)` that convert enum back to string. [IRac.cpp:1562](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRac.cpp#L1562)
* When i use `strToOpmode()` return value is enum, but when i turn it back to string it turnt into lowercase one.
* I mean someting like this

code :
```
const char* Mode = "COOL";
Serial.println(Mode);
opmode_t mode_t = IRac::strToOpmode(Mode); 
Serial.println( IRac::opmodeToString(mode_t);
```
Serial output
```
COOL
cool
```
* Right now i make project that require save last IRac state into SPIFFS on json file.
   - json file : 

```
{"protocol":"DAIKIN160","model":"1","power":"1","mode":"cool","celsius":"0","degress":"17.00","fan":"auto","swingv":"auto","swingh":"off","quiet":"0","turbo":"0","econo":"0","light":"0","filter":"0","clean":"0","beep":"0","sleep":"0","clock":"0"}
```
* When i set AC state, i save it i using  xxxToString() function.
* Because of that Inconsistent naming, when i read back the json file. AC state doesn't look same.
* Can you you make string variable definition consistent? or
* Is there any best way to save state to json ?

Thank you

crankyoldgit, 2019-09-13:
There is no reasonable way the result of `IRac::strToXyz()` can remember the original string passed to it. e.g. `IRac::strToOpmode("CoOl");` How could it remember the capitalisation you used? You want the string capitalisation to be remembered or formatted in a certain way, then you'll will have to do that in your program.

If you want the results of `IRac::XyzToString(variable);` to be uppercase, as it returns a `String` you can just use: `IRac::XyzToString(variable).toUpperCase();` (See: https://www.arduino.cc/reference/en/language/variables/data-types/string/functions/touppercase/)

There could be an argument for/against the protocol ones which return upper case strings, but that's that way to be visually consistent with the `enum decode_type_t` usage, which is case sensitive.

> Can you you make string variable definition consistent? or

I'm not sure what you mean by this.

> Is there any best way to save state to json ?

Serialisation (https://en.wikipedia.org/wiki/Serialization) is a challenge. If you want to serialise it to json, then **you're** the one making up the format and structure to serialise and deserialise the `state_t` structures. There is often no "best way" per se. This is a problem almost as old as Computer Science itself.

If you want to see how I've done it, look in the IRMQTTServer code in the `sendJsonState()` procedure. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L2644

k2biru, 2019-09-17:
Thanks for your attention. i have bad weekend, sorry for late update.
> If you want the results of IRac::XyzToString(variable); to be uppercase, as it returns a String you can just use: IRac::XyzToString(variable).toUpperCase();

Yes, I already do it. 
> Serialisation (https://en.wikipedia.org/wiki/Serialization) is a challenge. If you want to serialise it to json, then you're the one making up the format and structure to serialise and deserialise [...]

>If you want to see how I've done it, look in the IRMQTTServer code in the sendJsonState() procedure.

Thanks for this suggestion. I really appreciate it.

crankyoldgit, 2019-09-17:
Is there anything else unanswered/undone in this issue? If not, can I close it?

k2biru, 2019-09-17:
Sure, Close it. Thanks

closed 2019-09-17 by k2biru