crankyoldgit/IRremoteESP8266 #915: How to show received IR signal data at receiving side  in serial console ?

happytm, 2019-09-14:
### Version/revision of the library used
2.6.3

I am:
1. Using IRsmartrepeater example sketch on receiving side.
2. Sending side I am using irsend to broadcast Midea protocol 48 bit format.
3. IR data sent is 01:10:20:30:40:50.

### Actual behavior
Following is serial console log of receiving (repeater) device:

010056.883: A 48-bit MIDEA message was successfully retransmitted.
010077.407: A 48-bit MIDEA message was successfully retransmitted.
010097.897: A 48-bit MIDEA message was successfully retransmitted.

#### Circuit diagram and hardware used (if applicable)
ESP8266 Wemos d1 mini on both side. Sender IR emitter pin is 16 and on receiver detector pin is 2 and IR emitter pin is 16.

### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)

Yes.

My question is how to print this data received which is human readable in decimal format in my serial console.

Thanks

crankyoldgit, 2019-09-15:
> ### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
> Yes.

Some how I don't think you did. See: [I want to change the example code to make it do something different, but I don't know how. Please help me.](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#I_want_to_change_the_example_code_to_make_it_do_something_different_but_I_dont_know_how_Please_help_me)

I've even pointed this out to you earlier here: https://github.com/crankyoldgit/IRremoteESP8266/issues/887#issuecomment-528763042

I even suggested the code to you earlier in: https://github.com/crankyoldgit/IRremoteESP8266/issues/887#issuecomment-528745806

You need _something_ like:

```
  if (irrecv.decode(&results)) {
    if (result.decode_type == MIDEA) {
      serialPrintUint64(results.value, HEX);  // Should be your 48bit MAC Address.
      Serial.println("");
    }
    irrecv.resume();  // Receive the next value
  }
```

You should be able to figure out how & where to incorporate that sort of thing into the code.

Also, as pointed out fairly prominently in the [source code](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/SmartIRRepeater/SmartIRRepeater.ino#L18-L22) of that example file:

```
 *                               W A R N I N G
 *   This code is just for educational/example use only. No help will be given
 *   to you to make it do something else, or to make it work with some
 *   weird device or circuit, or to make it more usable or practical.
 *   If it works for you. Great. If not, Congratulations on changing/fixing it.
```

If you want the result in decimal rather than hex. Per the [function description](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRutils.cpp#L38-L47), use:
```
      serialPrintUint64(results.value, DEC);  // Should be your 48bit MAC Address in Decimal (Base 10).
```

closed 2019-09-15 by crankyoldgit

happytm, 2019-09-15:
Works like a charm.

Thanks.