crankyoldgit/IRremoteESP8266 #916: Add clear all retained messages and reboot to IRMQTTServer

sheppy99, 2019-09-15:
I've had a few issues where odd channels of IRMQTTServer Multi wouldn't work with the common ac framework. The fix seems to be deleting all retained messages and rebooting.
Adding a "clear all retained messages and reboot" function would help with this, although perhaps clearing all retained messages whenever GPIO's are changed would also be a good option?
As I'm still testing IRMQTTServer I'd advise caution on whether this is the full fix to the problem as each time I've had the problem I've ended up doing a lot of GPIO changing, retained messages clearing and testing before I got full functionality.

crankyoldgit, 2019-09-15:
FYI https://github.com/crankyoldgit/IRremoteESP8266/tree/clear_mqtt

closed 2019-09-15 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

PR #917 (2019-09-15): IRMQTTServer: Add option to clear retained settings.
* Add option to clear retained settings. (Under the "admin" page)
* Add HA discovery and LWT to things to be cleared as well.

Fixes #916
commit b8cd5840 (2019-09-15): IRMQTTServer: Add option to clear retained settings. (#917)

* Add HA discovery and LWT to things to be cleared.
* Add `MQTT_CLEAR_ENABLE` define to control if this gets compiled in or 
not.
  - It's disabled by default as most people shouldn't need this.
* Clean up the text for this option a little.

Note: If enabled, it costs about 1.1k of flash space.

Fixes #916