crankyoldgit/IRremoteESP8266 #918: mode when AC is Off

sheppy99, 2019-09-16:
The common ac framework shows the mode as "off" when the AC is turned Off, is this by design?
Looking at my remotes, 
FUJITSU_AC has a shorter packet when its turned off, I've not fully decoded it so I don't know if that packet includes the mode and what its set to.
Mitsubishi Electric has mode set at its last value when its turned off
DAIKIN2 has mode set at its last value when its turned off
Off is sent to the mode topics of ` ir_server/ac/stat/mode `and `ir_server/ac/stat/json` and its reverted by the framework when the ac is turned on again. Off is stored as the value in the mode retained topic, so perhaps this will be a problem across reboots?

crankyoldgit, 2019-09-16:
> The common ac framework shows the mode as "off" when the AC is turned Off, is this by design?
> Off is sent to the mode topics of ir_server/ac/stat/modeand ir_server/ac/stat/json and its reverted by the framework when the ac is turned on again.

Yes, this is on purpose.

See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/examples/IRMQTTServer/IRMQTTServer.ino#L2797

It's need to work with Home Assistant. It works best when there is an operation mode for "off".

> Off is stored as the value in the mode retained topic, so perhaps this will be a problem across reboots?

It shouldn't be a problem, hasn't been a problem so far, and no one has reported a problem with it yet. HA, when instructing an mqtt climate ([per it's protocol](https://www.home-assistant.io/components/climate.mqtt/)) will set the power to on & the mode to whatever the user has requested. Hence it hasn't been a problem.

> FUJITSU_AC has a shorter packet when its turned off, I've not fully decoded it so I don't know if that packet includes the mode and what its set to.

The `FUJITSU_AC` protocol sends a _special_ "short" message for a number of actions. The "off" message for example does not (as far as I am aware) contain mode information in it. Hence it is a non-issue, AFAIUI. Fujitsu, while not being unique in this approach, are certainly odd-ball'ish in this approach.
Most other A/C protocols just have the full state message control everything.
The library tries to do the right/safest thing at all times. Trust it, ... until it lets you down. :)

> Mitsubishi Electric has mode set at its last value when its turned off

Yep, that's normal and most typical of the majority of A/C protocols. There is no "off" mode as such on the device, nor does the library try to send it, and if it could, the device wouldn't accept it.

crankyoldgit, 2019-09-16:
There. I've added some documentation, and made it an easy setting for you to disable if you need/want to.

sheppy99, 2019-09-16:
I thought there might be an explanation. I’ve mapped the mode to an item and it wasn’t always showing the mode after turning the AC on. I’ll work around it, thanks for your help.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

sheppy99, 2019-09-16:
Wow, all this in less time than it took me to drive into town. Thanks very much!
I’ll take a look in the morning.

<sub>Sent with <a href="http://githawk.com">GitHawk</a></sub>

sheppy99, 2019-09-16:
I've just tested this and `ir_server/ac_x/stat/json` keeps its mode, but `ir/server/ac_x/stat/mode` still changes to off

crankyoldgit, 2019-09-17:
I _think_ I've fixed it. Untested. Please download and try it out.

sheppy99, 2019-09-17:
That fixed it, thanks. Regarding the multi stuff do you want me to do any form of testing apart from using it day to day and reporting problems?

crankyoldgit, 2019-09-17:
Excellent.

Re: Multi.

It's up to you. I think it's working (I wouldn't have merged it if I knew it had problems). So "use and see" seems appropriate to me.

commit ad960cd8 (2019-09-17): More changes for MQTT_CLIMATE_HA_MODE

Fixes #918
PR #919 (2019-09-17): IRMQTTServer: Add flag & documentation for Home Assistant mode.
* `MQTT_CLIMATE_HA_MODE` setting controls if we use HA mode. Default: 
Yes, we do!

Fixes #918
commit 9c24e083 (2019-09-17): IRMQTTServer: Add flag & documentation for Home Assistant mode. (#919)

* `MQTT_CLIMATE_HA_MODE` setting controls if we use HA mode. Default:  Yes, we do!

Fixes #918
closed 2019-09-17 by crankyoldgit