crankyoldgit/IRremoteESP8266 #921: Panasonic AC model 6, swing V

hpirila, 2019-09-17:
### Version/revision of the library used
2.6.5, IRMQTTServer

### Expected behavior
Send and receive Panasonic AC model 6 Swing V HIGH, MIDDLE AND LOW should work but they are not working. Only HIGHEST, LOWEST and AUTO are working.

### Actual behavior
The AC Swing goes to HIGHEST position when I choose HIGH or MIDDLE and it goes to LOWEST position when I choose LOW. 

#### Output of raw data from IRrecvDumpV2.ino (if applicable)

```
In log below I cycle SWING-V in this sequence using Panasonic AC original remote:
AUTO -> HIGHEST -> HIGH -> MIDDLE -> LOW -> LOWEST
Only the AUTO, HIGHEST and LOWEST are detected (Rx) by IRremoteESP8266 v.2.6.5
Same applies to sending (Tx), only  AUTO, HIGHEST AND LOWEST are working.
Other settings below are T 25 C, fan MAX, mode COOL, model 6, Protocol Panasonic AC
Below is MQTT log, my comments marked with //

-----------------------
2019-09-17 20:05:43
Topic: mh1ir/received
Qos: 0
49,0220E004000000060220E004003932807F00000EE00000890000E7
2019-09-17 20:05:43
Topic: mh1ir/ac/stat/swingv
Qos: 0
auto  // THIS IS CORRECT

2019-09-17 20:05:46
Topic: mh1ir/received
Qos: 0
49,0220E004000000060220E004003932807100000EE00000890000D9
2019-09-17 20:05:46
Topic: mh1ir/ac/stat/swingv
Qos: 0
highest // THIS IS CORRECT

2019-09-17 20:05:50
Topic: mh1ir/received
Qos: 0
49,0220E004000000060220E004003932807200000EE00000890000DA
2019-09-17 20:05:50
Topic: mh1ir/ac/stat/swingv
Qos: 0
auto // THIS IS NOT CORRECT, SHOULD BE HIGH

2019-09-17 20:05:52
Topic: mh1ir/received
Qos: 0
49,0220E004000000060220E004003932807300000EE00000890000DB
// THIS IS NOT DETECTED, SHOULD BE MIDDLE

2019-09-17 20:05:55
Topic: mh1ir/received
Qos: 0
49,0220E004000000060220E004003932807400000EE00000890000DC
// THIS IS NOT DETECTED, SHOULD BE LOW

2019-09-17 20:05:57
Topic: mh1ir/received
Qos: 0
49,0220E004000000060220E004003932807500000EE00000890000DD
2019-09-17 20:05:57
Topic: mh1ir/ac/stat/swingv
Qos: 0
lowest // THIS IS CORRECT
```
### I have followed the steps in the [Troubleshooting Guide](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Troubleshooting-Guide) & read the [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions)
_Yes_

### Has this library/code previously worked as expected for you?
_No._

crankyoldgit, 2019-09-17:
Can you please capture and annotate the same data but using the `IRrecvDumpV2` program?
It will give me what the library is really seeing, rather than just what `IRMQTTServer` is giving you.

crankyoldgit, 2019-09-17:
Looking at the code, we only have three values that the library recognises for Vertical Swing with Panasonic:
```
const uint8_t kPanasonicAcSwingVAuto = 0xF;
const uint8_t kPanasonicAcSwingVUp = 0x1;
const uint8_t kPanasonicAcSwingVDown = 0x5;
```

I can probably guess that values 2-4 are the ones in between "Up" and "Down". Is "Up" fully up? and is "Down" fully down? or are they "Move the vent up a bit" and "down a bit" respectively?

The data/info you collect from `IRrecvDumpV2` will help with that.

crankyoldgit, 2019-09-17:
FYI, when the library doesn't recognise the value, it tends to default to `Auto` hence the behaviour you're seeing.

hpirila, 2019-09-17:
Requested log below. I would say this AC does not have Up / Down control for Swing-V. It has only these 5 fixed positions and Auto, also no Off. Actually when I look the log below, the Fan also not perfect. This AC has 5 fixed fan speeds and Auto. I will look this more tomorrow.

IRrecvDumpV2 dump of same sequence as above:
```
Timestamp : 000694.111
Library   : v2.6.5

Encoding  : PANASONIC_AC
Code      : 0220E004000000060220E004003932807F00000EE00000890000E7 (216 bits)
Mesg Desc.: Model: 6 (RKR), Power: On, Mode: 3 (COOL), Temp: 25C, Fan: 4 (High), Swing (Vertical): 15 (AUTO), Swing (Horizontal): 0 (UNKNOWN), Quiet: Off, Powerful: Off, Clock: 00:00, On Timer: Off, Off Timer: Off
Raw Timing[439]:
   +  3542, -  1684,    +   450, -   418,    +   444, -  1288,    +   476, -   416, 
   +   450, -   416,    +   452, -   376,    +   490, -   366,    +   500, -   366, 
   +   498, -   368,    +   496, -   378,    +   490, -   368,    +   524, -   416, 
   +   450, -   416,    +   450, -   378,    +   488, -  1258,    +   472, -   420, 
   +   452, -   362,    +   502, -   296,    +   570, -   416,    +   450, -   416, 
   +   450, -   416,    +   452, -   414,    +   452, -  1256,    +   472, -  1260, 
   +   448, -  1286,    +   474, -   418,    +   476, -   416,    +   450, -  1256, 
   +   476, -   380,    +   488, -   378,    +   488, -   372,    +   492, -   386, 
   +   482, -   416,    +   450, -   376,    +   492, -   370,    +   496, -   374, 
   +   488, -   416,    +   450, -   380,    +   484, -   416,    +   450, -   416, 
   +   452, -   372,    +   516, -   418,    +   448, -   418,    +   452, -   372, 
   +   492, -   414,    +   452, -   414,    +   450, -   416,    +   454, -   362, 
   +   500, -   416,    +   452, -   366,    +   498, -   382,    +   484, -   416, 
   +   450, -   416,    +   448, -   418,    +   450, -   416,    +   476, -   380, 
   +   488, -   414,    +   452, -   414,    +   450, -  1256,    +   474, -  1258, 
   +   474, -   390,    +   476, -   388,    +   480, -   416,    +   450, -   372, 
   +   492, -   384,    +   482, -  9952,    +  3568, -  1660,    +   474, -   418, 
   +   422, -  1284,    +   502, -   418,    +   450, -   380,    +   488, -   414, 
   +   448, -   418,    +   448, -   390,    +   474, -   420,    +   452, -   374, 
   +   460, -   422,    +   470, -   392,    +   478, -   418,    +   446, -   420, 
   +   448, -  1258,    +   476, -   416,    +   450, -   378,    +   514, -   384, 
   +   480, -   418,    +   420, -   446,    +   450, -   378,    +   458, -   420, 
   +   442, -  1288,    +   474, -  1260,    +   472, -  1262,    +   444, -   446, 
   +   448, -   418,    +   422, -  1286,    +   472, -   420,    +   448, -   416, 
   +   448, -   420,    +   448, -   418,    +   450, -   416,    +   474, -   392, 
   +   448, -   446,    +   420, -   420,    +   472, -   392,    +   472, -   392, 
   +   448, -   422,    +   442, -   426,    +   442, -   424,    +   442, -  1288, 
   +   444, -   448,    +   420, -   420,    +   470, -  1262,    +   444, -  1290, 
   +   472, -  1260,    +   446, -   446,    +   448, -   422,    +   440, -   424, 
   +   444, -  1288,    +   472, -   392,    +   474, -   420,    +   420, -  1288, 
   +   442, -  1288,    +   444, -   424,    +   440, -   426,    +   468, -   398, 
   +   440, -   424,    +   470, -   396,    +   444, -   424,    +   440, -   426, 
   +   466, -   396,    +   500, -   420,    +   444, -  1260,    +   446, -  1288, 
   +   470, -  1262,    +   442, -  1290,    +   442, -  1290,    +   446, -  1288, 
   +   444, -  1290,    +   440, -  1290,    +   444, -   424,    +   442, -   448, 
   +   448, -   392,    +   472, -   422,    +   446, -   420,    +   422, -   444, 
   +   420, -   446,    +   444, -   448,    +   420, -   446,    +   418, -   448, 
   +   418, -   448,    +   446, -   390,    +   472, -   396,    +   442, -   424, 
   +   440, -   428,    +   438, -   450,    +   418, -   424,    +   442, -   424, 
   +   442, -  1288,    +   442, -  1290,    +   442, -  1292,    +   440, -   450, 
   +   444, -   424,    +   442, -   424,    +   442, -   424,    +   440, -   450, 
   +   418, -   448,    +   418, -   424,    +   442, -   424,    +   442, -   424, 
   +   440, -  1290,    +   444, -  1290,    +   440, -  1292,    +   442, -   450, 
   +   418, -   424,    +   442, -   424,    +   442, -   426,    +   468, -   448, 
   +   418, -   448,    +   418, -   424,    +   442, -   424,    +   438, -   428, 
   +   440, -   426,    +   438, -   428,    +   438, -   426,    +   440, -   424, 
   +   442, -   426,    +   440, -   424,    +   442, -   448,    +   418, -  1288, 
   +   442, -   426,    +   440, -   450,    +   442, -  1292,    +   442, -   450, 
   +   418, -   422,    +   442, -   426,    +   438, -  1292,    +   442, -   426, 
   +   440, -   426,    +   438, -   428,    +   440, -   426,    +   438, -   428, 
   +   436, -   426,    +   440, -   426,    +   440, -   426,    +   440, -   424, 
   +   442, -   426,    +   468, -   426,    +   438, -   426,    +   438, -   426, 
   +   438, -   426,    +   442, -   450,    +   416, -   426,    +   440, -  1290, 
   +   442, -  1290,    +   442, -  1292,    +   440, -   450,    +   418, -   448, 
   +   418, -  1288,    +   442, -  1292,    +   440, -  1292,    +   442

[... cut ...] , 426,  440, 1292,  440, 426,  438, 430,  436, 428,  438, 428,  438, 428,  438, 428,  438, 428,  438, 452,  416, 450,  442, 426,  438, 426,  438, 426,  438, 428,  438, 426,  440, 426,  438, 428,  438, 1292,  440, 452,  416, 1290,  440, 1292,  440, 1292,  442, 426,  436, 1294,  442, 1292,  440};  // PANASONIC_AC
uint8_t state[27] = {0x02, 0x20, 0xE0, 0x04, 0x00, 0x00, 0x00, 0x06, 0x02, 0x20, 0xE0, 0x04, 0x00, 0x39, 0x32, 0x80, 0x75, 0x00, 0x00, 0x0E, 0xE0, 0x00, 0x00, 0x89, 0x00, 0x00, 0xDD};
```

crankyoldgit, 2019-09-17:
I think I've covered this in branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue921 / PR #923 
Can you please download and test it out?

As for the fan speed, if you find it is incorrect, please create a new/separate issue for it.

hpirila, 2019-09-18:
I have verified Issue921 branch and I can confirm Vertical Swing works now as expected. All 5 positions and auto are correct. Both Tx and Rx verified.

I will now make more comprehensive test on all functions of this AC machine.

Thank you very much for your excellent work on this important part of home automation.

crankyoldgit, 2019-09-18:
Thanks for confirm it works!

closed 2019-09-18 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

PR #923 (2019-09-18): PanasonicAc: Add better common a/c SwingV support.
Fixes #921
commit 9a5e442f (2019-09-18): PanasonicAc: Add better SwingV support for common a/c framework. (#923)

Fixes #921