crankyoldgit/IRremoteESP8266 #922: BitMask fields must be changed to make ir_Goodweather work

ribeirodanielf, 2019-09-17:
### Version/revision of the library used
Master branch and version 2.6.5

While using the lib, it was expected to turn on/off and command the AC but it was not possible. After compare with dump command it was found some differences on the message bits.

To make it work, it was necessary a change on the Bitmask definition, to address correctly the fields definition. It was also change some changes on the timers.

The code attached was tested and works fine.

[ir_Goodweather.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3623560/ir_Goodweather.cpp.txt)
[ir_Goodweather.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3623561/ir_Goodweather.h.txt)

ribeirodanielf, 2019-09-17:
Just small correction about send method. It works too and might be mantained.

#if SEND_GOODWEATHER
// Send a Goodweather message.
//
// Args:
//   data: The raw message to be sent.
//   nbits: Nr. of bits of data in the message. (Default is kGoodweatherBits)
//   repeat: Nr. of times the message is to be repeated. (Default = 0).
//
// Status: BETA / tested.
//
// Ref:
//   https://github.com/crankyoldgit/IRremoteESP8266/issues/697
void IRsend::sendGoodweather(const uint64_t data, const uint16_t nbits,
                             const uint16_t repeat) {
  if (nbits != kGoodweatherBits)
    return;  // Wrong nr. of bits to send a proper message.
  // Set IR carrier frequency
  enableIROut(38);

  for (uint16_t r = 0; r <= repeat; r++) {
    // Header
    mark(kGoodweatherHdrMark);
    space(kGoodweatherHdrSpace);

    // Data
    for (int16_t i = 0; i < nbits; i += 8) {
      uint16_t chunk = (data >> i) & 0xFF;  // Grab a byte at a time.
      chunk = (~chunk) << 8 | chunk;  // Prepend a inverted copy of the byte.
      sendData(kGoodweatherBitMark, kGoodweatherOneSpace,
               kGoodweatherBitMark, kGoodweatherZeroSpace,
               chunk, 16, false);
    }
    // Footer
    mark(kGoodweatherBitMark);
    space(kGoodweatherHdrSpace);
    mark(kGoodweatherBitMark);
    space(kDefaultMessageGap);
  }
}
#endif  // SEND_GOODWEATHER

crankyoldgit, 2019-09-18:
Hi, I've had a look at the attached files and compared them to the existing code.

As you've noted, I'm going to keep the original `sendGoodweather()` code as it compiles smaller and is ever so slightly faster.

The altered timing values you proposed cause your previous message to not be successfully matched. So I'm hesitant to change those just yet.

Running previous data through the newer detailed message inspection seems to indicate the fan speeds are now different from what you previously said were correct.

Can you please provide me some new rawData (and associated `uint64_t data = ` line) and documentation as to what the message really (when viewing the remote) should be.

e.g. A message which you previously indicated was:
_Power: On, Mode: 1 (COOL), Temp: 20C, **Fan: 3 (Low)**, Turbo: -, Light: -, Sleep: -, Swing: 0 (Fast), Command: 0 (Power)_
is now decoded as:
_Power: On, Mode: 1 (COOL), Temp: 20C, **Fan: 2 (Medium)**, Turbo: -, Light: -, Sleep: -, Swing: 0 (Fast), Command: 0 (Power)_

Is that what you would expect?

ribeirodanielf, 2019-09-18:
Hi, I've just tested and need small adjustment on the variables below to give a good match.
// Timing
const uint16_t kGoodweatherBitMark = 580;
const uint16_t kGoodweatherOneSpace = 580;
const uint16_t kGoodweatherZeroSpace = 1860;
const uint16_t kGoodweatherHdrMark = 6820;
const uint16_t kGoodweatherHdrSpace = 6820;

ribeirodanielf, 2019-09-18:
The latest version I'm using here.

[ir_Goodweather.cpp.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3624083/ir_Goodweather.cpp.txt)
[ir_Goodweather.h.txt](https://github.com/crankyoldgit/IRremoteESP8266/files/3624084/ir_Goodweather.h.txt)

crankyoldgit, 2019-09-18:
> Can you please provide me some new rawData (and associated `uint64_t data = ` line) and documentation as to what the message really (when viewing the remote) should be.

Again, I ask the above.

crankyoldgit, 2019-09-18:
I think I've incorporated your suggestions as best as I can.

Can you please download and test branch: https://github.com/crankyoldgit/IRremoteESP8266/tree/Issue922 / PR #924 and let me know if it does all that you need now?

crankyoldgit, 2019-09-20:
Have you tested it yet?

ribeirodanielf, 2019-09-20:
Sorry for the late reply. Yes, it is working fine.
Thanks for your assistance.
Best Regards,

crankyoldgit, 2019-09-20:
Thanks for the confirmation!

closed 2019-09-20 by crankyoldgit

crankyoldgit, 2019-09-23:
FYI, the changes mentioned above are included in the newly released version of the library (v2.6.6).

PR #924 (2019-09-20): Goodweather: Adjust timings & minor fixes.
* Adjust timings per #922
* Set an initial value for the IRGoodweather class.
* Add extra `tolerance` to help with matching.

Fixes #922
commit 00bf01c0 (2019-09-20): Goodweather: Adjust timings & minor fixes. (#924)

* Adjust timings per #922
* Set an initial value for the IRGoodweather class.
* Add extra `tolerance` to help with matching.

Fixes #922