crankyoldgit/IRremoteESP8266 #926: Samsung AC AJ035NB1DEH support

nav208, 2019-09-20:
Hello, i have samsung AC model AJ035NB1DEH.
Its decode as 

```
Encoding  : SAMSUNG_AC
Code      : 02B20F000000C001D20F000000000102FF71A001C0 (168 bits)
Mesg Desc.: Power: Off, Mode: 0 (AUTO), Temp: 26C, Fan: 0 (Auto), Swing: Off, Beep: Off, Clean: Off, Quiet: Off, Powerful: Off
Raw Timing[349]:
   +   642, - 17636,    +  2986, -  8924,    +   512, -   476,    +   516, -  1470, 
   +   514, -   474,    +   518, -   474,    +   516, -   476,    +   514, -   480, 
   +   514, -   478,    +   512, -   500,    +   492, -   498,    +   492, -  1494, 
   +   490, -   502,    +   536, -   456,    +   574, -  1408,    +   572, -  1410, 
   +   516, -   476,    +   516, -  1466,    +   516, -  1468,    +   516, -  1468, 
   +   516, -  1468,    +   514, -  1472,    +   512, -   500,    +   492, -   498, 
   +   492, -   502,    +   490, -   502,    +   542, -   450,    +   564, -   426, 
   +   574, -   418,    +   544, -   446,    +   518, -   474,    +   514, -   476, 
   +   518, -   474,    +   516, -   474,    +   516, -   474,    +   518, -   474, 
   +   516, -   476,    +   518, -   476,    +   514, -   478,    +   514, -   478, 
   +   514, -   498,    +   492, -   498,    +   494, -   500,    +   490, -   502, 
   +   532, -   458,    +   570, -   420,    +   576, -   418,    +   572, -   418, 
   +   518, -   472,    +   518, -   474,    +   518, -   472,    +   518, -   472, 
   +   518, -   474,    +   518, -   472,    +   518, -   474,    +   516, -   476, 
   +   516, -  1470,    +   514, -  1490,    +   576, -  2882,    +  3066, -  8868, 
   +   490, -  1492,    +   566, -   422,    +   574, -   420,    +   574, -   414, 
   +   548, -   444,    +   544, -   448,    +   518, -   472,    +   546, -   446, 
   +   544, -   446,    +   520, -  1462,    +   546, -   446,    +   542, -   450, 
   +   544, -  1440,    +   542, -   470,    +   520, -  1464,    +   522, -  1462, 
   +   536, -  1446,    +   576, -  1408,    +   572, -  1414,    +   544, -  1436, 
   +   520, -   474,    +   546, -   442,    +   550, -   444,    +   548, -   444, 
   +   546, -   446,    +   546, -   446,    +   546, -   446,    +   546, -   466, 
   +   522, -   470,    +   524, -   470,    +   522, -   468,    +   532, -   462, 
   +   564, -   428,    +   570, -   420,    +   574, -   418,    +   570, -   422, 
   +   546, -   444,    +   548, -   444,    +   548, -   444,    +   570, -   422, 
   +   572, -   420,    +   572, -   420,    +   546, -   446,    +   574, -   420, 
   +   544, -   468,    +   548, -   442,    +   550, -   440,    +   550, -   444, 
   +   546, -   446,    +   530, -   462,    +   536, -   454,    +   572, -   420, 
   +   574, -   418,    +   572, -   418,    +   572, -   418,    +   574, -   416, 
   +   574, -  2910,    +  3044, -  8860,    +   578, -  1406,    +   578, -   416, 
   +   578, -   414,    +   576, -   438,    +   552, -   438,    +   554, -   438, 
   +   552, -   442,    +   552, -   440,    +   510, -   480,    +   532, -  1452, 
   +   572, -   420,    +   574, -   418,    +   576, -   414,    +   578, -   416, 
   +   578, -   412,    +   580, -   412,    +   578, -  1404,    +   582, -  1402, 
   +   576, -  1410,    +   576, -  1428,    +   556, -  1430,    +   554, -  1430, 
   +   504, -  1478,    +   568, -  1416,    +   574, -  1410,    +   580, -   414, 
   +   576, -   414,    +   578, -   412,    +   580, -  1404,    +   580, -  1404, 
   +   578, -  1408,    +   576, -   436,    +   554, -   436,    +   556, -   434, 
   +   556, -   438,    +   552, -   440,    +   506, -   486,    +   562, -  1420, 
   +   572, -   420,    +   574, -  1408,    +   580, -  1404,    +   580, -   412, 
   +   578, -   414,    +   578, -   412,    +   580, -   412,    +   578, -   416, 
   +   576, -   416,    +   576, -   436,    +   556, -   434,    +   556, -   438, 
   +   554, -   438,    +   552, -   440,    +   504, -   486,    +   536, -   456, 
   +   566, -  1416,    +   576, -  1408,    +   580

uint16_t rawData[349] = {642, 17636,  2986, 8924,  512, 476,  516, 1470,  514, 474,  518, 474,  516, 476,  514, 480,  514, 478,  512, 500,  492, 498,  492, 1494,  490, 502,  536, 456,  574, 1408,  572, 1410,  516, 476,  516, 1466,  516, 1468,  516, 1468,  516, 1468,  514, 1472,  512, 500,  492, 498,  492, 502,  490, 502,  542, 450,  564, 426,  574, 418,  544, 446,  518, 474,  514, 476,  518, 474,  516, 474,  516, 474,  518, 474,  516, 476,  518, 476,  514, 478,  514, 478,  514, 498,  492, 498,  494, 500,  490, 502,  532, 458,  570, 420,  576, 418,  572, 418,  518, 472,  518, 474,  518, 472,  518, 472,  518, 474,  518, 472,  518, 474,  516, 476,  516, 1470,  514, 1490,  576, 2882,  3066, 8868,  490, 1492,  566, 422,  574, 420,  574, 414,  548, 444,  544, 448,  518, 472,  546, 446,  544, 446,  520, 1462,  546, 446,  542, 450,  544, 1440,  542, 470,  520, 1464,  522, 1462,  536, 1446,  576, 1408,  572, 1414,  544, 1436,  520, 474,  546, 442,  550, 444,  548, 444,  546, 446,  546, 446,  546, 446,  546, 466,  522, 470,  524, 470,  522, 468,  532, 462,  564, 428,  570, 420,  574, 418,  570, 422,  546, 444,  548, 444,  548, 444,  570, 422,  572, 420,  572, 420,  546, 446,  574, 420,  544, 468,  548, 442,  550, 440,  550, 444,  546, 446,  530, 462,  536, 454,  572, 420,  574, 418,  572, 418,  572, 418,  574, 416,  574, 2910,  3044, 8860,  578, 1406,  578, 416,  578, 414,  576, 438,  552, 438,  554, 438,  552, 442,  552, 440,  510, 480,  532, 1452,  572, 420,  574, 418,  576, 414,  578, 416,  578, 412,  580, 412,  578, 1404,  582, 1402,  576, 1410,  576, 1428,  556, 1430,  554, 1430,  504, 1478,  568, 1416,  574, 1410,  580, 414,  576, 414,  578, 412,  580, 1404,  580, 1404,  578, 1408,  576, 436,  554, 436,  556, 434,  556, 438,  552, 440,  506, 486,  562, 1420,  572, 420,  574, 1408,  580, 1404,  580, 412,  578, 414,  578, 412,  580, 412,  578, 416,  576, 416,  576, 436,  556, 434,  556, 438,  554, 438,  552, 440,  504, 486,  536, 456,  566, 1416,  576, 1408,  580};  // SAMSUNG_AC
uint8_t state[21] = {0x02, 0xB2, 0x0F, 0x00, 0x00, 0x00, 0xC0, 0x01, 0xD2, 0x0F, 0x00, 0x00, 0x00, 0x00, 0x01, 0x02, 0xFF, 0x71, 0xA0, 0x01, 0xC0};

A001C0

```

If use ir_Samsung.h, is send only 112 bits 
Code      : 02920F000000F001C2FE719015F0 (112 bits)

Thanks for help

crankyoldgit, 2019-09-20:
That looks like an extended Samsung message. Try irsend.sendSamsungExtended(state);

nav208, 2019-09-20:
> sendSamsungExtended

sendSamsungExtended(state) not exist in 2.6.5 version..

crankyoldgit, 2019-09-20:
Sorry, try irsend.sendSamsungAC(state, kSamsungAcExtendedStateLength);

crankyoldgit, 2019-09-23:
How did you go with that?

nav208, 2019-09-23:
Its ok, thanks :)

1043717432, 2019-09-24:
> Its ok, thanks :)

How do you add a new a / C? Thank you!

crankyoldgit, 2019-09-24:
@1043717432 As previously, https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

closed 2019-09-24 by crankyoldgit

1043717432, 2019-09-25:
> @1043717432 As previously, https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol

Thank you!!! My DAKIN A/C code doesn't know how to import it into the state.xlsx table. It failed. Limited capacity, unable to continue, declared withdrawal.