crankyoldgit/IRremoteESP8266 #927: not sending and showing unknown remote codes

ankit993, 2019-09-20:
board esp8266.
sometimes its not showing decoded codes.
i working fine with LG AC remote, but other hand with Voltas AC, it is not decoding, when  i try to decode voltas remote and others, its showing nothing.

crankyoldgit, 2019-09-20:
You are going to need to supply a lot more information if you want some help.

For instance, the suggested steps from the New Issue template that you didn't see, read or fill out.

https://github.com/crankyoldgit/IRremoteESP8266/blob/master/.github/issue_template.md#output-of-raw-data-from-irrecvdumpv2ino-if-applicable

crankyoldgit, 2019-09-20:
Or the FAQ: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#the-libraryexample-code-doesnt-understand-my-air-conditioner-remote-

crankyoldgit, 2019-09-23:
Do you have any results from using the `IRrecvDumpV2` example code, and/or are you sure they are IR remotes, not RF or Bluetooth?

ankit993, 2019-09-23:
Fro IRrecvDumpV2, i am getting good results, but from mqttserver, not getting.

crankyoldgit, 2019-09-23:
And what is the result from `IRrecvDumpV2`? Does the library recognise it?

ankit993, 2019-09-23:
Is shows only hex_code and raw_code and type unknown and bits_size is 116.
And other remote, its shows everything, name, code, etc...

crankyoldgit, 2019-09-23:
As it is `UNKNOWN`, the library doesn't recognise it. Thus, `IRMQTTServer` won't either. It is as yet unsupported.

See https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-IR-protocol & https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol for how to get it added.

closed 2019-09-23 by crankyoldgit