crankyoldgit/IRremoteESP8266 #931: A possible way to reduce code size

jimmys01, 2019-09-25:
Through the code there are multiple uses of the words "Max" "Min" "Auto" "Toggle" "Off" etc.
Perhaps there is a way to make the 30 instances of "OFF" to just occupy 1*sizeof("OFF") and not 30*sizeof("OFF")?

crankyoldgit, 2019-09-26:
There is, and I'll get to it, eventually.

crankyoldgit, 2019-09-26:
Alas, I expect it won't save a heck of a lot of space, but I'm currently working on a major re-write of the A/C stuff just to save ~1K of space (and to produce safer/cleaner code).

I want to add Internationalisation support too, which will probably happen at the same time.

crankyoldgit, 2019-10-02:
Just a FYI update. I've started work on this but it will have to wait till #934 is merged to reduce a "merge conflict hell". i.e. This isn't forgotten.

crankyoldgit, 2019-10-05:
Progress update.

I've completed most of the work for it. I've got a few more cosmetic/practical stuff to do yet which shouldn't affect the resulting binary size etc. The changes save about 1.5k+. Which honestly was more than I was expecting. However, it breaks the use of `F()` macros on the base strings for some reason. That's probably a small price to pay honestly.

closed 2019-10-06 by crankyoldgit

crankyoldgit, 2019-10-30:
@jimmys01 FYI, the changes mention above have been included in the v2.7.0 release of the library.

PR #946 (2019-10-06): Centralise copies of strings and add I18n support.
* Create a framework for doing internationalisation(i18n).
* Centralise strings to `IRtext`. Saves approximately 1.5k+ of prog space.
* Add linter coverage to src/locale.
* Clean up/standardise text for A/Cs.
* Add `modelToStr()` & `addDayToString()` functions.
* Change `IRrecvDumpV2` behaviour
  * Remove displaying the legacy "Raw Data" dump. Just rely on the `uint16_t rawData[]` output.
  * i18n the text in the example code.
* More string consolidation.
* Remove `F()` around `IRtext` strings as it causes the compiler to berk when built under [... cut]
commit 1ad29079 (2019-10-06): I18n support & centralise strings. (#946)

* Create a framework for doing internationalisation(i18n).
* Centralise strings to `IRtext`. Saves approximately 1.5k+ of prog space.
* Add linter coverage to `src/locale`.
* Clean up/standardise text for A/Cs.
* Add `modelToStr()` & `addDayToString()` functions.
* Change `IRrecvDumpV2` behaviour
  * Remove displaying the legacy "Raw Data" dump. Just rely on the `uint16_t rawData[]` output.
  * i18n the text in the example code.
* More string consolidation.
* Remove `F()` around `IRtext` strings as it causes the compiler to berk when built under the A [... cut]