crankyoldgit/IRremoteESP8266 #932: Hacktoberfest 2019! Free T-Shirt!

crankyoldgit, 2019-09-26:
## Hacktoberfest
It's (almost) October and that means it's Hacktoberfest time [again](https://github.com/crankyoldgit/IRremoteESP8266/issues/537)! #hacktoberfest
See https://hacktoberfest.digitalocean.com/ for details.

This year, to get a free T-Shirt, you need to [register](https://hacktoberfest.digitalocean.com/register) & submit 4 PRs during October (1-31). They don't have to be in this project. See the link for the rules. You can [register](https://hacktoberfest.digitalocean.com/register) now!

Here are just some suggestions for things that would improve this project/library to help you get started:

## Easy:
* Fix some typos. I'm a horrendous speller and have shocking grammar. There are plenty to find and fix.
* Improve some of the documentation.

## Medium:
* Add some function description comments. There are plenty of examples in the code where I've done this. Use that as a template for any function etc that doesn't have a description at the start of it.
* Improve/fix some of the code style issues. See our [Code Lint And Style guide](https://github.com/markszabo/IRremoteESP8266/wiki/Library-Maintainers-Guide#code-lint-and-style) and the [Google C++ Style Guide](https://google.github.io/styleguide/cppguide.html) for things to look for. I guarantee there are plenty to fix/correct. ;-)
* Improve our code examples. e.g. Add comments, tips or features you think folk might need when first using the library.
* Create/add [README.md](https://en.m.wikipedia.org/wiki/README)s for the code examples or other directories in the library. [Further reading](https://help.github.com/en/articles/about-readmes).

## Hard:
* Write some more unit tests or add more unit test cases.
* Find & **fix** some bugs.
* Make the Example code prettier/easier to read/look better.
* Improve our Travis tests.

## Awesome!!!1!one!:
* Add support for a new protocol. ([It's even easier now!](../wiki/Adding-support-for-a-new-IR-protocol))
* Create unit tests for some of the protocols with out any!
* Write a CLI tool to parse & decode a `uint16_t rawData[] = {...};` line, using the library, into it's `protocol` & `value`/`state[]` form.

Go crazy and contribute to open-source!

crankyoldgit, 2019-09-30:
FYI, It's now Oct 1st somewhere in the world. Go nuts. Bring on the PRs!

crankyoldgit, 2019-10-08:
As we have support for multiple languages now, added a suggestion of:
* Know a non-English language? Consider adding or improving [internationalisation (i18n)](https://en.wikipedia.org/wiki/Internationalization_and_localization) support. See [src/locale](../tree/master/src/locale) for details.

Knackie, 2019-10-10:
i can translate makefile in french if you want ?

crankyoldgit, 2019-10-10:
@Knackie  Sure. And/Or you could add French language support for the library!
See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/locale/README.md

Knackie, 2019-10-10:
you want I translate the locale/README ?

crankyoldgit, 2019-10-10:
I only just added it, so it might get changed soon. I was thinking of you adding 'fr-FR.h'. i.e. like what locale/README.md suggests. :-)

Knackie, 2019-10-10:
ok i can if you want 😄

crankyoldgit, 2019-10-10:
Well it will help your fellow French speakers/users of the library.

Knackie, 2019-10-10:
ok, i'll translate the comment 😄

crankyoldgit, 2019-10-23:
Just a reminder to all, if you haven't got your 4 PR's in yet, you've only got a week or so left to earn a free T-shirt. Get those PRs in! There is still plenty of time.

crankyoldgit, 2019-10-30:
Last days! Get you're PR's in!
Still plenty of typos and simple mistakes to find, comments to be added etc. You could still earn a T-shirt!

crankyoldgit, 2019-11-01:
Till next year.

closed 2019-11-01 by crankyoldgit