crankyoldgit/IRremoteESP8266 #937: Coolix kCoolixDefaultState has wrong Fan definition

ribeirodanielf, 2019-10-01:
Accordingly to the setMode method, AUTO mode can be set only using FAN kCoolixFanAuto0 (0b000) but it is defined with kCoolixFanAuto (0b101).

// On, 25C, Mode: Auto, Fan: Auto, Zone Follow: Off, Sensor Temp: Ignore.
const uint32_t kCoolixDefaultState = 0b10110010**101**111111100**10**00;  // 0xB2BFC8

It is causing the AC to not start after code initialization.
Could please change to the value below?

// On, 25C, Mode: Auto, Fan: Auto, Zone Follow: Off, Sensor Temp: Ignore.
const uint32_t kCoolixDefaultState = 0b10110010**000**111111100**10**00;  // 0xB21FC8

Regards,
Daniel

crankyoldgit, 2019-10-02:
Hey Daniel,
 As it's #Hacktoberfest (See #932) and it is relatively a simple change, why not create a Pull Request (PR) for it?

i.e. Go to https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Coolix.h#L83-L84, scroll up, and click edit!

Then you're on your way to earning a quarter of a t-shirt. See #932 for a the URL to register first to be eligble! :)

Otherwise, yes, I can fix it, but wouldn't you rather a free t-shirt? :)

closed 2019-10-02 by crankyoldgit

ribeirodanielf, 2019-10-02:
Looks like somebody else did :)
Thanks for the fix.
Regards

crankyoldgit, 2019-10-30:
FYI, the changes mention above have been included in the v2.7.0 release of the library.

PR #941 (2019-10-02): Fixed Coolix kCoolixDefaultState (file Update ir_Coolix.h)
Closes #937: Coolix kCoolixDefaultState has wrong Fan definition.
Ref #932
commit 2db55f1a (2019-10-02): Fixed Coolix kCoolixDefaultState (file Update ir_Coolix.h) (#941)

Closes #937: Coolix kCoolixDefaultState has wrong Fan definition.
* New default state changes some of the unit test results. Fixing.
Ref #932