crankyoldgit/IRremoteESP8266 #945: Mitsubishi 112 - New Protocol

kuchel77, 2019-10-04:
### Version/revision of the library used
v2.6.6

### Other useful information

Ok, thanks to some of the tips provided to someone else, I have finally got the decoding of my Mitsubishi AC IR remote working, and I will work on writing some documentation to help others achieve the same goal.

The only issue I am having now is with calculating the checksums.

I have put together a spreadsheet similar the to Mitsubishi one, with some relevant codes, and just going up from 16 to 31 degrees, and then changing the vertical vane and coming down. I can see the checksums should be the last byte (or 2 nibbles), but I can't seem to work out how to calculate them. Because they are going up and down in a consistent fashion, it would suggest a sum, but I can't quickly figure out what should be added together.

If anyone can help that would be great.

https://docs.google.com/spreadsheets/d/1Vv_Qv-AaJjlN3FJCEsKBPZd8UEozi52B7S6-T3s2B8E/edit?usp=sharing

Once I figure this out, and double check my code, I will PR it into the project properly.

Also, I am not sure if there is a standard for new names for protocols, so i just went with Mitsubishi 112 as it is 112 bits long. The remote on the back says KPOA. Once I have the checksum working, I will get the AC model number as well.

Thanks.

crankyoldgit, 2019-10-05:
First off, kudos for attempting something so mammoth. It's a really big effort. Congrats on what you've done so far. (I saw your fork of the repo earlier)

I'll take a look and see if I can see anything obvious for the checksum.

Can you please make the document you linked read-able by everyone?

Feel free to start a PR now, we can work through it together.

crankyoldgit, 2019-10-05:
I had a quick look at the data.
It appears to be a simple "sum of the preceding bytes" algorithm. See: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Adding-support-for-a-new-AC-protocol#1-a-sum-of-the-preceding-bytes

See: https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRutils.cpp#L710-L716
e.g. `sumBytes(remote_state, kMitsubishi112StateLength - 1) == remote_state[kMitsubishi112StateLength - 1]` .. where `kMitsubishi112StateLength - 1` should be `13` If I'm reading your data right.

Using your data from here: https://docs.google.com/spreadsheets/d/1Vv_Qv-AaJjlN3FJCEsKBPZd8UEozi52B7S6-T3s2B8E/edit#gid=0&range=D11:AE11

_0x23 + 0xCB + 0x26 + 0x01 + 0x00 + 0x24 + 0x03 + 0x0E + 0x12 +0x00 + 0x00 + 0x00 + 0x04 = 0x260_ ... If you _mod 256_ it. i.e. `% 0x100` or `& 0xFF` it. You get your last byte, which is `0x60`. i.e. `state[13]`

kuchel77, 2019-10-06:
You were spot on. I was missing a vital column in my local Excel version of the spreadsheet I was using to try and calculate this. I have it all working now and I can control my air conditioner now. 

Just for those wanting to try, the model number of my Air Condition is MSH-A24WV for the part inside, and MUH-A24WV for the part outside. The remote has a code KPOA on the outside. I will add this to the code and documentation next, but if people are googling this is the meantime it should at least give them somewhere to start.

crankyoldgit, 2019-10-07:
FYI @kuchel77 I've updated your PR / master branch to include some fixes/corrections/review alterations etc.
There are still a few unresolved questions I need you to answer.

closed 2019-10-21 by crankyoldgit

crankyoldgit, 2019-10-30:
FYI, the changes mention above have been included in the v2.7.0 release of the library.

PR #968 (2019-10-21): Cleanup new Mitsubishi(112) A/C code.
* Use the i18n system for text.
* Use `setBit()`/`GETBIT8()` etc.

Fixes #945
Ref #947
FYI @kuchel77
commit 0b3b4a5d (2019-10-21): Cleanup new Mitsubishi(112) A/C code. (#968)

* Use the 18n system for text.
* Use setBit/GETBIT8() etc.

Fixes #945
Ref #947