crankyoldgit/IRremoteESP8266 #961: Utilize this library to make a standalone tool

arihantdaga, 2019-10-11:
@crankyoldgit . Sorry for not using the standard issue template, and its actually not an issue, but i wanted to make a standalone tool using this library for decoding remote codesets. Since this library has improved so much and is able to decode so many protocols now. 
Without esp8266 also, i was assuming that we can detect protocols for given raw codes. 

I wanted to make  javascript based tool, which can use compiled binaries or by using llvm and emscripten (I am not sure about weather it can be done though). I do not have much experience with c++ however, so naively i tried to compile a new `main.cpp` having code similar to `IrRecvDumpV2.ino`. And ofcourse it failed saying 
```
Undefined symbols for architecture x86_64:
  "IRrecv::enableIRIn(bool)", referenced from:
      _main in main-182832.o
  "IRrecv::IRrecv(unsigned short, unsigned short, unsigned char, bool)", referenced from:
      ___cxx_global_var_init in main-182832.o
  "IRrecv::~IRrecv()", referenced from:
      ___cxx_global_var_init in main-182832.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
```
By a little understanding of  C++ i have, i guessed that we may have to compile dependencies also. 
I am not sure if it can be done. But if it can be done, i would like to contribute and build this tool. @crankyoldgit  If you can point me in some direction ?

crankyoldgit, 2019-10-11:
Based on that error message, it looks like you failed to included the `IRrecv.o` object file to the linker.
See here for all the files you'll need to link against to get it to work.
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/Makefile#L44-L70

The work you want to do _may_ have some-what already been done for you, see the command line tool 
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/gc_decode.cpp

It will take GlobalCache codes, and Pronto codes as input, and decode them. It wouldn't be too much effort to add _RAW_ as an input format.

If you are going to compile on a **non** arduino platform, such as X86 etc. You will need to compile with `-DUNIT_TEST` to the compiler. e.g. https://github.com/crankyoldgit/IRremoteESP8266/blob/master/tools/Makefile#L21

Yes, it will work fine on the X86 platform. All of the unit tests (and those programs in the _tools/_ directory all do too) run on X86 regularly.

crankyoldgit, 2019-10-11:
Cool. I look forward to your success and eventual Pull Request!

FYI, everything under the `test/` & `tools/` directories is designed to run PCs etc. Mostly Linux, but with some effort it should run under Windows too.

crankyoldgit, 2019-10-13:
Have you now solved your own issue? :) i.e. Can we close this issue now your PR is merged?

arihantdaga, 2019-10-13:
Haha. Yes ofcourse. Thank you :)

closed 2019-10-13 by arihantdaga

crankyoldgit, 2019-10-30:
FYI, the changes mention above have been included in the v2.7.0 release of the library.

arihantdaga, 2019-10-30:
I am working on a GUI Tool, i'll soon update.