crankyoldgit/IRremoteESP8266 #964: Is there a limit to Sending/Recieving Characters?

icefire555, 2019-10-12:
### Version/revision of the library used
v2.66

### Expected behavior
I wanted to be able to send basic binary bits or longer strings of data

### Actual behavior
I can't seem to Send/Recieve more than 8 Values at a time, and if I double the Irsend command it is very slow to send the second signal.

#### Circuit diagram and hardware used (if applicable)
I'm using a DOIT ESP32 30pin, all of the code seems to be working wonderfully, 

### Other information
I really want to send the quickest amount of data I can all at once. Preferably binary, since it's very basic to send over light. However I'm not sure how to do that in this library. 

I was looking through the files and noticed a "sendLasertag" function as part of the IRsend files and thought it might be more of what I need. But I'm unsure how to use it properly.

If anyone has some insight into how to send extra values, or how to just send binary, I would love it! I tried the send raw, but the light was much dimmer than I wanted. And when I how the receiver read the light, it tended to show up as random characters.

icefire555, 2019-10-12:
Update: I'm using the exsample code for IRrecvDemo and IRsendDemo

icefire555, 2019-10-12:
LOL, I changed the serialPrintUint64(results.value, HEX); to serialPrintUint64(results.value, BIN); and it made the output into binary. That's half the battle.

crankyoldgit, 2019-10-12:
This library is _not_ geared towards sending arbitrary sized data via IR. It's aimed at decoding and reproducing existing _established_ protocols. Some protocols have fixed sizes ranging from 12 bits to 48 bits of arbitrary data. Some other protocols can send much longer messages i.e. ~20-30-ish 8-bit chars, but have special check values involved, which the library can calculate for you. I think there is only one arbitrary length established protocol supported by this library, but the data has to be specially formatted first. The library doesn't have the routines to format the data for you. It can send it, & decode it, it just can't format it.

Or, you could invent & add your own protocol and decoder (not that hard, but not trivial)

Typical IR protocols are capable of sending a single bit every 1-1.7ish milliseconds.
Most IR protocols have a 40-100 millisecond gap between messages to allow other messages to get through. So, its quite common for a 32 bit message to take 1/8th+ of a second. 

So the question back to you is, _how much data do you want to send at a time?_
That's going to need to be answered before I can suggest a protocol/method to you.

Also, check out #887

icefire555, 2019-10-12:
Actually. I think I counted around 30 bits of binary on my longest message sent. That is more than enough. I originally was upset because it was converting to hex and confusing me. TBH, your library would probably work for this if I converted a binary message into hex to send it. But I don't understand it well enough to know when it would cut the message short. I noticed the ir send code allowed you to select the length, I am curious if that would be the max binary length.

crankyoldgit, 2019-10-12:
You've got several options to send 32 bits of data.
1. `sendWhynter()` (see https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Whynter.cpp) will let you send arbitrary data (i.e. no check bits or patterns)
2. Alternatively, the most common/abused protocol is NEC. `sendNEC()` (see https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_NEC.cpp) will let you also send 32 bits of data, but it will be detected as one of two protocols depending on the payload. NEC has a specific bit/check pattern, if the payload follows that pattern it will be detected as `NEC`, if it doesn't it will be detected as `NEC_LIKE`. i.e. You will have to check that the protocol to see if it is either `NEC` or `NEC_LIKE`.
3. `sendGoodweather()`. Not as fast, but more data (48 bits) (See https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Goodweather.cpp). It duplicates the message by sending an inverted copy a byte for every byte it sends. So, more reliable, but at least twice as long a message.

> But I don't understand it well enough to know when it would cut the message short. I noticed the ir send code allowed you to select the length, I am curious if that would be the max binary length.

That, there is a coding problem for your end/program. You need to design a system for that.
e.g. If you never need to send a NULL byte (`0x00`/`0b00000000`) then you could use a C-style NULL terminated string method. But that relies on you never needing to send a value of `0x00`.

You don't want to send messages with non-standard bit length for a given protocol, as the library will most likely ignore them as `UNKNOWN`. Only do that if you know exactly what you are doing, and TBH, it doesn't sound like you do yet.

icefire555, 2019-10-12:
What is the goal of the null byte? Is it just a pickup so the receiver knows a signal is coming? 

I was planning on sending binary and adding a 1 or 2 parity bits to the end to try and correct errors on the fly. Although I haven't tested anything outside, and I have a feeling the sun will wreck havoc on IR signals. 
And yeah I have a lot to learn, I really appreciate the advice. My plan earlier was to abuse the NEC command and convert binary into Hex. Although I'l look into the Whynter functions to see how it reacts.

crankyoldgit, 2019-10-13:
> What is the goal of the null byte? Is it just a pickup so the receiver knows a signal is coming?

It was just a suggestion of method of indicating the end of an arbitrary/variable length piece of data. e.g. Just like a C-style string. You don't need to use it, but if you want an arbitrary length encoding, you will have to come up with something yourself.

> Although I haven't tested anything outside, and I have a feeling the sun will wreck havoc on IR signals.

IR and direct sunlight are not a reliable environment for success. I suggest you try to see if you can get successful transmission sooner rather than later to potentially save wasted effort.

`Whynter` should offer a solution that is less susceptible to data corruption during transmission.

icefire555, 2019-10-13:
I got it working!
Using 
int Test = (B11111111 * 16777216) + (B11111111 * 65536) + (B11111111 * 256) + B11110000;
I can store binary into an int variable. Unfortuantly, if you want more than 8 bits, you have to use multiplication to designate where it's at in the binary equation.

But then using 
irsend.sendNEC(0x00 + Test);

Will result in 
"11111111111111111111111111110000"

icefire555, 2019-10-13:
The only downside is it will not send 0's at the front of the binary code.
Like writing the number 005. It doesn't makes sense to write the 00s without a reason.

icefire555, 2019-10-13:
It would also appear that using
irsend.sendInax(0x00+Test,32,1);
irsend.sendPioneer(0x00+Test,32,1);
or
irsend.sendRC6(0x00+Test,32,1);
Will output a matching secondary signal.

icefire555, 2019-10-13:
irsend.sendRC6(0x00+Test,32,1);
Seems to have the fastest and brightest signal. But it also has the largest gap.

closed 2019-10-13 by icefire555

crankyoldgit, 2019-10-13:
> The only downside is it will not send 0's at the front of the binary code.
> Like writing the number 005. It doesn't makes sense to write the 00s without a reason.

You are conflating _zeros being sent_ with _zeros being printed_. They are not the same thing. It is in fact always sending 32 bits of data. It _is_ sending those leading zeros. It's the print routine that isn't showing them, because they are implied.

To make your programming job easier, I _strongly_ suggest you learn about [BitMath](https://playground.arduino.cc/Code/BitMath/), especially the _bitshift_ operators [<<](https://www.arduino.cc/reference/en/language/structure/bitwise-operators/bitshiftleft/) & [>>](https://www.arduino.cc/reference/en/language/structure/bitwise-operators/bitshiftright/).

e.g. Your code:
```
int Test = (B11111111 * 16777216) + (B11111111 * 65536) + (B11111111 * 256) + B11110000;
```
becomes
```
uint32_t Test = B11111111 << 24 | B11111111 << 16 | B11111111 << 8 | B11110000;
```

In addition, in the master branch (will be in the next release), are routines for [setBits()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRutils.cpp#L1128-L1183) and [GETBITSnn()](https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/IRutils.h#L79-L89).

> It would also appear that using
irsend.sendInax(0x00+Test,32,1);
irsend.sendPioneer(0x00+Test,32,1);
or
irsend.sendRC6(0x00+Test,32,1);
**Will output a matching secondary signal.**

Yes, that's because you've specified it to do a single "repeat" with the `..., 1)` argument. So, working as intended.

P.S. Your `0x00+Test` is a [no-op](https://en.wikipedia.org/wiki/NOP_(code)). I don't think it's doing what you think it's doing. i.e. Using `Test` instead is exactly the same.

icefire555, 2019-10-13:
> You are conflating zeros being sent with zeros being printed. They are not the same thing. It is in fact always sending 32 bits of data. It is sending those leading zeros. It's the print routine that isn't showing them, because they are implied.
That makes sense. If the value is 32bits, they should all be there. At first I was just going to live with a 1bit loss on transmission. But I don't think it would be hard to get it back if I think through things a little. 

> To make your programming job easier, I strongly suggest you learn about BitMath, especially the bitshift operators << & >>.
This seems super helpful so far! I'll give it a read!

>Yes, that's because you've specified it to do a single "repeat" with the ..., 1) argument. So, working as intended.
When I tell it to repeat on any other functions, it either sends a non matching sequence, or it sends the repeat twice in a line(Receive function) with only 1 repeat being called.

> P.S. Your 0x00+Test is a no-op. I don't think it's doing what you think it's doing. i.e. Using Test instead is exactly the same.
Yeah, I wasn't sure, and it was there with the sendNEC() function so I thought it might be important.