crankyoldgit/IRremoteESP8266 #970: daikin 152 bit. can we help me with checksum?

curla92, 2019-10-17:
```
uint8_t DaikinTemplate_152[] = {
    // header
    0x11, 0xDA, 0x27, 0x00, 0x00, 0x30, 0x34, 0x00, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC5, 0x40, 0x00, 0xAB};
    //  0     1     2     3     4     5     6     7    8    9    10    11    12    13    14    15    16    17   18 

checksum_152 = 0x00;
            
for (int i=0; i<18; i++) {
      checksum_152 += DaikinTemplate_152[i];
}
            
DaikinTemplate_152[18] = checksum_152;
```

Can we explain what's wrong? i am not able to find the error. 
thank y

crankyoldgit, 2019-10-18:
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.cpp#L2952-L2956
```
// Calculate and set the checksum values for the internal state.
void IRDaikin152::checksum(void) {
  remote_state[kDaikin152StateLength - 1] = sumBytes(
      remote_state, kDaikin152StateLength - 1);
}
```
Where `const uint16_t kDaikin152StateLength = 19;`

So, your code looks fine to me. i.e. `DaikinTemplate_152[18]`  should contain `0xAB` at the end of your code's execution. `checksum_152` should contain `0x2AB` at the end.
Perhaps try: `uint8_t checksum_152 = 0x00;` but that shouldn't make a difference.

curla92, 2019-10-18:
> kDaikin152StateLength

i always received 154 bit as response. i have 2 bits plus, it's very strange

crankyoldgit, 2019-10-18:
Can you supply the full output of `IRrecvDumpV2`? What is the model number of the remote that is producing the code?

curla92, 2019-10-18:
I don't have the remote i want replace this on my firmware.
This is my complete function:

```
uint8_t DaikinTemplate_152[19] = {
    // header
    0x11, 0xDA, 0x27, 0x00, 0x00, 0x30, 0x34, 0x00, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC5, 0x40, 0x00, 0xAB};
    //  0     1     2     3     4     5     6     7    8    9    10    11    12    13    14    15    16    17   18

uint8_t checksum_152;

DaikinTemplate_152[5] = 0x00; // OFF
            
            // Calculate the checksum
            checksum_152 = 0x00;
            
            for (int i=0; i<18; i++) {
                checksum_152 += DaikinTemplate_152[i];
            }
            
            DaikinTemplate_152[18] = checksum_152;
 

            taskENTER_CRITICAL();
            
            // Header
            ir_mark(3492);
            ir_space(1718);
            
            // First Header
            for (int i=0; i<19; i++) {
                sendIRbyte(DaikinTemplate_152[i], 433, 433, 1529);
            }
            
            ir_mark(433);
            ir_space(0);
            
            taskEXIT_CRITICAL();
```

crankyoldgit, 2019-10-18:
So, I'm guessing you're using the code from: https://github.com/ToniA/arduino-heatpumpir/

There are two issues with that signal.
1) It's missing the leader to the message. ie. 5 zero bits, and a gap of ~25000usecs.
e.g.
https://github.com/crankyoldgit/IRremoteESP8266/blob/master/src/ir_Daikin.cpp#L2839-L2844
That isn't being produced by the code for that project.

2) The message you described has a state code of:
```
uint8_t state[kDaikin152StateLength] = {0x11, 0xDA, 0x27, 0x00, 0x00, 0x00, 0x2C, 0x00, 0xA0, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, 0xC5, 0x00, 0x00, 0xD4};
```
The final byte should be `0xA3`, not `0xD4` .. And that according to my reading of the code for `heatpumpir` and my library.

Something is wrong with what ever checksum calculation code you are using.

curla92, 2019-10-18:
ok perfect i ll read all. actually d esp,216 and 160 works on my firmware. very thanks.

crankyoldgit, 2019-10-23:
Any luck?

crankyoldgit, 2019-10-27:
Friendly progress chase up. As the code you are using isn't really our library's code, and I've offered improved `DAIKIN152` handling & heard no feedback, I'm going to close this issue unless I heard back from you in a day or so.

TL;DR: If you find an issue with our code & library, I'm happy to keep this open if I hear from you soon.

closed 2019-10-30 by crankyoldgit

crankyoldgit, 2019-10-30:
FYI, the changes mention above have been included in the v2.7.0 release of the library.