crankyoldgit/IRremoteESP8266 #978: Question: Can this library be used for communicating via ASCII messages?

Silipwn, 2019-10-25:
Was wondering if we could use this library to communicate via ASCII messages .Tried using the `sendData` function but I wasn't able to receive any data. 

Any tips on where to begin?

crankyoldgit, 2019-10-26:
This topic has come up a few times, so I've written an [FAQ](https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#can-i-send-arbitrary-data-with-this-library-eg-ascii-text) for it.

Please read: https://github.com/crankyoldgit/IRremoteESP8266/wiki/Frequently-Asked-Questions#can-i-send-arbitrary-data-with-this-library-eg-ascii-text

`sendData()` is used as part of a protocol. It's not a protocol in of itself. For the library to detect it as is, it needs to be an established protocol. If you want to create your own protocol, then yes, `sendData` could be a part of it, just not on it's own.

See also #964

crankyoldgit, 2019-10-27:
Was the FAQ text any help to you?

Silipwn, 2019-10-28:
Sorry for the late reply, I think I got my answer for now, I'll try to communicate via the an existing IR protocol for now, and maybe then try to design my protocol later on. Thanks for the FAQ. :+1: :smiley:

closed 2019-10-28 by Silipwn