crankyoldgit/IRremoteESP8266 #979: Problem sending ir code

ByteHD, 2019-10-28:
```#include <ESP8266WiFi.h>
#include <WiFiClient.h>
#include <ESP8266WebServer.h>
#include <IRremoteESP8266.h>
#include <IRsend.h>
#include <IRrecv.h>
#include <IRutils.h>

IRsend irsend(3);
ESP8266WebServer server(80);

void setup() {
  WiFi.mode(WIFI_STA);
  WiFi.begin("Test", "testnet123");
  irrecv.enableIRIn();
  irsend.begin();

  server.on("/send", []() {
    uint32_t code;
    for (uint8_t i = 0; i < server.args(); i-=-1) {
      if (server.argName(i) == "code") {
        code = strtoul(server.arg(i).c_str(), NULL, 10);
        irsend.sendNEC(code, 32);
      }
    }
    server.send(200, "text/html", "sending code " + String(code));
  });
  
  server.begin();
}

void loop() {
  server.handleClient();
}
```

I'm trying to send IR codes through a webpage and wrote a sketch based on the IRServer example.
When I send a GET request ex. http://ip/send?code=25245 the device I want to turn on doesn't.
The IR led lights up under my phone camera. When I repeat a received code ( irsend.sendNEC(results.value); ) it works fine.
Has somebody got an idea to fix it?
thank you

crankyoldgit, 2019-10-28:
First off, you can remove `irrecv.enableIRIn();` from your code as supplied by above. I'm amazed it compiles.

You can also remove the `, 32` from the `irsend.sendNEC(code, 32)` call. That shouldn't make an difference as it is the default.
Also, it took me a minute to work out what `i-=-1` did too. i.e. Use `i++` instead. ;-)

> When I repeat a received code ( irsend.sendNEC(results.value); ) it works fine.

My guess is you're using the wrong value (`code`) in  the`irsend.sendNEC()` call. How sure are you that 25245 (dec) / 0x629D (hex) is the correct value? If you are trying to emulate an actual remote you have, what is the output of `IRrecvDumpV2` for that button press?

I'm assuming that `String(code)` is providing you with the value you expect.

ByteHD, 2019-10-29:
```Encoding  : NEC
Code      : 40BE629D (32 bits)
Timing[67]: 
     +9050, -4500     + 550, - 600     + 550, -1700     + 550, - 600
     + 550, - 550     + 600, - 550     + 550, - 600     + 550, - 550
     + 600, - 550     + 600, -1650     + 600, - 550     + 550, -1700
     + 600, -1650     + 600, -1650     + 600, -1700     + 550, -1700
     + 550, - 600     + 550, - 550     + 600, -1650     + 600, -1700
     + 550, - 550     + 600, - 550     + 550, - 600     + 550, -1700
     + 550, - 600     + 550, -1700     + 550, - 550     + 600, - 550
     + 600, -1650     + 600, -1650     + 600, -1700     + 550, - 550
     + 600, -1650     + 600
unsigned int  rawData[67] = {9050,4500, 550,600, 550,1700, 550,600, 550,550, 600,550, 550,600, 550,550, 600,550, 600,1650, 600,550, 550,1700, 600,1650, 600,1650, 600,1700, 550,1700, 550,600, 550,550, 600,1650, 600,1700, 550,550, 600,550, 550,600, 550,1700, 550,600, 550,1700, 550,550, 600,550, 600,1650, 600,1650, 600,1700, 550,550, 600,1650, 600};  // NEC 40BE629D
unsigned int  data = 0x40BE629D;
```
Found the problem, it wasn't my sending code... I used an ESP-01 to receive with my own receiving code and I messed up conversion between the value (uint32_t) and threw it in a uint16_t

Thank you for your help!

crankyoldgit, 2019-10-30:
I'm glad you found your problem. Good luck with the rest of it.

closed 2019-10-30 by crankyoldgit